raspberrypi/linux #3563: LIRC does not work with 4.19 kernel, need dowgrade instruction

AlexanderShniperson, 2020-04-21:
Hi LIRC doesn't work with kernel `4.19` including fresh new
Can you please share instruction to downgrade to kernel `4.14.98`?
several users told LIRC works fine there

pelwell, 2020-04-21:
Please read this guide on migrating to gpio-ir: https://www.raspberrypi.org/forums/viewtopic.php?t=235256

AlexanderShniperson, 2020-04-21:
it doesn't work

pelwell, 2020-04-21:
There are lots of people using LibreELEC etc. who would disagree.

AlexanderShniperson, 2020-04-25:
after patching LIRC by [this instruction](https://www.raspberrypi.org/forums/viewtopic.php?t=235256) for `Raspbian Buster` i see logs
```
Apr 22 11:06:21 DietPi lircd[3352]: lircd-0.10.1[3352]: Error: error processing command: SEND_ONCE TANK-64 KEY_0
Apr 22 11:06:21 DietPi lircd[3352]: lircd-0.10.1[3352]: Error: transmission failed
Apr 22 11:06:21 DietPi lircd[3352]: lircd-0.10.1[3352]: Info: removed client
Apr 22 11:06:21 DietPi lircd-0.10.1[3352]: Error: invalid send buffer
Apr 22 11:06:21 DietPi lircd-0.10.1[3352]: Error: this remote configuration cannot be used to 
transmit
Apr 22 11:06:21 DietPi lircd-0.10.1[3352]: Error: error processing command: SEND_ONCE TANK-64 
KEY_0
Apr 22 11:06:21 DietPi lircd-0.10.1[3352]: Error: transmission failed
```
also try to setup LIRC with [that instruction](https://github.com/raspberrypi/linux/issues/2993) from `elmicha commented on 30 May 2019` without luck
also updated kernel to latest possible without luck 
`Linux DietPi 4.19.115+ #1305 Fri Apr 17 11:47:30 BST 2020 armv6l GNU/Linux`

AlexanderShniperson, 2020-04-25:
so looks like i got working config and placed setup [at the repo](https://github.com/AlexanderShniperson/rpi-lirc)
seems to be working and data sends but receiver decode signal wrong, that we can see at `journalctl -fu lircd`

AlexanderShniperson, 2020-04-27:
it works, i summarized all [my work in repo](https://github.com/AlexanderShniperson/rpi-lirc)

closed 2020-04-27 by AlexanderShniperson