raspberrypi/linux #3569: DRM VC4: atomic KMS API doesn't synchronize with VBlank

mszyprow, 2020-04-23:
** Issue **
DRM VC4: atomic KMS API doesn't synchronize with VBlank

** Description **
I've tried to use atomic KMS API of the VC4 driver (firmware KMS), but it looks that the page flip event is not properly synchronized with vblank on Raspberry Pi4 board. I've tried various versions of the kernel from this git repo, from rpi-4.19 to rpi-5.6 branches, but in all cases the results are the same. 

I've attached my test application that demonstrates this issue. It performs very simple page flipping + drawing on the back buffer, all done using the atomic KMS API. When launched on Raspberry Pi4, one can observe ugly tearing and flashing of the screen, what means that the buffer, which is being drawn is still being displayed.

I've also check the same application with mainline v5.6 kernel with RPi4 HDMI support patches for VC4 driver (https://lore.kernel.org/dri-devel/cover.6c896ace9a5a7840e9cec008b553cbb004ca1f91.1582533919.git-series.maxime@cerno.tech/T/ ) and it works fine without tearing/flashing, but that driver doesn't support both HDMI outputs of the RaspberryPi4 board yet. I've also checked the test application and it works fine on the different board: Samsung Exynos Odroid XU4 with ExynosDRM driver.

* Which model of Raspberry Pi? 
Pi4b

* Which OS and version (`cat /etc/rpi-issue`)? 
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2

* Which firmware version (`vcgencmd version`)?
Feb 12 2020 12:36:21 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

**To reproduce**
$ ./tearing_test /dev/dri/by-path/platform-soc\:gpu-card
$ ./tearing_test /dev/dri/by-path/platform-soc\:gpu-card background

** Test app **
[tearing_test.c.txt](https://github.com/raspberrypi/linux/files/4520942/tearing_test.c.txt)

mszyprow, 2020-04-23:
One more information: similar issue can be observed on Raspberry Pi3b+ board, but only when 'background' switch is passed to tearing_test tool. I also had to add "dtoverlay=vc4-fkms-v3d" to /boot/config.txt

popcornmix, 2020-04-23:
You can run vc4-kms -v3d on pi4 with "next" firmware/kernel.
See https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769

I'm afraid the tearing/vsync issue with fkms is known about but unlikely to be fixed, as development effort is going to the better kms driver.

mszyprow, 2020-04-23:
Indeed, switching to the rpi-5.4.y kernel with "dtoverlay=vc4-kms-v3d-pi4" fixed this issue. Thanks!

FallingSnow, 2021-05-17:
Oh my god. I've been hitting my head against this issue for weeks. Using `vc4-kms-v3d-pi4` has indeed solved the issue.

pelwell, 2021-05-18:
`dtoverlay=vc4-kms-v3d` is sufficient, unless using an old image.