raspberrypi/linux #3572: rpi4: aarch64 - kernel upgrade from 5.4.28 to 5.4.32 does not detect sd card anymore

nkichukov, 2020-04-23:
**Describe the bug**
sd card is no longer recognized, a lot of errors:
 Failed to get clock (-22) 
 xhci_hcd 0000:01:00.0: ERROR: unexpected setup context command completion code 0x11.
 Mismatch between completed Set TR Deq Ptr command & xHCI internal state.\
 ...

**To reproduce**
1.) clone the latest release of the kernel from rpi-5.4.y branch, currently pointing to: 5eab66dbeb358608e04907a9ed0058d004b84bc7
2.) run make oldconfig, accept the two changes and compile.
3.) Install the new kernel, device tree blobs and kernel modules
4.) reboot and observe the problem

**Expected behaviour**
Expected the system to boot fine, able to recognize the sd card

**Actual behaviour**
sd card is not recognized, system is sluggish

**System**
Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi4 4GB
* Which OS and version? GNU/Gentoo Linux, aarch64
* Which firmware version (`vcgencmd version`)? vcgencmd does not work on aarch64, firmware is latest version
* Which kernel version (`uname -a`)? 5.4.28-v8+

**Logs**
dmesg output attached.
[rpi4-failure-dmesg.txt](https://github.com/raspberrypi/linux/files/4523752/rpi4-failure-dmesg.txt)

**Additional context**
As per the logs, the root is on external usb disk, attached via UAS, thus booting is partially successful.
I also tried to switch to the kms for pi4 as seen in the log above, but this is not relevant for the sd card recognition:
dtoverlay=vc4-kms-v3d-pi4
dtoverlay=rpivid-v4l2

same problem still occurs with:
dtoverlay=vc4-fkms-v3d,cma-512

Thanks!

popcornmix, 2020-04-23:
This suggestion may be relevant.
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&start=125#p1646456

nkichukov, 2020-04-23:
Unfortunately, this patched bcm2711-rpi-4-b.dtb file did not make any difference.

klslz, 2020-04-27:
I also had the boot issue when going from 5.4.29 to .32 on aarch64.
Not sure if this is related.

I figured there were issues with these configs:

start_file=start4cd.elf
fixup_file=fixup4cd.dat

System booted again after disabling these.

I do also have the feeling that the system with .32 became kind of sluggish resp. less responsive.

timg236, 2020-04-27:
It's not safe to force start_file=start4cd.elf it will end up with the wrong amount of GPU memory due to a long-standing limitation in the fixup generation.

klslz, 2020-04-27:
Q: What  changed from .29 >> .32 ? It's been working under .29.

pelwell, 2020-04-27:
The main highlights are:
```
kernel: configs: Enable CONFIG_ARM_BCM2835_CPUFREQ
See: raspberrypi/linux#3554

kernel: Unicam Frame Start event
See: raspberrypi/linux#3551

kernel: Backporting imx219 fixes
See: raspberrypi/linux#3553

kernel: Revert firmware: raspberrypi: register clk device
 
kernel: Bump to 5.4.32 …

firmware: isp_tuner: Add in the slave AWB tuner handling …

firmware: arm_dt: Apply os_prefix to device_tree= files

firmware: clock_2711: Fix PLL analog setup

firmware: board_info: Also include CM3+ for pmic trait
See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=267576&start=25#p1643032
 
kernel: dtoverlay: Add overlay_map functionality …

kernel: clk-raspberrypi: Allow cpufreq driver to also adjust gpu clocks
```
I'm running Raspbian with a self-built top-of-tree rpi-5.4.y 64-bit kernel and it's fine, so it's not broken for everyone. I am getting occasional problems with the vc4-kms-v3d driver starting up (*), but otherwise it looks OK.

(*) Here's the end of the kernel log, in case you see something similar:
```
[   24.862651] Bluetooth: RFCOMM ver 1.11
[   29.150578] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
[   39.390577] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
[   49.630626] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:59:plane-2] flip_done timed out
[   59.870628] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
[   70.110624] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
[   80.350636] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
[   90.590648] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:59:plane-2] flip_done timed out
```
In this state it is also low to get round to rebooting.

nkichukov, 2020-06-21:
Same problem on kernel 5.6.19.

But I see others are able to boot that OK: https://github.com/raspberrypi/linux/issues/3577

So this has to be a kernel option that is not enabled (as it was not needed before) but is required for whatever changed after kernels 5.4.28/29.

nkichukov, 2020-06-21:
maybe the bug I referenced above is for 32bit kernel/userspace, so perhaps not related to this.

nkichukov, 2020-08-11:
I've tried latest rpi-5.4.51 kernel and it produced the same errors...
failure to load firmware
failure to load the clocks
failure to recognize the sdcard
failure to load the xhci_hcd devices

Recompiled the kernel with various different options, nothing helped.

However, limiting the available memory to 1024MB resolved the issue completely without any kernel changes required. Also, no sluggishness is observed, the system works as fast as with kernel 5.4.28.

What worked:
total_mem=1024

Other memory options did not work:
total_mem=3072
or 
total_mem=2048

I remember there were some issues with recognizing sdcard and usb devices on rpi4 2 and 4 GB models with earlier kernels, however those were fixed and limiting the visible memory was no longer required.

This seems to be a regression or perhaps I am missing some kernel option or setting (introduced/used after kernel 5.4.28).

pelwell, 2020-08-11:
You must have a problem with your config or .dtb files - nobody else is reporting this regression.

nkichukov, 2020-08-11:
the .dtbo files are being generated with the kernel build, so this has to be a kernel configuration option.

However why does it work without it on 5.4.28 and it fails so bad on 5.4.32?

What's the in-kernel firmware load has to do with the amount of visible memory?

nkichukov, 2020-08-12:
Hi, 
this is a regression, which got fixed in 5.8.0-rc7. I am currently running kernel 5.8.0-rc7 and everything looks fine, without additional kernel options (except for the newly added from oldconfig from 5.4.28).

Same problem was also reported [here](http://rglinuxtech.com/?p=2768) and gave me a pointer to attempt 5.8 kernels.

This issue can now be closed.