raspberrypi/linux #3573: How to resolve "ERROR: linux-raspberrypi-1_4.19.88+gitAUTOINC+988cc7beac-r0 do_fetch: Fetcher failure  for URL: 'git://github.com/raspberrypi/linux.git;branch=rpi-4.19.y'. Unable to fetch URL from any source."

Siddharthav1, 2020-04-24:
Hello, 

I am trying to build the Base image for RPI3 board using yocto. I am using " bitbake -k core-image-base" command for that. 

  Everything is going well but while building: Task "(/home/siddhu/Documents/yocto/sources/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.19.bb:do_fetch) failed with exit code '1'  " task is failing.

Blow is the part of error log:

During Linux fetching :
     _Cloning into bare repository 
      '/home/siddhu/Documents/yocto/build/downloads/git2/github.com.raspberrypi.linux.git'...
      remote: Enumerating objects: 85, done.        
       remote: Counting objects: 100% (85/85), done.        
       remote: Compressing objects: 100% (67/67), done.        
       _fatal: read error: Connection reset by peer1.62 GiB | 340.00 KiB/s      
        fatal: early EOF
       fatal: index-pack failed__

At the end build failure:
            ERROR: linux-raspberrypi-1_4.19.88+gitAUTOINC+988cc7beac-r0 do_fetch: Fetcher failure 
             for URL: 'git://github.com/raspberrypi/linux.git;branch=rpi-4.19.y'. Unable to fetch URL from 
            any source.
 
Please let me know what's going wrong here in my proceedure.

My work environment is as below:
local host: Laptop with Ubuntu.18.04 OS
Processor: Intel i3 quadcore processor.

Please help me on this. I need your suggestion what I need to do to get complete build image.

Thanks,
Siddhartha V
9972583616

pelwell, 2020-04-24:
GitHub seems to be suffering from heavy loads during the US mornings - I'm seeing slowdowns from around 1500GMT. At times this had been bad enough to prevent me adding a comment on an issue, so I can believe that it might affect cloning.

If you still think there might be a problem with your build process, try following our kernel building instructions: https://www.raspberrypi.org/documentation/linux/kernel/building.md
Because each step is typed (rather than running some build script) you can see where it is going wrong. Or not.

Siddharthav1, 2020-06-17:
Hello RPI-Linux team. I am trying to buils the yocto image for RPI-4-64. But again I am facing the same problem. May I know what can I do ,what wrong I am doing here.please

Regards,
Siddhartha V

timg236, 2020-06-17:
Sorry, we can't support other build systems. You should ask the Yocto maintainers