raspberrypi/linux #3575: missing /dev/dri/card0

sro0, 2020-04-25:
**Describe the bug**
Since 358259f the /dev/dri/card0 device is missing.

**To reproduce**

1. git checkout 126504b7181a30f7591de29c09920367dd824e71
2. compile and install to RPI -> /dev/dri/card0 gets created and works as intended
3. git checkout 358259f8b26987ea15661322d11252c6ccaed037
4. compile and install to RPI -> /dev/dri/card0 is missing -> no GUI

**Expected behaviour**
/dev/dri/card0 always gets created
**Actual behaviour**
/dev/dri/card0 is missing since 358259f

**System**

* Which model of Raspberry Pi: ``Raspberry Pi Zero Rev 1.3``
* Which OS and version: ``gentoo``
* Which firmware version (`vcgencmd version`): ``Apr 20 2020 15:41:24
Copyright (c) 2012 Broadcom
version ed60c22eaee6ee7216b132591919025cba0f68e5 (clean) (release) (start)``
* Which kernel version (`uname -a`): ``Linux rpi 5.4.32+ #119 Sat Apr 25 07:05:33 CEST 2020 armv6l ARMv6-compatible processor rev 7 (v6l) BCM2835 GNU/Linux``

**Logs**
[dmesg_126504b_ok.log](https://github.com/raspberrypi/linux/files/4532401/dmesg_126504b_ok.log)
[dmesg_358259f_fail.log](https://github.com/raspberrypi/linux/files/4532402/dmesg_358259f_fail.log)
[kernel.config.txt](https://github.com/raspberrypi/linux/files/4532403/kernel.config.txt)
[boot_config.txt](https://github.com/raspberrypi/linux/files/4532407/boot_config.txt)

**Additional context**
Possible related messages from dmesg_358259f_fail.log:
```
[    2.333934] vc4_hvs 20400000.hvs: Couldn't get core clock
[    2.334133] vc4-drm soc:gpu: failed to bind 20400000.hvs (ops 0xc074b634): -561414144
[    2.335645] vc4-drm soc:gpu: master bind failed: -561414144
[    2.335689] vc4-drm: probe of soc:gpu failed with error -561414144
```

6by9, 2020-04-25:
Yup, https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/gpu/drm/vc4/vc4_hvs.c#L243
```
	hvs->core_clk = devm_clk_get(&pdev->dev, NULL);
```
needs to be ```devm_clk_get_optional``` for Pi0-3.

Actually looking at the v2 that's just been [sent to dri-devel](https://patchwork.freedesktop.org/patch/362974/), it's been put in an ```if (hvs->hvs5)``` conditional which also works.
I'll be looking to revert the v1 patchset and push v2 on Monday.

sro0, 2020-05-01:
Works again since 5f4e299. Thank you for the quick fix.

closed 2020-05-01 by sro0