raspberrypi/linux #3577: RPi4, Kernel 5.6.x, Kodi does not start up

mkreisl, 2020-04-25:
Hi, having a problem and would need help please.

When running kernel 5.6.x. Kodi does not come up it stucks without any error message. If I install older kernel (5.5.15 in my case), everything works as expected. Kernel 5.6 was build from the working config of 5.5.15. System is running XBian

Do I miss anything? Do I have to add additional things in /boot/config.txt?
This is what I have in /boot/config.txt
```
[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
```

Neither in Kodi's log nor in dmesg (output see below)  I can see any errors, everything seems to be ok:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.6.7+ (manfred@kmnote5) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #1 SMP PREEMPT Sat Apr 25 12:11:46 CEST 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000017400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 933888
[    0.000000]   DMA zone: 1920 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 163840 pages, LIFO batch:31
[    0.000000]   HighMem zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 19 pages/cpu s48524 r8192 d21108 u77824
[    0.000000] pcpu-alloc: s48524 r8192 d21108 u77824 alloc=19*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 931968
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 video=HDMI-A-1:1920x1080M@50 smsc95xx.macaddr=DC:A6:32:09:2D:5F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  telnet zswap.enabled=1 zswap.compressor=lz4 console=tty1 partswap vnc cnet=bond0 root=iSCSI=iqn.2017-12.com.kmhome:kmxbilr2,192.168.1.6:3260,UUID=29e1091f-1092-4787-9748-271d9a670f41 rootflags=subvol=root/@,autodefrag,compress=none rootfstype=btrfs rootwait logo.nologo quiet noswap loglevel=0 startevent=mountall selinux=0 splash nohdparm net.ifnames=0 biosdevname=0 --startup-event mountall
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x103f4000-0x143f4000] (64MB)
[    0.000000] Memory: 3332436K/3735552K available (10240K kernel code, 934K rwdata, 3240K rodata, 2048K init, 878K bss, 140972K reserved, 262144K cma-reserved, 3080192K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31025 entries in 61 pages
[    0.000000] ftrace: allocated 61 pages with 5 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x330/0x500 with crng_init=0
[    0.000003] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000010] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000043] bcm2835: system timer (irq = 17)
[    0.000287] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000294] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000301] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000307] Switching to timer-based delay loop, resolution 18ns
[    0.000426] Console: colour dummy device 80x30
[    0.000442] printk: console [tty1] enabled
[    0.000466] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000476] pid_max: default: 32768 minimum: 301
[    0.000610] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000621] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001219] Disabling memory control group subsystem
[    0.001265] CPU: Testing write buffer coherency: ok
[    0.001523] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.024330] Setting up static identity map for 0x200000 - 0x20003c
[    0.032324] rcu: Hierarchical SRCU implementation.
[    0.048357] smp: Bringing up secondary CPUs ...
[    0.080678] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.112692] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.144744] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.144818] smp: Brought up 1 node, 4 CPUs
[... 461 lines cut ...]
uinput                 20480  0
rfkill                 28672  5 bluetooth,cfg80211
brcmutil               20480  1 brcmfmac

```

Installed firmware files:
```
vcgencmd version
Dec  5 2019 14:42:20 
Copyright (c) 2012 Broadcom
version dfab159feffeba8de79ecb6dc973fab33d07004f (clean) (release) (start_x)
```
Installing lates fw files
```
vcgencmd version
Apr 20 2020 15:40:55 
Copyright (c) 2012 Broadcom
version ed60c22eaee6ee7216b132591919025cba0f68e5 (clean) (release) (start_x)
```
does not make any difference

I noticed that vnc-server (dispmanx) does not work, is stucks anywhere and vnc client does not get any response (writing infinite connecting ...)

Any help would be very appreciated. Thanks in advance

mkreisl, 2020-04-25:
I gave another try, cloned config from 5.5.15 with `make savedefconfig` and used this generated config, but unfortunately absolutely no difference. Kernel source is unmodified from this repository (5.6.4)

Absolutely frustrating :(

pelwell, 2020-04-25:
Try with bcm2711_defconfig.

mkreisl, 2020-04-25:
Did it already (partly) merged  bcm2711_defconfig and defconfig together with my last try. Being sure that all required Pi4 settings are in this config, but will try a native bcm2711_defconfig tomorrow (just building btrfs not as module)

mkreisl, 2020-04-25:
> Try with bcm2711_defconfig.

Did it. Absolutely **no** difference. When Kodi starts, splash screen will be shown forever. Nothing more

So, for me the 5.6 kernel is completely broken :angry:

mkreisl, 2020-05-21:
Tested again and still not working, but found what the culprit is:

Something is broken with the ethernet driver. Usually I'm running bond0 with eth0 and wlan0 as bond devices, with eth0 as default.

If I stop eth0 with `sudo ifdown eth0`, Kodi starts without any issue and even vnc server works again as expected

Removing bond0 and using native eth0 as network connection did not help, do its GENET driver and nothing else!

mkreisl, 2020-05-21:
Found workaround:

```
sudo  ethtool -K eth0 tx off
```

makes it work again