raspberrypi/linux #3579: rpi-5.6.y -- vdso version of clock_gettime is wrong, syscall is correct

jackschmidt, 2020-04-26:
**Is this the right place for my bug report?**

I believe so. With default Raspbian kernel and userland, the program produces correct output. Building 5.6 kernel following https://www.raspberrypi.org/documentation/linux/kernel/building.md and booting the result, the program produces erratic output.

**Describe the bug**

The VDSO version of `clock_gettime( CLOCK_REALTIME, ts );` produces erratic, incorrect output. The syscall version works fine.

**To reproduce**

Boot a stock raspberry pi 4b rev 1.2 with https://downloads.raspberrypi.org/raspbian_lite_latest (2020-02-13) and run apt-get update and apt-get upgrade.

Build the kernel:
```
sudo apt install git bc bison flex libssl-dev make
git clone --depth=1 --branch rpi-5.6.y https://github.com/raspberrypi/linux
cd linux
mkdir ../build
make O=../build bcm2711_defconfig
make O=../build -j4
sudo cp ../build/arch/arm/boot/zImage /boot/kernel7l.img
sudo reboot
```
Now build the program with MUSL libc (I used musl-cross-make)

```C
#include <stdio.h>
#include <unistd.h>
#include <time.h>
#include <syscall.h>

int main(int argc, char**argv) {
  struct timespec ts_sc[1];
  struct timespec ts_vd[1];
  syscall(SYS_clock_gettime64, CLOCK_REALTIME, ts_sc);
  clock_gettime(CLOCK_REALTIME, ts_vd); // musl will use vdso here
  printf(
    "vd: %lld %lld (%llx %llx)\n"
    "sc: %lld %lld (%llx %llx)\n",
    (long long) ts_vd->tv_sec, (long long) ts_vd->tv_nsec,
    (long long) ts_vd->tv_sec, (long long) ts_vd->tv_nsec,
    (long long) ts_sc->tv_sec, (long long) ts_sc->tv_nsec,
    (long long) ts_sc->tv_sec, (long long) ts_sc->tv_nsec
  );
}
```

**Expected behaviour**

The time in seconds (first column) should increase by about 1 per second. This should happen for both rows, VDSO and syscall.

```
$ ./mix
vd: 1587861652 218823486 (5ea4d894 d0afb3e)
sc: 1587861652 218784111 (5ea4d894 d0a616f)
$ ./mix
vd: 1587861652 838336559 (5ea4d894 31f8002f)
sc: 1587861652 838282913 (5ea4d894 31f72ea1)
$ ./mix
vd: 1587861653 427229788 (5ea4d895 1977025c)
sc: 1587861653 427190569 (5ea4d895 19766929)
$ ./mix
vd: 1587861654 20806558 (5ea4d896 13d7b9e)
sc: 1587861654 20748798 (5ea4d896 13c99fe)
$ ./mix
vd: 1587861654 589989110 (5ea4d896 232a84f6)
sc: 1587861654 589934787 (5ea4d896 2329b0c3)
```

**Actual behaviour**

The seconds column for VDSO is erratic. The seconds column for syscall increases by about 1 per second.

```
$ ./mix
vd: 1602382300 871248665 (5f8269dc 33ee3319)
sc: 1587861514 880693130 (5ea4d80a 347e4f8a)
$ ./mix
vd: 1592986975 290581759 (5ef30d5f 1151ecff)
sc: 1587861515 313451055 (5ea4d80b 12aee22f)
$ ./mix
vd: 1600536634 931908777 (5f66403a 378bcca9)
sc: 1587861515 766646428 (5ea4d80b 2db2189c)
$ ./mix
vd: 1591309079 582261861 (5ed97317 22b49c65)
sc: 1587861516 220063133 (5ea4d80c d1de59d)
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?
```sh
$ dmesg | grep Rasp
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
```

* Which OS and version (`cat /etc/rpi-issue`)?

```sh
$ cat /etc/rpi-issue
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
```

* Which firmware version (`vcgencmd version`)?

From the 5.6 kernel:
```
$ vcgencmd version
VCHI initialization failed
```

From stock kernel:
```
$ vcgencmd version
Feb 12 2020 12:36:21
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?

```sh
uname -a
Linux raspberrypi 5.6.4-v7l+ #1 SMP Sat Apr 25 19:34:29 EDT 2020 armv7l GNU/Linux
```

**Logs**

Let me know if you want anything else.

**Additional context**

Related problem report at https://github.com/richfelker/musl-cross-make/issues/96

pelwell, 2020-04-30:
I'm prepared to look at this issue, but "Now build the program with MUSL libc (I used musl-cross-make)" is a bit brief (especially after you say how to build a kernel). What happened to clear, line-by-line instructions?

pelwell, 2020-04-30:
Although, reading your issue report (https://github.com/richfelker/musl-cross-make/issues/96) and the response from the musl-cross-make maintainer it is a general problem with the upstream Linux implementation of the VDSO clock_gettime on ARM.

Closing, as it's a generic ARM Linux problem.

closed 2020-04-30 by pelwell

popcornmix, 2020-05-05:
Just FYI I spotted this: https://github.com/knaerzche/LibreELEC.tv/commit/3efb84748a208f1a65dca79265af13e08af1f98e

was reported to fix random time readings on rockchip. It may be the cause of the issue you are seeing.

arndb, 2020-05-20:
I see commit 45939ce292b4 ("ARM: 8957/1: VDSO: Match ARMv8 timer in cntvct_functional()") getting merged between linux-5.6-rc7 and linux-5.6, this also changes the behavior on bcm2711 (rpi-4) but not bcm2837 (rpi-3) so the clock_gettime and gettimeofday calls should no longer get patched out.

Jack, can you confirm that the kernel you tested was earlier than the final linux-5.6 and did not include that commit?

After this commit, I would expect clock_gettime() and clock_gettime64() to behave the same way, either both returning correct times, or both returning bogus times.

jackschmidt, 2020-05-20:
It was the linux from this repository (https://github.com/raspberrypi/linux.git) at commit 85313d9203e26197d536938b47e983509f5f0e55 with tag rpi-5.6.y
 
I believe it reports itself as 5.6.4, so should include the change. I'm not sure of a reasonable way to ask git the question, but according to scrolling through github's commit log pages 0a27a29496060843ae3a8fe78aaec0062cbd5dfa is an ancestor, gregkh's 5.6.4 release.

arndb, 2020-05-20:
Ok, I see the change in there, and I also now see that the bcm2711.dtsi sets the "arm,cpu-registers-not-fw-configured" property that also leads to the vdso getting patched out on 32-bit kernels, which is likely what happened then, and when the virtual timer is not set up, that also explains why the __arch_get_hw_counter() function returns incorrect data.

pelwell, 2020-05-21:
Can you try with the patch suggested by @popcornmix (https://github.com/knaerzche/LibreELEC.tv/commit/3efb84748a208f1a65dca79265af13e08af1f98e)?

lategoodbye, 2020-05-30:
Okay, now i remember the discussion with Marc Zyngier https://marc.info/?l=linux-arm-kernel&m=156457054122140&w=2

So for my understand there is also an issue in the arm stubs.

popcornmix, 2020-05-31:
@lategoodbye can you clarify what is missing in armstubs?
Are you taking about CNTFRQ and CNTVOFF setup?

lategoodbye, 2020-06-01:
> Are you taking about CNTFRQ and CNTVOFF setup?

Yes, this is what i understood. Only the armstub7.S should be affected.