raspberrypi/linux #3582: vncserver not working

bobypierre, 2020-04-28:
Hello, 

When I do: 
vncserver
I have in return :
vncserver: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory

What's happen please?

pelwell, 2020-04-28:
Please provide some basic information:

1. What type of Raspberry Pi are you using?

2. Which OS (e.g. Raspbian) are you running? `cat /etc/os-release` may provide the information.

3. Which kernel versions? The output of `uname -a` will tell us that.

bobypierre, 2020-04-28:
yes.
- Raspi 3
- PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
- Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- I am using Raspi as server via SSH
Thanks

bobypierre, 2020-04-28:
hello,

I have resolve the issue in installing vnc4server package.

closed 2020-04-28 by bobypierre