raspberrypi/linux #3586: Need to set VC clock to 250 Mhz in the bcm2835-unicam driver

naushir, 2020-04-29:
**Describe the bug**
If the VC core clock drops below 250 MHz, Unicam may underrun and produce corrupt images.  This is handled correctly in the FW driver, but not in the kernel driver.

**To reproduce**
Run a camera usecase that uses the kernel Unicam driver.  However, this is very intermittent, and seems to be board dependent.

**Expected behaviour**
No corrupt images should be produced.

**Actual behaviour**
Corrupt images are produced.

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Found happening on a small number of Pi4 boards.

**Additional context**
Current workaround is to add core_freq_min=250 in /boot/config.txt

naushir, 2020-04-29:
@pelwell or @popcornmix, can this be assigned to me please?

popcornmix, 2020-04-30:
Kernel driver just needs to use mailbox to request a minumum core frequency when active.
@6by9 did this for kms driver during mode switch so can advise.
Here is commit: https://github.com/raspberrypi/linux/commit/b98b151ee36bb894cdb05b59a2446a29fc5f8899