raspberrypi/linux #3593: Missing PPS modules for 64-bit builds of 5.4

trejan, 2020-05-03:
64-bit builds of the 5.4 kernel are missing the PPS modules (pps_core, pps-gpio and pps-ldisc). The other 32-bit builds of 5.4 and all builds of 4.19 do have the PPS modules.

```
./4.19.97+/kernel/drivers/pps/clients/pps-gpio.ko
./4.19.97+/kernel/drivers/pps/clients/pps-ldisc.ko
./4.19.97+/kernel/drivers/pps/pps_core.ko
./4.19.97-v7+/kernel/drivers/pps/clients/pps-gpio.ko
./4.19.97-v7+/kernel/drivers/pps/clients/pps-ldisc.ko
./4.19.97-v7+/kernel/drivers/pps/pps_core.ko
./4.19.97-v7l+/kernel/drivers/pps/clients/pps-gpio.ko
./4.19.97-v7l+/kernel/drivers/pps/clients/pps-ldisc.ko
./4.19.97-v7l+/kernel/drivers/pps/pps_core.ko
./4.19.97-v8+/kernel/drivers/pps/clients/pps-gpio.ko
./4.19.97-v8+/kernel/drivers/pps/clients/pps-ldisc.ko

./5.4.35+/kernel/drivers/pps/clients/pps-gpio.ko
./5.4.35+/kernel/drivers/pps/clients/pps-ldisc.ko
./5.4.35+/kernel/drivers/pps/pps_core.ko
./5.4.35-v7+/kernel/drivers/pps/clients/pps-gpio.ko
./5.4.35-v7+/kernel/drivers/pps/clients/pps-ldisc.ko
./5.4.35-v7+/kernel/drivers/pps/pps_core.ko
./5.4.35-v7l+/kernel/drivers/pps/clients/pps-gpio.ko
./5.4.35-v7l+/kernel/drivers/pps/clients/pps-ldisc.ko
./5.4.35-v7l+/kernel/drivers/pps/pps_core.ko

```
Diff between 4.19 and 5.4 version of arch/arm64/configs/bcm2711_defconfig shows these two config statements are missing:

```
CONFIG_PPS_CLIENT_LDISC=m
CONFIG_PPS_CLIENT_GPIO=m
```

pelwell, 2020-05-04:
I can't explain how those options got lost - the history has been squashed - but they have been restored in rpi-5.4.y and rpi-5.6.y, and hopefully will remain from now on.

trejan, 2020-05-04:
Thanks for the fix

closed 2020-05-04 by trejan

lightswitch05, 2023-11-23:
I'm using the 64-bit build of bookworm on a Pi 3 Model B+, and I'm having some issue getting a `dev/pps` device created. Looking at `lsmod | grep pps`, I see the `pps_gpio` module is loaded, but I'm no longer seeing `pps_core` as an available module. 

I'm way outside of my depth here, so please forgive me if I'm mistake, but could the `pps_core` module been dropped again?