raspberrypi/linux #3594: Warning in boot log about vc4/vc4_hdmi_phy.c

tompreston, 2020-05-04:
**Is this the right place for my bug report?**
Possibly, it's to do with the HDMI component of the vc4 driver. I don't know if this should live here or in [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware).

**Describe the bug**
On Raspberry Pi 4 with `dtoverlay=vc4-kms-v3d` or `dtoverlay=vc4-kms-v3d-pi4`, I see a warning at boot time relating to the vc4 HDMI component.
```
[    6.055586] WARNING: CPU: 1 PID: 135 at drivers/gpu/drm/vc4/vc4_hdmi_phy.c:414 vc5_hdmi_phy_init+0x7b4/0x2078 [vc4]
```

**To reproduce**
Connect rpi4 to HDMI display. Build and boot rpi-5.4.y on Raspberry Pi 4. Check dmesg.

**Expected behaviour**
There should be no warning.

**Actual behaviour**
A warning appears in the boot log

**System**
* Which model of Raspberry Pi? *Raspberry Pi 4*
* Which OS and version (`cat /etc/rpi-issue`)? *Custom OS, but I believe this is a kernel issue.*
* Which firmware version (`vcgencmd version`)? I'm not sure how to check this without `vcgencmd`
* Which kernel version (`uname -a`)? `Linux localhost 5.4.35-v8 #1 SMP PREEMPT Fri May 1 17:22:32 UTC 2020 aarch64 GNU/Linux`

**Logs**
Full dmesg https://pastebin.com/4NYaaEffper

**Additional context**
The warning comes from the [VC4_SET_FIELD macro](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/gpu/drm/vc4/vc4_hdmi_regs.h#L11), which checks that the register value is contained within the register mask. It looks like the warning is because [phy_get_cp_current](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/gpu/drm/vc4/vc4_hdmi_phy.c#L185) returns an 8-bit value, but the mask is only 6 bits. Either way the mask is applied so the 0xc8 turns into 0x08.

I need to find out what the register is supposed to be set to and what the effect of setting it to 0x08 is, but I don't know where the datasheet for the BCM2711 HDMI registers is.

Please can someone point me to the datasheet containing the registers for the BCM2711 HDMI peripheral.

Thanks!

6by9, 2020-05-04:
I thought @popcornmix had already committed a fix for this. The correct value is 0x18.

popcornmix, 2020-05-04:
It was cherry picked by you into https://github.com/raspberrypi/linux/pull/3535 but that PR got delayed.
I'll commit it directly.

6by9, 2020-05-04:
Thanks @popcornmix - I'm glad someone remembers these things!

tompreston, 2020-05-04:
@6by9 You mean the return value from `phy_get_cp_current()` is 0x18? That makes more sense, but I still don't understand why. Presumably 0x18 and 0x1c are (electrical) current values which need to be set depending on the vco frequency? :shrug: :smile: 

Out of curiosity, is the datasheet publicly available?

Thanks!

6by9, 2020-05-04:
> You mean the return value from phy_get_cp_current() is 0x18? That makes more sense, but I still don't understand why.

They are values provided by Broadcom.
There was a typo in transferring the value from the Broadcom code to the firmware, and that was replicated in the KMS driver. It's now fixed.

> Out of curiosity, is the datasheet publicly available?

No.

tompreston, 2020-05-04:
I'm just testing this, then I'll close the issue.

tompreston, 2020-05-04:
I can confirm the warning goes away with https://github.com/raspberrypi/linux/commit/71d5db1b342097fa4dc561202837beb934648b25. Thanks!

closed 2020-05-04 by tompreston