raspberrypi/linux #3595: Same Raspian System, one RPiZw will boot, one won't.

JimBinMaine, 2020-05-04:
So, I have two RPZWs. One will boot an 64G SD Card I have Raspbian on, and one won’t. I’ve tried rewriting the SD Card a couple of times with different SW, but nothing works. The rest of the system is the same, same SD Card, same PS, same cables, only Keyboard & Mouse and HDMI display for peripherals.

So here’s the console messages of the bad RPZW when booting:

Starting Operating System Debian_Buster_Raspbian_FULL_2019.10.img192
[ 21.691818] Unable to handle kernel paging request at virtual address 80a97458
[ 21.692106] pgd =a446917b
[ 21.692204] [80a97458] *pgd=00ao841e(bad)
[ 21.692343] Internal Error: Oops: 8000000d [#1] SMP ARM

Any idea what the problem is?

pelwell, 2020-05-04:
If the same software works in one Pi but not in what ought to be an identical Pi then it suggests that one of them has a hardware problem.

1. What's the history of those two Zero Ws? Do they appear to be identical?
2. Do you have any conditional sections in your config.txt file - things like `[12345678]` (where the string is a serial number) or `[edid=...]`?
3. Can you post any more of the log from the boot failure? A photo is fine if you can't capture it any other way.

JimBinMaine, 2020-05-05:
I bought them off ebay.  The one that doesn't work is a kit from Vilros.

How do I turn on verbose logging?

pelwell, 2020-05-05:
> How do I turn on verbose logging?

Remove the `quiet` from cmdline.txt (or change it to `xquiet`, to help remind you that it used to be there).

JimBinMaine, 2020-05-10:
I removed quiet from the one line cmdline.txt.  It was actually in there twice.  It actually did not give me any more console messages.  There are a lot more message initially before berryboot runs, put only the few lines after berryboot.

JimBinMaine, 2020-05-10:
[image]
[image]
[image]

JimBinMaine, 2020-05-10:
I guess only the first pic is usable.