raspberrypi/linux #3596: RPi4, Kernel 5.4.32 and higher, HEVC Video playback broken with Kodi

mkreisl, 2020-05-04:
Kernel version 5.4.32 and higher on RPi4 breaks playback of HEVC videos using @popcornmix leia_pi4_18.6 under XBian.  It seems like something broke with codec support or detection under this kernel.  Downgrading to 5.4.27 restores normal operation.

Neither rpivid_mem nor rpivid_hevc is working

See thread https://forum.kodi.tv/showthread.php?tid=354154

graysky2, 2020-05-04:
@mkreisl - Do you know which commit corresponds to 5.4.32 xbian used?  I'd like to build it and try it on Arch ARM.  https://github.com/raspberrypi/linux/commits/rpi-5.4.y

mkreisl, 2020-05-04:
> Do you know which commit corresponds to 5.4.27 xbian used? I'd like to build it and try it on Arch ARM

No, unfortunately I don't. I built that kernel at March 23, that's all I know. Source has gone after upgrading to 5.4.32 :disappointed:

pelwell, 2020-05-04:
Does your build system not record a commit hash or top commit string?

mkreisl, 2020-05-04:
Not sure, but I think it does not..

mkreisl, 2020-05-04:
But that wouldn't help much either. The fact is that now the kernel 5.4 is broken, it was not enough to break the 5.6 kernel

graysky2, 2020-05-04:
> No, unfortunately I don't. I built that kernel at March 23, that's all I know. Source has gone after upgrading to 5.4.32 disappointed

See [this](https://github.com/raspberrypi/linux/commits/rpi-5.4.y?after=4f10063ffdc292908e3d40e844cc37902e9e65de+104) page.  If I am not mistaken, on 32-March you would have pulled https://github.com/raspberrypi/linux/commit/d6a0423e9b0a274fd209d86e4c3765adba970adb

mkreisl, 2020-05-04:
Might be possible, I already got that far. But the fact that there hasn't been a single commit for 4 weeks (Feb 17 - Mar 25) seems strange to me

JamesH65, 2020-05-05:
> But that wouldn't help much either. The fact is that now the kernel 5.4 is broken, it was not enough to break the 5.6 kernel

Both those kernels are still under test on Raspberry Pi, you should not be expecting them to be robust enough for deployment in critical areas. Our current release kernel is 4.19.

graysky2, 2020-05-05:
@JamesH65 - Is there an official source of which kernel is current or stable for the hardware?  Thanks!

JamesH65, 2020-05-05:
The current release kernel is the one you get from apt. Which is currently 4.19.97.

graysky2, 2020-05-05:
Probably safe to close this ticket since it is related to a branch of Kodi.  See [this thread](https://forum.kodi.tv/showthread.php?tid=354154) for a solution.

pelwell, 2020-05-05:
I think @mkreisl should do that once he confirms that a change to Kodi solves his problem.

mkreisl, 2020-05-05:
Tried the workaround and this fixes the issue for me. So will close this issue now

closed 2020-05-05 by mkreisl