raspberrypi/linux #3598: 64-bit 5.4 kernel shows a page_alloc warning in bcm2708_fb if no display attached

trejan, 2020-05-05:
**Describe the bug**
64-bit 5.4 kernel shows a [__alloc_pages_nodemask](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/mm/page_alloc.c#L4727) warning for invalid order in bcm2708_fb if no display attached. 32-bit 5.4 kernel, forcing HDMI output or attaching a display will avoid this warning.

Size of GPU RAM allocation doesn't affect it. It will give the warning whether you have 16MB or 64MB.

**To reproduce**
Use 64-bit 5.4 kernel on a Pi 4 with no HDMI display and no hdmi_force_hot_plug=1. Warning shown in dmesg. /boot/config.txt is stock except for arm_64bit=1

**Expected behaviour**
```
[    0.983182] bcm2708_fb soc:fb: Unable to determine number of FB's. Assuming 1
[    0.983215] bcm2708_fb soc:fb: FB found 1 display(s)
[    0.984414] raspberrypi-firmware soc:firmware: Request 0x00048003 returned status 0x80000001
[    0.984454] bcm2708_fb soc:fb: Failed to allocate GPU framebuffer (-22)
[    0.984522] bcm2708_fb soc:fb: probe failed, err -22
[    0.984560] bcm2708_fb: probe of soc:fb failed with error -22
```

**Actual behaviour**
```
[    0.982031] bcm2708_fb soc:fb: Unable to determine number of FB's. Assuming 1
[    0.982061] bcm2708_fb soc:fb: FB found 1 display(s)
[    0.982922] ------------[ cut here ]------------
[    0.982962] WARNING: CPU: 0 PID: 1 at mm/page_alloc.c:4727 __alloc_pages_nodemask+0x2d4/0x338
[    0.982982] Modules linked in:
[    0.983004] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.4.35-v8+ #1313
[    0.983021] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[    0.983039] pstate: 20000005 (nzCv daif -PAN -UAO)
[    0.983058] pc : __alloc_pages_nodemask+0x2d4/0x338
[    0.983081] lr : __dma_direct_alloc_pages+0x180/0x230
[    0.983095] sp : ffffffc01002b860
[    0.983109] x29: ffffffc01002b860 x28: ffffff80f6244168 
[    0.983129] x27: ffffff80f6244080 x26: ffffff80f6244080 
[    0.983148] x25: 0000000000000cc1 x24: 0000000000000000 
[    0.983167] x23: 000fffffffffffff x22: ffffffc010e28000 
[    0.983185] x21: 0000000000000000 x20: 0000000000000034 
[    0.983203] x19: 0000000000000cc1 x18: ffffffc010e28888 
[    0.983221] x17: 00000000edb1e13a x16: 00000000a229a22b 
[    0.983239] x15: ffffffc09002ba07 x14: ffffffffffffffff 
[    0.983257] x13: ffffffc01000d000 x12: ffffffffffffffff 
[    0.983275] x11: 0000000000000040 x10: 0000000000000000 
[    0.983292] x9 : 0000000000000000 x8 : ffffffc010e28888 
[    0.983310] x7 : 0000000000000000 x6 : 0000000000000000 
[    0.983327] x5 : 0000000000000000 x4 : 0000000000000000 
[    0.983345] x3 : 0000000000000000 x2 : 0000000000000000 
[    0.983362] x1 : 0000000000000000 x0 : ffffffc010e28888 
[    0.983380] Call trace:
[    0.983397]  __alloc_pages_nodemask+0x2d4/0x338
[    0.983415]  __dma_direct_alloc_pages+0x180/0x230
[    0.983432]  arch_dma_alloc+0xe0/0x190
[    0.983449]  dma_direct_alloc+0x80/0x98
[    0.983467]  dma_alloc_attrs+0x88/0xf8
[    0.983487]  bcm2708_fb_set_par+0x154/0x3b8
[    0.983505]  bcm2708_fb_probe+0x318/0x630
[    0.983523]  platform_drv_probe+0x54/0xb0
[    0.983539]  really_probe+0x110/0x390
[    0.983555]  driver_probe_device+0x60/0x108
[    0.983572]  device_driver_attach+0x74/0x80
[    0.983588]  __driver_attach+0xc0/0x100
[    0.983603]  bus_for_each_dev+0x78/0xe0
[    0.983619]  driver_attach+0x30/0x40
[    0.983634]  bus_add_driver+0x158/0x210
[    0.983649]  driver_register+0x68/0x118
[    0.983666]  __platform_driver_register+0x54/0x60
[    0.983688]  bcm2708_fb_init+0x20/0x28
[    0.983706]  do_one_initcall+0x54/0x2b8
[    0.983724]  kernel_init_freeable+0x200/0x2b8
[    0.983744]  kernel_init+0x18/0x110
[    0.983760]  ret_from_fork+0x10/0x1c
[    0.983780] ---[ end trace 9033f6bceb5c0a26 ]---
[... 211 lines cut ...]
[    0.983781]  kernel_init+0x18/0x110
[    0.983798]  ret_from_fork+0x10/0x1c
[    0.983817] ---[ end trace 462f4fbe43a65efb ]---
[    0.983934] raspberrypi-firmware soc:firmware: Request 0x00048003 returned status 0x80000001
[    0.983966] bcm2708_fb soc:fb: Failed to allocate GPU framebuffer (-22)
[    0.984026] bcm2708_fb soc:fb: probe failed, err -22
[    0.984065] bcm2708_fb: probe of soc:fb failed with error -22
[    0.985923] raspberrypi-clk firmware-clocks: CPU frequency range: min 600000000, max 1500000000
[    0.988210] Serial: 8250/16550 driver, 1 ports, IRQ sharing enabled
[    0.990159] iproc-rng200 fe104000.rng: hwrng registered
[    0.990503] vc-mem: phys_addr:0x00000000 mem_base=0x3f000000 mem_size:0x3f600000(1014 MiB)
[    0.991383] gpiomem-bcm2835 fe200000.gpiomem: Initialised: Registers at 0xfe200000
[    0.991760] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    1.002010] brd: module loaded
[    1.013660] loop: module loaded
[    1.014717] Loading iSCSI transport class v2.0-870.
[    1.016503] libphy: Fixed MDIO Bus: probed
[    1.017029] bcmgenet fd580000.ethernet: failed to get enet clock
[    1.017062] bcmgenet fd580000.ethernet: GENET 5.0 EPHY: 0x0000
[    1.017093] bcmgenet fd580000.ethernet: failed to get enet-wol clock
[    1.017123] bcmgenet fd580000.ethernet: failed to get enet-eee clock
[    1.017159] bcmgenet: Skipping UMAC reset
[    1.026631] libphy: bcmgenet MII bus: probed
[    1.078684] unimac-mdio unimac-mdio.-19: Broadcom UniMAC MDIO bus
```

pelwell, 2020-05-05:
The warning is caused by attempting to allocate a zero-sized framebuffer, which gets surprisingly far down the stack.

@6by9 This used to result in a sequence like this:
```
[    0.473819] bcm2708_fb soc:fb: Unable to determine number of FB's. Assuming 1
[    0.473847] bcm2708_fb soc:fb: FB found 1 display(s)
[    0.474176] bcm2708_fb soc:fb: Failed to allocate GPU framebuffer (-22)
[    0.474227] bcm2708_fb soc:fb: probe failed, err -22
[    0.474263] bcm2708_fb: probe of soc:fb failed with error -22
```
which still isn't great, but it's better than a backtrace.

Since we have no idea how large to make the image, and there's no way to get the image displayed if a cable is eventually plugged in, I think it's preferable to bail out early (the line numbers will be out because of additional debug code):
```diff
@@ -1104,10 +1105,9 @@ static int bcm2708_fb_probe(struct platform_device *dev)
         * set one display
         */
        if (ret || num_displays == 0) {
-               num_displays = 1;
                dev_err(&dev->dev,
-                       "Unable to determine number of FB's. Assuming 1\n");
-               ret = 0;
+                       "Unable to determine number of FBs. Disabling driver.\n");
+               return -ENOENT;
        } else {
                fbdev->firmware_supports_multifb = 1;
        }
```

6by9, 2020-05-05:
This was @JamesH65 's code, but if we have the option of bailing out, then that seems reasonable to me.

JamesH65, 2020-05-05:
Not sure why I defaulted to one, I suspect I either wasn't thinking about the no display at all use case, or I was considering backwards compatibility - ISTR it was one before the changes for multi displays.

6by9, 2020-05-05:
We had reverted the patch that did a CMA / dma_alloc_coherent allocation for the frame buffer, and went back to relying on a reloc heap allocation for it. That was part of trying to reduce reloc heap usage, but we didn't resolve what to do should the CMA heap be insufficient for the FB. 
I know we did want to add __GFP_NOWARN to avoid the warning in that case, but I think this is coming up with a dodgy resolution and asking for something silly.

satmandu, 2020-05-05:
For what it is worth, I have noticed this when ```dtoverlay=vc4-kms-v3d``` is set on the 5.6 kernels but not when ```dtoverlay=vc4-fkms-v3d``` is set in config.txt

6by9, 2020-05-05:
> For what it is worth, I have noticed this when dtoverlay=vc4-kms-v3d is set on the 5.6 kernels but not when dtoverlay=vc4-fkms-v3d is set in config.txt

That implies something is wrong in the DT config.  Both of those should disable the framebuffer node that loads bcm2708_fb.
Or are you on a Pi4? I'm not sure if the dtoverlay=vc4-fkms-v3d changes for Pi4 have been merged into 5.6, so it may be doing nothing.

pelwell, 2020-05-05:
The initial issue should be solved by the commit "video: bcm2708_fb: Disable FB if no displays found".

trejan, 2020-05-05:
Apologies. The warning is only shown on 64-bit 5.4 kernel if vc4-fkms-v3d is not loaded. I was mistaken when I said I hadn't altered /boot/config.txt apart from enabling 64-bit mode.

pelwell, 2020-05-05:
@satmandu Are you sure? I'm not seeing any problem with either vc4-kms-v3d or vc4-fkms-v3d on either 5.4 or 5.6 - only when neither overlay is selected, and that is solved by the recent commit.

satmandu, 2020-05-06:
@pelwell With the caeat that this is a slightly modified kernel, the issue was still there as of earlier today:
```dtoverlay=vc4-fkms-v3d``` dmesg: https://paste.ubuntu.com/p/Qt4xN8WnKT/
```dtoverlay=vc4-kms-v3d``` dmesg: https://paste.ubuntu.com/p/yWPnSGpMfh/

(I'll try with the new commit and see what happens.)

satmandu, 2020-05-06:
@pelwell The error message is gone now with a 5.6 kernel compiled from https://github.com/raspberrypi/linux/commit/0716e3e044223434e4160a5dd50c91d05751766d and using ```dtoverlay=vc4-kms-v3d```.

pelwell, 2020-05-06:
If you don't mind a bit of testing, could you try with vc4-kms-v3d-pi4 instead? I suspect that either your firmware is old or your build system hasn't included overlay_map.dtb in the overlays directory, meaning that it can't load the Pi 4-specific overlay automatically.

You can also debug this by reverting the patch and using `sudo vcdbg log msg`, assuming the command is available to you.

pelwell, 2020-05-06:
Actually, you don't even need to revert my patch - the presence of any bcm2708_fb messages in the log suggests that the vc4-kms-v3d(-pi4) hasn't been applied correctly.

satmandu, 2020-05-06:
@pelwell Here's my log output: https://paste.ubuntu.com/p/SVbqwbbXSJ/
(And yes I'm using Feb 12 firmware.)

pelwell, 2020-05-06:
> Here's my log output:

So you've used vc4-kms-v3d-pi4, and the problem has gone away - no mention of the frame buffer, and the v3d driver can be seen initialising. Your firmware is too old to be using vc4-kms-v3d, but vc4-kms-v3d-pi4 is a good substitute until you update.

satmandu, 2020-05-06:
@pelwell Here's what I see with the current 5.4 kernel8.img: https://paste.ubuntu.com/p/btwT6StMkj/
```
[    4.453129] ------------[ cut here ]------------
[    4.453194] WARNING: CPU: 1 PID: 123 at drivers/gpu/drm/vc4/vc4_hdmi_phy.c:420 vc5_hdmi_phy_init+0x18a8/0x1df0 [vc4]
[    4.453198] Modules linked in: spidev v3d vc4(+) gpu_sched drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops i2c_bcm2835 spi_bcm2835 drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd
[    4.453247] CPU: 1 PID: 123 Comm: systemd-udevd Not tainted 5.4.35-v8+ #1313
[    4.453252] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[    4.453259] pstate: 20000005 (nzCv daif -PAN -UAO)
[    4.453301] pc : vc5_hdmi_phy_init+0x18a8/0x1df0 [vc4]
[    4.453342] lr : vc4_hdmi_encoder_enable+0x1ac/0x1a78 [vc4]
[    4.453346] sp : ffffffc0128e3150
[    4.453351] x29: ffffffc0128e3150 x28: ffffff80f09e3cf0 
[    4.453359] x27: 0000000000000000 x26: 0000000000000038 
[    4.453367] x25: ffffffc008b99d00 x24: ffffff80f16cfd80 
[    4.453374] x23: 0000000000002060 x22: 0000000000000300 
[    4.453381] x21: 000000010989e7c0 x20: 0000000008d9ee20 
[    4.453387] x19: ffffff80f6781080 x18: ffffffc010e28888 
[    4.453395] x17: 0000000010624dd3 x16: 0000000000000014 
[    4.453401] x15: ffffff80f660091c x14: 0000000000000000 
[    4.453408] x13: ffffffc01000d000 x12: 0000000000000001 
[    4.453414] x11: 0000000000000007 x10: 0000000000000000 
[    4.453421] x9 : ffffffc01000d038 x8 : 0000000000000001 
[    4.453428] x7 : 0000000000000000 x6 : ffffffc010e29000 
[    4.453434] x5 : 000000000000004c x4 : ffffffc008b99d00 
[    4.453441] x3 : ffffffc008b8cc24 x2 : 0000000000000000 
[    4.453447] x1 : ffffffc011581f00 x0 : 00000000dc8984ff 
[    4.453454] Call trace:
[    4.453495]  vc5_hdmi_phy_init+0x18a8/0x1df0 [vc4]
[    4.453534]  vc4_hdmi_encoder_enable+0x1ac/0x1a78 [vc4]
[    4.453580]  drm_atomic_helper_commit_modeset_enables+0x210/0x240 [drm_kms_helper]
[    4.453619]  vc4_atomic_complete_commit+0xe4/0x560 [vc4]
[    4.453657]  vc4_atomic_commit+0xf4/0x190 [vc4]
[    4.453756]  drm_atomic_commit+0x54/0x60 [drm]
[    4.453851]  drm_client_modeset_commit_atomic.isra.12+0x180/0x1f0 [drm]
[    4.453943]  drm_client_modeset_commit_force+0x68/0x1a0 [drm]
[    4.453986]  drm_fb_helper_restore_fbdev_mode_unlocked+0x74/0xc8 [drm_kms_helper]
[    4.454028]  drm_fb_helper_set_par+0x34/0x68 [drm_kms_helper]
[    4.454043]  fbcon_init+0x4c8/0x558
[    4.454053]  visual_init+0xb8/0x110
[    4.454062]  do_bind_con_driver+0x1d8/0x3c8
[    4.454069]  do_take_over_console+0x144/0x1d0
[    4.454076]  do_fbcon_takeover+0x80/0x108
[    4.454084]  fbcon_fb_registered+0x144/0x158
[    4.454090]  register_framebuffer+0x1cc/0x320
[    4.454131]  __drm_fb_helper_initial_config_and_unlock+0x2e4/0x470 [drm_kms_helper]
[    4.454173]  drm_fbdev_client_hotplug+0xf4/0x188 [drm_kms_helper]
[    4.454215]  drm_fbdev_generic_setup+0xa0/0x140 [drm_kms_helper]
[    4.454254]  vc4_drm_bind+0x1a4/0x1b8 [vc4]
[    4.454263]  try_to_bring_up_master+0x160/0x1d0
[    4.454270]  component_master_add_with_match+0xcc/0x118
[    4.454309]  vc4_platform_drm_probe+0xdc/0x110 [vc4]
[    4.454317]  platform_drv_probe+0x54/0xb0
[    4.454323]  really_probe+0x110/0x390
[    4.454330]  driver_probe_device+0x60/0x108
[    4.454336]  device_driver_attach+0x74/0x80
[    4.454341]  __driver_attach+0xc0/0x100
[    4.454347]  bus_for_each_dev+0x78/0xe0
[    4.454352]  driver_attach+0x30/0x40
[    4.454357]  bus_add_driver+0x158/0x210
[    4.454363]  driver_register+0x68/0x118
[    4.454370]  __platform_driver_register+0x54/0x60
[    4.454408]  vc4_drm_register+0x4c/0x58 [vc4]
[    4.454416]  do_one_initcall+0x54/0x2b8
[    4.454426]  do_init_module+0x60/0x20c
[    4.454434]  load_module+0x2000/0x24b0
[    4.454442]  __do_sys_finit_module+0xe0/0xf8
[    4.454449]  __arm64_sys_finit_module+0x28/0x38
[    4.454459]  el0_svc_common.constprop.1+0xe0/0x1a8
[    4.454466]  el0_svc_handler+0x34/0xa0
[    4.454472]  el0_svc+0x8/0xc
[    4.454477] ---[ end trace 1710a4108796f406 ]---
```
With the May1st Firmware and a newer kernel, I no longer get those issues: https://paste.ubuntu.com/p/YrDRfjptdh/

closed 2020-05-15 by trejan