raspberrypi/linux #3606: 8 UVCs can't work well on RP4B

steven-hz, 2020-05-11:
6 uvc can works fine with 2 powered hubs, but when the num of uvc is 8 , RP4B will report :

USB isochronous frame lost (-18).
`[2020/05/11 16:17:43][  575.912821] uvcvideo: USB isochronous frame lost (-18).
`
Then, the v4l2 api can't get the the random 2 video streams through VIDIOC_DQBUF cmd of ioctl.

Follow is th log of lsusb  and lsusb -v 
```pi@raspberrypi:$ lsusb
Bus 002 Device 002: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 003: ID 2109:0817 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 1d6b:0102 Linux Foundation EEM Gadget
Bus 001 Device 006: ID 1d6b:0102 Linux Foundation EEM Gadget
Bus 001 Device 005: ID 1d6b:0102 Linux Foundation EEM Gadget
Bus 001 Device 004: ID 1d6b:0102 Linux Foundation EEM Gadget
Bus 001 Device 003: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 012: ID 1d6b:0102 Linux Foundation EEM Gadget
Bus 001 Device 011: ID 1d6b:0102 Linux Foundation EEM Gadget
Bus 001 Device 010: ID 1d6b:0102 Linux Foundation EEM Gadget
Bus 001 Device 009: ID 1d6b:0102 Linux Foundation EEM Gadget
Bus 001 Device 008: ID 2109:2817 VIA Labs, Inc. 
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:$ lsusb -v

Bus 002 Device 002: ID 2109:0813 VIA Labs, Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x0813 
  bcdDevice           90.11
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes           19
          Transfer Type            Interrupt
[... 637 lines cut ...]
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       6
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x1400  3x 1024 bytes
        bInterval               1

  
 

```

nschurando, 2020-06-02:
Wouldn't that be because of usb bandwidth limitations?

JamesH65, 2020-06-02:
What's a UVC? (Never assume people know abbreviations in your bug reports). Link to the particular 'UVC' you are using?

ptesarik, 2020-06-02:
USB video device class: https://en.wikipedia.org/wiki/USB_video_device_class

P33M, 2020-06-02:
When the errors are appearing, what is the output of `sudo cat /sys/kernel/debug/usb/devices`?

steven-hz, 2020-06-04:
> Wouldn't that be because of usb bandwidth limitations?

Maybe it is, but  how can I confirm this?

steven-hz, 2020-06-04:
> When the errors are appearing, what is the output of `sudo cat /sys/kernel/debug/usb/devices`?

 This output will be given by next test cycle while now is testing 6 UVCS.

JamesH65, 2020-06-04:
> > Wouldn't that be because of usb bandwidth limitations?
> 
> Maybe it is, but how can I confirm this?

Work out what the bandwidth each device is taking - video is fairly easy to approximate - video size, bit depth/pixel and fps gives you a basic number.  Multiply it up by number of devices. Always assume that actual bandwidth of USB is about half the stated theoretical max. Drop a bit more for contention handling. Approximate, but useful numbers.

kbingham, 2020-12-02:
Bandwidth could indeed be an issue here.
One of my colleagues has made a patch / workaround to be able to manually reduce the bandwith consumed by UVC.
It might be worth trying/investigating.

If it solves your problem do let us know, as we're struggling to figure out a good way to solve this for upstream, generic use cases, so any feedback on it actually solving something real will be helpful.

https://www.thegoodpenguin.co.uk/blog/multiple-uvc-cameras-on-linux/