raspberrypi/linux #3610: mcp251x: Probe failed (spi0.0), with Error -34

embedience, 2020-05-14:
Hi,
I just recently used a Pi1 with a CAN-MCP2515 (Board from NiRen) with the most recent raspbian distro.
As I want to communicate with another Pi I added the same wiring to a Pi2 board and configured it the same.
It returns an error in the kernel messages (dmesg)
"mcp251x: probe of spi0.0 failed with error -34"

I could not find a description of the error -34, as most websites with described issues with error 16 or 19.
What can it be or where can I find further material?

best regards
Andreas

pelwell, 2020-05-14:
-34 is -ERANGE. It is only used by the mcp251x driver in one place:
```
	/* Sanity check */
	if (freq < 1000000 || freq > 25000000)
		return -ERANGE;
```
It's complaining that it has been configured with an oscillator which is running too fast or too slow. The default frequency from the overlay is 16MHz - have you changed this, for example by using the `oscillator` parameter, and mistyped the value?

embedience, 2020-05-19:
Thanks that was exacly what I was looking for! Thanks

closed 2020-05-19 by embedience

muazmalas, 2022-01-16:
@embedience can you elaborate on how you fixed the issue. I'm facing very similar issue.

embedience, 2022-01-20:
The code above @pelwell described tells that the value for the oscillator needs to be between the stated values.
Set it accordingly and the error will go away.