raspberrypi/linux #3615: nftables kernel modules missing

steveharriss, 2020-05-15:
Errors on running nft as kernel modules are missing from:
/lib/modules/5.4.40-v7l+/kernel/net/netfilter

Missing nf_tables.ko and all relevant nft_*.ko files

sudo apt install nftables
nftables v0.9.0 (Fearless Fosdick)
nftables.service                                                                                    loaded failed failed    nftables

If you delete the minimal nftables.conf file you can start the service
systemctl status nftables
● nftables.service - nftables
Loaded: loaded (/lib/systemd/system/nftables.service; enabled; vendor preset: enabled)
Active: active (exited) since Fri 2020-05-15 15:50:03 BST; 1min 40s ago
Docs: man:nft(8)
http://wiki.nftables.org
Process: 1986 ExecStart=/usr/sbin/nft -f /etc/nftables.conf (code=exited, status=0/SUCCESS)
Main PID: 1986 (code=exited, status=0/SUCCESS)

May 15 15:50:03 raspberrypi systemd[1]: Starting nftables...
May 15 15:50:03 raspberrypi systemd[1]: Started nftables.
 
But with anything in .conf file service fails with errors:
/etc/nftables.conf:2:1-14: Error: Could not process rule: Operation not supported flush ruleset

Same for running nft in an interactive session

Linux raspberrypi 5.4.40-v7l+ #1316 SMP Tue May 12 13:10:42 BST 2020 armv7l GNU/Linux
pi4 4Gb Ram
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage4
version 21bfdeee3a6ea823e2113b983390acd1eec8edfb (clean) (release) (start)

pelwell, 2020-05-15:
They do indeed appear to be missing from the 32-bit bcm2711_defconfig and the 64-bit bcmrpi3_defconfig. The squashing that takes place on our new branches makes it hard to work out what went wrong, but I suspect there was a merge problem and it ended up half-and-half.

That's fixed in the rpi-5.4.y branch of the kernel source, and the modules will appear in future releases.

mys721tx, 2021-01-24:
nftables on aarch64 does not support sets at 08ae2dd with the following minimal nftables.conf.

```
#!/usr/bin/nft -f

define IP_DROPS = {
    172.16.0.0/24
}

table inet filter {
    chain input {
        type filter hook input priority 0; policy accept;

        ip saddr $IP_DROPS drop
    }
}

```

mys721tx, 2021-01-24:
Also the output from nft:

```ShellSession
$ sudo nft flush ruleset; sudo nft -f /etc/nftables.conf
/etc/nftables.conf:3:19-1: Error: Could not process rule: Operation not supported
define IP_DROPS = {
 
/etc/nftables.conf:3:19-1: Error: Could not process rule: No such file or directory
define IP_DROPS = {
 
/etc/nftables.conf:11:9-31: Error: Could not process rule: No such file or directory
        ip saddr $IP_DROPS drop
        ^^^^^^^^^^^^^^^^^^^^^^^
```

mys721tx, 2021-01-26:
The problem is resolved at 9007908.

popcornmix, 2021-01-26:
@steveharriss are you happy this issue is resolved?

pelwell, 2021-05-10:
I'm happy to add `NF_LOG_ARP=m` (and `NF_LOG_NETDEV=m` - another missing option in that category) - see ad26fd4.