raspberrypi/linux #3619: [Pi 4] Other machines cannot connect to wireless AP when ieee80211w=1

hvenev, 2020-05-18:
**Is this the right place for my bug report?**
It might be related to brcmfmac firmware. I am using the firmware from https://github.com/RPi-Distro/firmware-nonfree

**Describe the bug**
It looks like brcmfmac does not support ieee80211w/MFP in AP mode. Clients are unable to connect to the network because the RSN IE message during handshake is different from the one in the beacon frame.

Disabling ieee80211w fixes the issue.

If I run the same hostapd.conf on my laptop (iwlwifi/iwlmvm), my phone connects to it just fine.

**To reproduce**
hostapd.conf:
```
ctrl_interface=/var/run/hostapd
ctrl_interface_group=wheel

driver=nl80211

macaddr_acl=0
wmm_enabled=1
country_code=BG
ignore_broadcast_ssid=0
hw_mode=g
ieee80211n=1
ieee80211w=1
ieee80211d=1
channel=6

interface=wlan0

ssid=REDACTED
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=REDACTED
```

**Expected behaviour**
Clients can connect to the network.

**Actual behaviour**
Clients cannot connect to the network.

**System**
Device: Raspberry Pi 4 Model B Rev 1.2
Kernel: rpi-5.6.y from 5377449ffb6ffe9ebfee939d52b2611130fbb836
Firwmare version: May 11 2020 18:59:41, 21bfdeee3a6ea823e2113b983390acd1eec8edfb (clean) (release) (start)
OS: Fedora 32 aarch64
hostapd: hostapd-2.9-3.fc32.aarch64

**Logs**
No errors appear on the Pi.

Here is what happens on my laptop when it tries to connect:
```
wpa_supplicant: wlan: SME: Trying to authenticate with dc:a6:32:87:2a:8d (SSID='REDACTED' freq=2437 MHz)
kernel: wlan: authenticate with dc:a6:32:87:2a:8d
kernel: wlan: send auth to dc:a6:32:87:2a:8d (try 1/3)
wpa_supplicant: wlan: Trying to associate with dc:a6:32:87:2a:8d (SSID='REDACTED' freq=2437 MHz)
kernel: wlan: authenticated
kernel: wlan: associate with dc:a6:32:87:2a:8d (try 1/3)
kernel: wlan: RX AssocResp from dc:a6:32:87:2a:8d (capab=0x411 status=0 aid=1)
wpa_supplicant: wlan: Associated with dc:a6:32:87:2a:8d
wpa_supplicant: wlan: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
kernel: wlan: associated
wpa_supplicant: wlan: WPA: IE in 3/4 msg does not match with IE in Beacon/ProbeResp (src=dc:a6:32:87:2a:8d)
wpa_supplicant: WPA: RSN IE in Beacon/ProbeResp - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 0c 00
wpa_supplicant: WPA: RSN IE in 3/4 msg - hexdump(len=26): 30 18 01 00 00 0f ac 04 01 00 00 0f ac 04 02 00 00 0f ac 02 00 0f ac 08 0c 00
kernel: wlan: deauthenticating from dc:a6:32:87:2a:8d by local choice (Reason: 17=IE_DIFFERENT)
wpa_supplicant: wlan: CTRL-EVENT-DISCONNECTED bssid=dc:a6:32:87:2a:8d reason=17 locally_generated=1
```

pelwell, 2020-05-21:
Do you have reason to think this _should_ be supported?

hvenev, 2020-05-22:
> Do you have reason to think this _should_ be supported?

Yes. It's a basic security measure that makes it more difficult for script kiddies to disconnect me from my network.

Furthermore, the behaviour of silently creating a broken network is quite confusing.

hvenev, 2020-05-24:
If I set `wpa_key_mgmt=SAE WPA-PSK`, I also get a similarly broken network, even if `ieee80211w=0`.

Before you ask, SAE should be supported because it makes offline brute force attacks against the network password much more expensive.

Note that most often people who conduct offline brute force attacks start by capturing a handshake. The easiest way to capture a handshake is to cause an existing client to disconnect and reconnect. If `ieee80211w=0`, anyone can just disconnect any client and observe a handshake.

burnbabyburn, 2020-06-06:
+1
802.11w is a basic and old standard (2009) to protect against deauth attacks.
If WIFI-AP is a valid usecase for any RPI it should be supported.

Specially at times where deauth runs on very cheap ESP chips
https://github.com/SpacehuhnTech/esp8266_deauther

hvenev, 2020-06-06:
> Did u try this hint?
> [billz/raspap-webgui#484](https://github.com/billz/raspap-webgui/issues/484)

What exactly? Configuring the network as WPA3 Enterprise?

cybern0id, 2020-06-09:
This bug is not just related to RPi4 and/or AP mode.

I have a RPi 3B+ running stock Raspberry Pi OS (Raspbian) Lite installed today, with kernel 4.19.118-v7+; using its on-board wifi chip, it is unable to connect as a client to my Openwrt router.

The access point is set up with "WPA2-PSK/WPA3-SAE Mixed Mode" and with 802.11w Management Frame Protection set to "Optional".

It is dual band with Qualcomm Atheros QCA9880 802.11acn 5 GHz and Atheros AR9287 802.11bgn 2 GHz.

All my other devices can connect to the access point without issue (3 different Android phones, Sony PS4, Nintendo Switch, Lenovo X230 running Devuan or Windows 10, Lenovo X230T running either Windows 10 or Android x86, even a little ESP8266 ESP-01 module in my Sinclair Spectrum Next).

To troubleshoot, I used a process of elimination to determine that the RPi 3 will only connect when the access point is in WPA2-PSK only mode, with 802.11w Management Frame Protection disabled.

Setting the AP to either WPA2-PSK/WPA3-SAE Mixed Mode with 802.11w Management Frame Protection disabled OR to WPA2-PSK only mode with 802.11w Management Frame Protection set to either "Optional" or "Required" results in no connection.

Adding combinations of `pmf=1`, `proto=RSN` `key_mgmt=SAE` and `sae_password=#######` to the wpa_supplicant.conf config file did not have the desired effect.

The router logs show the following repeatedly when the RPi 3B tries unsuccessfully to connect:

`Tue Jun  9 04:15:02 2020 daemon.info hostapd: wlan1-1: STA b8:27:eb:xx:xx:xx IEEE 802.11: authenticated`
`Tue Jun  9 04:15:02 2020 daemon.info hostapd: wlan1-1: STA b8:27:eb:xx:xx:xx IEEE 802.11: associated (aid 1)`
`Tue Jun  9 04:15:02 2020 daemon.notice hostapd: wlan1-1: AP-STA-POSSIBLE-PSK-MISMATCH b8:27:eb:xx:xx:xx`
`Tue Jun  9 04:15:03 2020 daemon.notice hostapd: wlan1-1: AP-STA-POSSIBLE-PSK-MISMATCH b8:27:eb:xx:xx:xx`
`Tue Jun  9 04:15:04 2020 daemon.notice hostapd: wlan1-1: AP-STA-POSSIBLE-PSK-MISMATCH b8:27:eb:xx:xx:xx`
`Tue Jun  9 04:15:05 2020 daemon.notice hostapd: wlan1-1: AP-STA-POSSIBLE-PSK-MISMATCH b8:27:eb:xx:xx:xx`
`Tue Jun  9 04:15:11 2020 daemon.info hostapd: wlan1-1: STA b8:27:eb:xx:xx:xx IEEE 802.11: deauthenticated due to local deauth request`

[This article ](https://dokuwiki.alu4u.com/doku.php?id=stellar-wireless-wpa3-sae-raspberry-pi) suggests that both the kernel and wpa_supplicant must be patched to support WPA3-Personal (SAE/PMF), though I'm unsure as to whether this is still necessary as it is dated June 2019.

[This bug](https://github.com/billz/raspap-webgui/issues/484) also has a little related background info.

I agree with all the other comments in saying this is quite an important bug to resolve due to the security implications. It is [trivial and cheap](https://majzel.blogspot.com/2019/07/wireless-networking-ieee-80211w-and.html) to make a device that can prevent any clients joining a WPA2 wireless network that doesn't have 802.11w Management Frame Protection enabled.

pelwell, 2020-06-09:
There is a ticket open with Cypress asking about support for 802.11w and WPA3-SAE.

pelwell, 2020-06-10:
I can confirm that the current firmware release doesn't include MFP or WPA3/SAE support (unlike recent general public releases), but that we've requested an update.

pelwell, 2020-06-17:
There is a trial firmware to download here: https://drive.google.com/file/d/1sOsgqaTK1OiGyyhYNNeEo02XqCQPCPcx/view?usp=sharing

(This firmware may look familiar, @hvenev - Cypress have suggested it)

Make a backup of your old firmware and install the new one using:
```
$ sudo cp /lib/firmware/brcm/brcmfmac43455-sdio.bin{,.orig}
$ sudo cp brcmfmac43455-sdio-2.bin /lib/firmware/brcm/brcmfmac43455-sdio.bin
```
In case you encounter any problems you can enable firmware logging by adding `brcmfmac.debug=0x100000` to cmdline.txt. Diagnostic output from the firmware will appear in the output of `dmesg`.

Reverting to the original firmware is accomplished with:
```
$ sudo mv /lib/firmware/brcm/brcmfmac43455-sdio.bin{.orig,}
```

hvenev, 2020-06-17:
> There is a trial firmware to download here: https://drive.google.com/file/d/1sOsgqaTK1OiGyyhYNNeEo02XqCQPCPcx/view?usp=sharing

Yes, I tried this firmware quite a while ago. It is also broken.

I am still unable to create a network with MFP optional (broken network created) or required (hostapd fails to start) or SAE (error in the kernel log, broken network created). I tried to create a network with `wpa_key_mgmt=WPA-PSK-SHA256`. My phone thinks that the network is using WPA(2?) Enterprise.

I cannot connect to a network with `ieee80211=1` and `wpa_key_mgmt=SAE WPA-PSK`.

pelwell, 2020-06-17:
> Yes, I tried this firmware quite a while ago. It is also broken.

In which case I'll presumably be reporting that fact back to them fairly soon.

Re: custom kernel builds, bcm2711_defconfig (always a good place to check) has:
```
CONFIG_BRCMDBG=y
```

hvenev, 2020-06-17:
Log with `wpa_key_mgmt=WPA-PSK ieee80211w=1`:
```
[   48.621861] brcmfmac: F1 signature read @0x18000000=0x15264345
[   48.627404] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   48.791199] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   48.803183] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:20:01 version 7.45.206 (r725000 CY) FWID 01-febaba43
[   48.890434] brcmfmac mmc1:0001:1 wlan: renamed from wlan0
[   48.898820] brcmfmac: CONSOLE: d 0
[   48.898824] brcmfmac: CONSOLE: 000000.063 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.206 (r725000 CY)
[   48.898826] brcmfmac: CONSOLE: 000000.063 TCAM: 256 used: 251 exceed:0
[   48.898828] brcmfmac: CONSOLE: 000000.064 reclaim section 1: Returned 118440 bytes to the heap
[   48.898830] brcmfmac: CONSOLE: 000000.064 reclaim section 4: Returned 44 bytes to the heap
[   48.898831] brcmfmac: CONSOLE: 000000.064 sdpcmd_dpc: Enable
[   48.898833] brcmfmac: CONSOLE: 000000.084 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
[   48.898835] brcmfmac: CONSOLE: 000000.086 wl0: unable to find iovar "rsdb_mode"
[   48.898837] brcmfmac: CONSOLE: 000000.086 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
[   48.898839] brcmfmac: CONSOLE: 000000.086 wl0: unable to find iovar "tdls_enable"
[   48.898841] brcmfmac: CONSOLE: 000000.086 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
[   48.898842] brcmfmac: CONSOLE: 000000.102 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[   48.898844] brcmfmac: CONSOLE: 000000.162 wl0: wlc_iovar_op: bw_cap BCME -5 (Not down)
[   48.938789] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   48.958838] brcmfmac: CONSOLE: 000000.209 wl0: unable to find iovar "toe_ol"
[   48.958841] brcmfmac: CONSOLE: 000000.209 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
[   48.958844] brcmfmac: CONSOLE: 000000.210 wl0: wl_open
[   61.099371] local: port 1(wlan) entered blocking state
[   61.100463] local: port 1(wlan) entered disabled state
[   61.101672] device wlan entered promiscuous mode
[   61.128837] brcmfmac: CONSOLE: 000012.363 wl0: bc/mc deauth_reason on STA BSS?
[   61.128841] brcmfmac: CONSOLE: 000012.386 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[   61.148802] brcmfmac: CONSOLE: 000012.397 wl0: wl_open
[   61.162133] brcmfmac: CONSOLE: 000012.408 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[   61.226539] IPv6: ADDRCONF(NETDEV_CHANGE): wlan: link becomes ready
[   61.229436] local: port 1(wlan) entered blocking state
[   61.231735] local: port 1(wlan) entered forwarding state
[   61.234052] IPv6: ADDRCONF(NETDEV_CHANGE): local: link becomes ready
[   61.265566] brcmfmac: CONSOLE: 000012.481 wl0: link up (wl0)
[   61.265571] brcmfmac: CONSOLE: 000012.481 wl0: link up (wl0)
[  102.375324] brcmfmac: CONSOLE: 000053.432 
[  102.375332] brcmfmac: CONSOLE: No PMKID found in RSNIE
[  102.755319] brcmfmac: CONSOLE: 000053.812 
[  102.755325] brcmfmac: CONSOLE: No PMKID found in RSNIE
[  104.265314] brcmfmac: CONSOLE: 000055.301 
[  104.265319] brcmfmac: CONSOLE: No PMKID found in RSNIE
[  105.265314] brcmfmac: CONSOLE: 000056.294 
[  105.265321] brcmfmac: CONSOLE: No PMKID found in RSNIE
[  106.221971] brcmfmac: CONSOLE: 000057.245 
[  106.221977] brcmfmac: CONSOLE: No PMKID found in RSNIE
[  108.011961] brcmfmac: CONSOLE: 000059.026 
[  108.011969] brcmfmac: CONSOLE: No PMKID found in RSNIE
[  108.741971] brcmfmac: CONSOLE: 000059.774 
[  108.741976] brcmfmac: CONSOLE: No PMKID found in RSNIE
[  109.955311] brcmfmac: CONSOLE: 000060.963 
[  109.955317] brcmfmac: CONSOLE: No PMKID found in RSNIE
[  113.418622] brcmfmac: CONSOLE: 000064.427 
[  113.418628] brcmfmac: CONSOLE: No PMKID found in RSNIE
```

hvenev, 2020-06-17:
Log with `wpa_key_mgmt=WPA-PSK-SHA256 ieee80211w=2`:
```
[  344.179729] local: port 1(wlan) entered blocking state
[  344.180656] local: port 1(wlan) entered disabled state
[  344.181840] device wlan entered promiscuous mode
[  344.208485] brcmfmac: CONSOLE: 000294.428 wl0: bc/mc deauth_reason on STA BSS?
[  344.208489] brcmfmac: CONSOLE: 000294.452 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[  344.231870] brcmfmac: CONSOLE: 000294.463 wl0: wl_open
[  344.231874] brcmfmac: CONSOLE: 000294.474 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[  344.307723] local: port 1(wlan) entered blocking state
[  344.310055] local: port 1(wlan) entered forwarding state
[  344.314770] ieee80211 phy1: brcmf_cfg80211_add_key: set wsec error (-52)
[  344.331892] device wlan left promiscuous mode
[  344.334353] local: port 1(wlan) entered disabled state
[  344.365250] brcmfmac: CONSOLE: 000294.547 wl0: link up (wl0)
[  344.365259] brcmfmac: CONSOLE: 000294.547 wl0: link up (wl0)
[  344.365264] brcmfmac: CONSOLE: 000294.557 wl0: wlc_iovar_op: wsec BCME -2 (Bad Argument)
[  344.835324] brcmfmac: CONSOLE: 000295.023 wl0: wl_open
```

(hostapd fails to start)

Are there any other configurations you want me to test?

pelwell, 2020-06-17:
Thanks - I've passed both of those on. You could try the worldwide-safe, Pi4-only clm_blob (https://drive.google.com/file/d/1Qoc90FCTO17d69PbBqUhkJKgqDMmdOui/view?usp=sharing - you'll have to rename it to `brcmfmac43455-sdio.clm_blob`, but I think you know that), but I don't expect it to make a difference to this issue.

hvenev, 2020-06-17:
Other than making channels 12 and 13 unusable, the worldwide-safe clm_blob does not change anything.

graysky2, 2020-06-17:
Some of this is over my head, but I gather that @hvenev is using his RPi4 to be an access point.  In my case, I wish to get my RPi4 to connect to an OpenWRT router/access point which is pure WPA3-SAE.  Does this issue apply to me?  See the discussion on [this] hostap mailing post(http://lists.infradead.org/pipermail/hostap/2020-June/041839.html).

cybern0id, 2020-06-17:
@graysky2 read my post above. Connecting to WPA3-SAE or WPA2 with MFP enabled or set to optional does not work (and neither does mixed-mode).

In other news; I performed the steps, detailed in the link I mention in my previous post, to patch and build the 64 bit Raspberry Pi 4 kernel using the most recent public release of the driver from Cypress and WPA3/SAE/MFP still does not work. I also tried a more recent version of wpa_supplicant by compiling it from Debian testing sources as well as updating the Broadcom wifi firmware using the trial version (dated May 20 2020).

Clearly there is more to this issue than simply patching the current Broadcom driver in the kernel.

graysky2, 2020-06-18:
@cybern0id - Thanks for the reply.  I assume you tried the [cypress drivers from 04-02-2020](https://community.cypress.com/docs/DOC-19375)?

pelwell, 2020-06-18:
There's another test firmware - the date is the same as the 2020-04-02 release but the options list looks different (I have no other information): https://drive.google.com/file/d/10ivocg5PrOwVxAYFKOzJEdv_gdCd-IUF/view?usp=sharing

Be sure to rename to brcmfmac43455-sdio.bin when installing it.

Unfortunately it sounds as though we may need to apply Cypress patches to the kernel, hostap and wpa_supplicant to support WPA3 - I'll post more as I know it.

graysky2, 2020-06-18:
@pelwell - The kernel patches apply cleanly, but the build fails.  Any thoguhts?

```
...
  CC [M]  net/sched/cls_basic.o
  CC [M]  drivers/net/wireless/broadcom/b43legacy/debugfs.o
  CC [M]  drivers/media/rc/keymaps/rc-trekstor.o
In file included from drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c:31:
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:6: warning: "LINUX_VERSION_CODE" is not defined, evaluates to 0 [-Wundef]
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |      ^~~~~~~~~~~~~~~~~~
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:27: warning: "KERNEL_VERSION" is not defined, evaluates to 0 [-Wundef]
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |                           ^~~~~~~~~~~~~~
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:41: error: missing binary operator before token "("
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |                                         ^
  LD [M]  drivers/media/usb/au0828/au0828.o
  AR      drivers/media/usb/b2c2/built-in.a
  CC [M]  drivers/media/usb/b2c2/flexcop-usb.o
  CC [M]  drivers/media/rc/keymaps/rc-tt-1500.o
  LD [M]  drivers/net/wireless/broadcom/b43/b43.o
  CC [M]  drivers/media/rc/keymaps/rc-twinhan-dtv-cab-ci.o
  CC [M]  net/netfilter/xt_cluster.o
  CC [M]  net/netfilter/xt_comment.o
In file included from drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c:31:
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:6: warning: "LINUX_VERSION_CODE" is not defined, evaluates to 0 [-Wundef]
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |      ^~~~~~~~~~~~~~~~~~
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:27: warning: "KERNEL_VERSION" is not defined, evaluates to 0 [-Wundef]
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |                           ^~~~~~~~~~~~~~
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:41: error: missing binary operator before token "("
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |                                         ^

...
```

pelwell, 2020-06-18:
Does the patched file end up including "<linux/version.h>"? Other files that use those macros do, and in my build tree there is `include/generated/uapi/linux/version.h` which contains:
```
#define LINUX_VERSION_CODE 328751
#define KERNEL_VERSION(a,b,c) (((a) << 16) + ((b) << 8) + (c))
```
Try adding `#include <linux/version.h>` in cfg80211.c (or in one of the headers it includes.

graysky2, 2020-06-18:
If I look in the build tree, I do indeed have `include/generated/uapi/linux/version.h` as you do after applying all those patches:

```
% cat include/generated/uapi/linux/version.h
#define LINUX_VERSION_CODE 328749
#define KERNEL_VERSION(a,b,c) (((a) << 16) + ((b) << 8) + (c))
```

Trying to build now applying: https://github.com/graysky2/PKGBUILDs/commit/1b5e56ec1bf9f5a01bb237047137632b21114188

pelwell, 2020-06-18:
Such an aggressive patch name - I hope it does what it says.

graysky2, 2020-06-18:
> Such an aggressive patch name - I hope it does what it says.

:D Yes, I am having a bad day!  I will update (build in progress).

graysky2, 2020-06-18:
It failed:
```
  CC [M]  drivers/net/wireless/broadcom/b43/radio_2056.o
  CC      fs/xfs/xfs_itable.o
In file included from drivers/net/wireless/broadcom/brcm80211/brcmfmac/fwil.c:15:
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:6: warning: "LINUX_VERSION_CODE" is not defined, evaluates to 0 [-Wundef]
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |      ^~~~~~~~~~~~~~~~~~
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:27: warning: "KERNEL_VERSION" is not defined, evaluates to 0 [-Wundef]
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |                           ^~~~~~~~~~~~~~
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:41: error: missing binary operator before token "("
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |                                         ^
distcc[10778] ERROR: compile drivers/net/wireless/broadcom/brcm80211/brcmfmac/fwil.c on 10.9.8.228:3635 failed
distcc[10778] (dcc_build_somewhere) Warning: remote compilation of 'drivers/net/wireless/broadcom/brcm80211/brcmfmac/fwil.c' failed, retrying locally
distcc[10778] Warning: failed to distribute drivers/net/wireless/broadcom/brcm80211/brcmfmac/fwil.c to 10.9.8.228:3635, running locally instead
  CC [M]  drivers/media/rc/keymaps/rc-gotview7135.o
drivers/net/wireless/broadcom/b43/tables_nphy.c: In function 'b43_nphy_get_tx_gain_table':
drivers/net/wireless/broadcom/b43/tables_nphy.c:3718:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
drivers/net/wireless/broadcom/b43/tables_nphy.c:3721:3: note: here
  CC [M]  drivers/media/rc/keymaps/rc-hisi-poplar.o
  CC [M]  drivers/media/rc/keymaps/rc-hisi-tv-demo.o
drivers/net/wimax/i2400m/usb.c: In function '__i2400mu_send_barker':
drivers/net/wimax/i2400m/usb.c:187:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
drivers/net/wimax/i2400m/usb.c:199:2: note: here
  LD [M]  drivers/net/wimax/i2400m/i2400m-usb.o
In file included from drivers/net/wireless/broadcom/brcm80211/brcmfmac/fwil.c:15:
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:6: warning: "LINUX_VERSION_CODE" is not defined, evaluates to 0 [-Wundef]
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |      ^~~~~~~~~~~~~~~~~~
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:27: warning: "KERNEL_VERSION" is not defined, evaluates to 0 [-Wundef]
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |                           ^~~~~~~~~~~~~~
drivers/net/wireless/broadcom/brcm80211/brcmfmac/bus.h:155:41: error: missing binary operator before token "("
  155 | #if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 16, 0))
      |                                         ^
make[6]: *** [scripts/Makefile.build:265: drivers/net/wireless/broadcom/brcm80211/brcmfmac/fwil.o] Error 1
make[5]: *** [scripts/Makefile.build:500: drivers/net/wireless/broadcom/brcm80211/brcmfmac] Error 2
make[4]: *** [scripts/Makefile.build:500: drivers/net/wireless/broadcom/brcm80211] Error 2
make[4]: *** Waiting for unfinished jobs....
```

pelwell, 2020-06-18:
Add it to core.h instead - that gets included all over the place.

graysky2, 2020-06-18:
Which `core.h`?  
```
% find . -type f -name 'core.c'|grep wireless
./net/wireless/core.c
./drivers/net/wireless/realtek/rtlwifi/core.c
./drivers/net/wireless/quantenna/qtnfmac/core.c
./drivers/net/wireless/mediatek/mt7601u/core.c
./drivers/net/wireless/mediatek/mt76/mt7603/core.c
./drivers/net/wireless/broadcom/brcm80211/brcmfmac/core.c
./drivers/net/wireless/ath/ath6kl/core.c
./drivers/net/wireless/ath/ath10k/core.c
```

hvenev, 2020-06-18:
I would guess `drivers/net/wireless/broadcom/brcm80211/brcmfmac/core.c`.

hvenev, 2020-06-18:
However, it looks like `bus.h` does not include `core.h`. You can just add it to every file under `drivers/net/wireless/broadcom/brcm80211/brcmfmac` that mentions `LINUX_VERSION_CODE`.

graysky2, 2020-06-18:
https://github.com/graysky2/PKGBUILDs/commit/9255e066d5e70586f65965a4f2583fb22b095638

graysky2, 2020-06-18:
@pelwell - Note there are some other steps outlined in the README but I am finding them a bit confusing wrt firmware manipulations.

> Unfortunately it sounds as though we may need to apply Cypress patches to the kernel, hostap and wpa_supplicant to support WPA3 - I'll post more as I know it.

* Patched kernel (https://github.com/graysky2/PKGBUILDs/commit/cd5a7668c3ca4646c7599da7cd37c676116ae6d0)
* Patched wpa_supplicant
* Modified firmware linked by @pelwell (https://github.com/raspberrypi/linux/issues/3619#issuecomment-645857826)

Still no connection:
```
# wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/blast.conf
Successfully initialized wpa_supplicant
wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=US
wlan0: Trying to associate with SSID 'blast'
wlan0: WPA: Failed to select authenticated key management type
wlan0: WPA: Failed to set WPA key management and encryption suites
```

Config I using:
```
# cat /etc/wpa_supplicant/blast.conf 
ctrl_interface=/run/wpa_supplicant
country=US

network={
 ssid="blast"
 sae_password="mycoolpw"
 key_mgmt=SAE
 ieee80211w=2
}
```

Note that in addition to the 3 above, I also build that backport modules steps and installed them to `/usr/lib/modules/extramodules-5.4-raspberrypi4/updates/` then run `depmod -a` followed by a reboot.

That seemingly broke wpa_supplicant:
```
# wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/blast.conf
Successfully initialized wpa_supplicant
wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=US
wlan0: Failed to initiate sched scan
...
wlan0: Failed to initiate sched scan
```

graysky2, 2020-06-19:
> * Patched wpa_supplicant
> * Modified firmware linked by @pelwell (#3619 (comment))
> * Patched kernel (graysky2/PKGBUILDs@cd5a766)

If anyone is running Arch ARM and would like to review the builds for these:
https://github.com/graysky2/PKGBUILDs/tree/linux-rpi4-cypress-patched/testing/wpa_supplicant
https://github.com/graysky2/PKGBUILDs/tree/linux-rpi4-cypress-patched/testing/linux-firmware
https://github.com/graysky2/PKGBUILDs/tree/linux-rpi4-cypress-patched/core/linux-raspberrypi4

Alternatively, [here](http://repo-ck.com/testing/) are the compiled.packages.

cybern0id, 2020-07-01:
Testing this again today:

Cross-compiled 64 bit kernel 5.4.49-v8+, pulled from this repo master, patched with most recent public release from Cypress dated [2020-06-25](https://community.cypress.com/docs/DOC-20044).

Most of the patches apply cleanly bar one - I had to fiddle with the sdio.c file so that the patch that changes the definition for the new firmware filenames (now located in /lib/firmware/cypress) applies properly.

Installed the kernel, modules and firmware (plus copying/renaming the brcmfmac43455-sdio.txt nvram file to the new location /lib/firmware/cypress/cyfmac43455-sdio.txt) to up-to-date Raspberry Pi OS 64 bit with most recent RPi firmware.

Even though the README file supplied with the public release of the Cypress fmac drivers states explicitly that the 43455 driver supports both WPA3-STA and WPA3-AP, it still does not work. However, I have not yet patched and recompiled hostap/wpa_supplicant which will be my next step.

As a comparison, I decided to try connecting using an Atheros based USB Wifi adapter, a TP-Link TL-WN722N, which uses the ath9k_htc driver and firmware. I used the same RPi 4 B+, connecting to the same Access Point, using kernel 5.4.42-v8+ (ath9k_htc wouldn't work with my compiled 5.4.49-v8+ for some reason), using the default wpasupplicant 2:2.7+git20190128+0c1e29f-6+deb10u2 .deb and it connects without issue.

My wpa_supplicant.conf:

`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev`
`update_config=1`
`country=GB`

`network={`
`	ssid="accesspoint"`
`	sae_password="password"`
`	key_mgmt=SAE`
`	ieee80211w=2`
`}`

Output from wpa_cli status when successfully connected using Atheros Wifi adapter:

`> status`
`bssid=##:##:##:##:##:##`
`freq=2412`
`ssid=accesspoint`
`id=0`
`mode=station`
`pairwise_cipher=CCMP`
`group_cipher=CCMP`
`key_mgmt=SAE`
`pmf=1`
`mgmt_group_cipher=BIP`
`sae_group=19`
`wpa_state=COMPLETED`
`ip_address=192.168.#.#`
`p2p_device_address=##:##:##:##:##:##`
`address=##:##:##:##:##:##`
`uuid=######################`

Has there been any feedback from Cypress as yet?

cybern0id, 2020-07-01:
And now I'm confused..... I've just tried WPA-PSK-SHA256 and ieee80211w=2 using RPi 4B+ onboard brcmfmac wifi and connecting to the same mixed-mode WPA2+WPA3-SAE access point works and management frame protection is enabled:

RPi firmware:
`:~$sudo rpi-eeprom-update`
`BCM2711 detected`
`VL805 firmware in bootloader EEPROM`
`BOOTLOADER: up-to-date`
`CURRENT: Thu 16 Apr 17:11:26 UTC 2020 (1587057086)`
` LATEST: Thu 16 Apr 17:11:26 UTC 2020 (1587057086)`
 `FW DIR: /lib/firmware/raspberrypi/bootloader/critical`
`VL805: up-to-date`
`CURRENT: 000137ad`
` LATEST: 000137ad`

BRCM43455 firmware:
`:~$dmesg | grep brcmfmac | grep Firmware`
`[    7.964360] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May 20 2020 00:36:51 version manifest (de496ab CY) FWID 01-12aa6092`

Kernel:
`:~$uname -r`
`5.4.42-v8+`

wpa_supplicant.conf:
`:~ $ cat /etc/wpa_supplicant/wpa_supplicant.conf`
`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev`
`update_config=1`
`country=GB`

`network={`
`	ssid="accesspoint"`
`	psk="password"`
`	key_mgmt=WPA-PSK-SHA256`
`	ieee80211w=2`
`}`

wpa_cli status:
`> status`
`bssid=##:##:##:##:##:##`
`freq=5180`
`ssid=accesspoint`
`id=0`
`mode=station`
`pairwise_cipher=CCMP`
`group_cipher=CCMP`
`key_mgmt=WPA2-PSK-SHA256`
`pmf=1`
`mgmt_group_cipher=BIP`
`wpa_state=COMPLETED`
`ip_address=192.168.#.#`
`p2p_device_address=##:##:##:##:##:##`
`address=##:##:##:##:##:##`
`uuid=########################`
`ieee80211ac=1`

It still doesn't work with pure WPA3-SAE though however, having MFP/PMF enabled on WPA2 is a win from my perspective!

cybern0id, 2020-07-01:
Well, I've patched wpa_supplicant v2.9 (the bare original source) using the Cypress 2020-06-25 release and ran it on my brcmfmac patched kernel 5.4.49-v8+ and connecting to WPA3-SAE access point with MFP enabled **actually works!** Now to try running the Pi in AP mode.

BRCM43455 firmware:
`:~ $dmesg | grep brcmfmac | grep Firmware`
`[    5.670602] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May 22 2020 21:24:34 version 7.45.214 (9c83742 CY) FWID 01-59feefd4`

Kernel:
`:~$uname -r`
`5.4.49-v8-brcmfmac+`

wpa_supplicant.conf:
`:~ $ cat /etc/wpa_supplicant/wpa_supplicant.conf `
`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev`
`update_config=1`
`country=GB`

`network={`
`	ssid="accesspoint"`
`	sae_password="password"`
`	key_mgmt=SAE`
`	ieee80211w=2`
`}`

wpa_cli status:
`> status`
`bssid=##:##:##:##:##:##`
`freq=5180`
`ssid=accesspoint`
`id=0`
`mode=station`
`pairwise_cipher=CCMP`
`group_cipher=CCMP`
`key_mgmt=SAE`
`pmf=1`
`mgmt_group_cipher=BIP`
`wpa_state=COMPLETED`
`ip_address=192.168.#.#`
`p2p_device_address=##:##:##:##:##:##`
`address=##:##:##:##:##:##`
`uuid=#######################`
`ieee80211ac=1`

cybern0id, 2020-07-01:
AP mode also works! After patching hostapd I had to add CONFIG_SAE=y to the end of the .config file. I also enabled CONFIG_IEEE80211N=y and CONFIG_IEEE80211AC=y in the .config file.

The kernel and brcm43455 firmware are the same as in my previous comment https://github.com/raspberrypi/linux/issues/3619#issuecomment-652157368, my hostapd.conf as follows:

`:~ $ cat /etc/hostapd/hostapd.conf `
`ctrl_interface=/var/run/hostapd`
`ctrl_interface_group=netdev`

`driver=nl80211`

`macaddr_acl=0`
`wmm_enabled=1`
`country_code=GB`
`ignore_broadcast_ssid=0`
`#hw_mode=g`
`hw_mode=a`
`ieee80211n=1`
`ieee80211w=2`
`ieee80211d=1`
`ieee80211ac=1`
`#channel=7`
`channel=36`

`interface=wlan0`

`ssid=testAPmode`
`wpa=2`
`#wpa_key_mgmt=WPA-PSK-SHA256`
`wpa_key_mgmt=SAE`
`rsn_pairwise=CCMP`
`#wpa_passphrase=testAPmode`
`sae_password=testAPmode`
`#sae_require_mfp=1`

Both SAE and WPA-PSK-SHA256 key management work, as do 2.4 GHz and 5 GHz modes. ieee80211w=1 optional MFP also works.

graysky2, 2020-07-01:
Nice to hear the new files are working.
@pelwell - what is the road to getting these included in the firmware package?

prateek-0, 2020-08-11:
I have the same issue with the mismatched RSN IE even with `ieee80211w=0`, when trying to use a Raspberry Pi as an AP: . It affects EAP-TLS but not WPA[2]-PSK though: https://superuser.com/questions/1576743/using-raspberry-pi-as-wifi-ap-wpa2-psk-works-but-eap-tls-results-in-ie-in-3

rbaoc, 2020-08-24:
> AP mode also works! After patching hostapd I had to add CONFIG_SAE=y to the end of the .config file. I also enabled CONFIG_IEEE80211N=y and CONFIG_IEEE80211AC=y in the .config file.

Hi @cybern0id, I am a new to Raspberry Pi and trying to setup an AP with WPA3. Could u tell me, where exactly is the .config file and what can I do with it?

popcornmix, 2020-08-25:
The .config file is used to build the kernel. https://www.raspberrypi.org/documentation/linux/kernel/building.md

rbaoc, 2020-08-25:
> The .config file is used to build the kernel. https://www.raspberrypi.org/documentation/linux/kernel/building.md

Thank you for reply xD. So,  it is not the .config file of hostapd, is it? 
As far as I understand, I need to add these CONFIG_xxx configurations to the end of the .config file, which is inside /linux? And then, could u instruct me, how can I apply these changes?

pelwell, 2020-08-25:
You need to read the kernel building guide that @popcornmix linked to. The [Kernel Configuration guide](https://www.raspberrypi.org/documentation/linux/kernel/configuring.md) may help to explain how the .config file works.

hvenev, 2020-08-25:
From my understanding, `CONFIG_SAE` and `CONFIG_IEEE80211AC` are options you need to set in hostapd's `.config`.

pelwell, 2020-08-25:
Indeed, those config settings don't appear to be used by the kernel - hostapd just uses the same naming and format as the kernel, which is either consistent or confusing depending on your point of view.

rbaoc, 2020-08-25:
Yes, I tried to edit ,config file from hostapd-2.9 and AP doesnot work

hvenev, 2020-08-25:
Did you apply the kernel and hostapd patches provided by Cypress?

rbaoc, 2020-08-25:
Ach my bad, I have downloaded the hostapd from w1.fi (idk why, but yesterday I cant access to Cypress website @@) and havent applied the kernel yet. Just wanna try to only edit hostapd file. I will applied these aboved changes and update later xD

rbaoc, 2020-08-27:
Hi guys, that's me again. I would like to ask your help one more time. After applied patches from @graysky2, install the kernel and make modules from Cypress patch, I stucked at updating firmware file in /lib/firmware/cypress.
`cannot stat '/lib/firmware/cypress': No such file or directory`
Do someone has any suggestion for me?

hvenev, 2020-08-28:
`mkdir /lib/firmware/cypress` maybe?

Also check if you need to put the firmware in `/lib/firmware/cypress`
or `/lib/firmware/brcm`.

On Thu, 2020-08-27 at 15:11 -0700, rbaoc wrote:
> Hi guys, that's me again. I would like to ask your help one more
> time. After applied patches from @graysky2, install the kernel and
> make modules from Cypress patch, I stucked at updating firmware [...]

rbaoc, 2020-08-28:
Thank you for reply. I follow the README of Cypress's patch:
`### Using Linux Stable v5.4.18
```bash
#1. Download Linux stable kernel source
    wget https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/snapshot/linux-5.4.18.tar.gz
    tar zxvf linux-5.4.18.tar.gz
#2. In Linux root folder, untar/apply cypress patches with below bash commands
    cd linux-5.4.18
    tar zxvf cypress-patch*.tar.gz
    for i in cypress-patch/*.patch; do patch -p1 < $i; done
#3. Set kernel .config and enable below options, then compile kernel image
#      CONFIG_BRCMUTIL=y
#      CONFIG_BRCMFMAC=y
#      CONFIG_BRCMFMAC_SDIO=y
#      CONFIG_BRCMFMAC_PROTO_BCDC=y
#      CONFIG_BRCMFMAC_PCIE=y
#      CONFIG_BRCMFMAC_PROTO_MSGBUF=y
#4. (optional) Backup original firmware files
    **cp /lib/firmware/cypress /lib/firmware/cypress-bak -r**
#5. Update firmware files in /lib/firmware/cypress
    tar zxvf cypress-firmware*.tar.gz
    **cp firmware/* /lib/firmware/cypress**
#6. Boot the system with the new kernel image`

Actually, I even dont have `/lib/firmware`. Just not sure, is this situation all the same with everyone? Or may any wrong installation has been done?

nicklowe, 2020-11-01:
Please may we get 802.11w and SAE both working 'out of the box' in Raspberry Pi OS?

MASHtm, 2021-01-08:
It would be great if at least WPA2 still works if WPA2/WPA3 is active on AP. This doesn't work as long as Raspi fails with optional 11w.

jottr, 2021-03-03:
> Nice to hear the new files are working.
> @pelwell - what is the road to getting these included in the firmware package?

So, what is required to get this working out of the box? 
What's the process?

Tehvan, 2021-03-12:
If this isn't going to be patched there needs to be a note in the config file. I was running WPA2 and WPA3 in mixed mode on OpenWRT and my PI 4B was not able to connect. I had to change it to WPA2 only and then disable 802.11w for the PI to connect.

jorgejones7711, 2021-03-19:
I also cannot connect to a wpa3-sae network with a rpi 4. All of my other devices can. Raspberry Pi is falling behind it seems.

jaccc, 2021-05-26:
It seems kernel 5.10 should support WPA3

brcmfmac: support SAE authentication offload in AP mode
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v5.10.40&id=d5f59c964e15e9e2cffaf8b0e9034616568717f9

mweinelt, 2021-08-07:
>Linux rpi 5.13.8 #1-NixOS SMP Wed Aug 4 10:47:57 UTC 2021 aarch64 GNU/Linux

> [   15.464751] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
> [   15.470890] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e

Unable to connect to my SAE home network. IWD just replies with "Operation unsupported" for SAE/MFP networks.

jottr, 2021-10-20:
There is an interesting pointer in an issue that references this one: https://github.com/raspberrypi/linux/issues/4093#issuecomment-855412224

According to @graysky2  it seems that connecting via [`iwd`](https://iwd.wiki.kernel.org/) to WPA3 networks works.

k-tatgenhorst, 2022-01-16:
This issue was opened in May of 2020 and my current up to date R-Pi 4 is unable to connect to 802.11w networks. This is a standard network type in response to a real world problem (deauth attacks), is there an an ETA for resolution?

Thank you

pelwell, 2022-01-21:
There is an open issue with Infineon about this. Although the firmware's build information appears to show support for Management Frame Protection (the more descriptive name for the feature known as 802.11w), when the brcmfmac driver queries the firmware's capabilities it claims not to support it.

[23 more replies]