raspberrypi/linux #3624: kea-admin error directly after installing, packages up to date

DirkLaurenz, 2020-05-19:
Raspbian Version: 10

Steps to reproduce:
apt-get install kea-admin
root@dc01:~# kea-admin
/usr/sbin/kea-admin: 39: .: Can't open /build/isc-kea-NiDGuv/isc-kea-1.4.0.P1/obj-arm-linux-gnueabihf/../src/bin/admin/admin-utils.sh

pelwell, 2020-05-19:
The kea-admin script is looking for /usr/share/kea/scripts/admin-utils.sh (and other things under /usr/share/kea), but the Debian package (which I doubt we've altered) is installed to /usr/share/kea-admin.

This isn't a kernel issue, so this isn't the right place to report it. @XECDesign Is this possibly a Raspbian-specific problem, or should this go straight to Debian?

DirkLaurenz, 2020-05-19:
Hello,

 

i’m sorry, thougth this is the raspbian bug report system.

Where should i post it?

XECDesign, 2020-05-20:
Raspbian doesn't seem to clear packages that have been removed from a given Debian release.

kea-admin is not meant to be available Buster. https://packages.debian.org/search?keywords=kea-admin

If you would like to report it, the place to do it would be https://bugs.launchpad.net/raspbian/

pelwell, 2020-05-20:
Thanks, @XECDesign.

closed 2020-05-20 by pelwell