raspberrypi/linux #3625: Kernel 5.4.y  and patch realtime

antonellocaroli, 2020-05-20:
Applied patch-rt patch-5.4.40-rt24 regularly.
Compilation is successful on aarch64.

But at boot it hangs here:

[image]

does anyone have any solutions?

schnitzeltony, 2020-05-20:
Smells like a port of [1-2] is required.

[1] https://github.com/schnitzeltony/meta-raspi-light/blob/master/recipes-kernel/linux/files/rt/0001-usb-dwc_otg-fix-system-lockup-when-interrupts-are-th.patch
[2] https://github.com/schnitzeltony/meta-raspi-light/blob/master/recipes-kernel/linux/files/rt/0002-usb-dwc_otg-fix-system-lockup-when-interrupts-are-th.patch

antonellocaroli, 2020-05-20:
Thank you.

I tried, but it stops at the same place.

reboot and other errors.

[image]

antonellocaroli, 2020-07-23:
did anyone happen to come up with any solutions?

I also tried with kernel 5.4.51 and patch-5.4.52-rt31.patch.xz 

the boot always stops here:

[image]

antonellocaroli, 2020-07-23:
to specify on raspberry3 aarch64 is not working.

on raspberry4 aarch64 it works instead.

poerob, 2020-09-24:
Hello!
I'm also experiencing boot issues. I use an Raspberry Pi 4B 4Gb with 64 Bit Raspberry Pi OS. I patched the current branch with patch-5.4.54-rt32.patch.gz, the patch was applied without any problems. Now, the system freezes with a black screen after the "Raspberry Pi OS Welcome Screen" comes up. Are there any solutions or workarounds for this?

pelwell, 2020-09-24:
I'm happy for you to use this issue to discuss problems with the realtime patches, and if you get to a stage where there is a combination of patches that work then I'll consider hosting them in an rpi-5.4.y-rt branch, just don't expect much support from the Pi devs.

poerob, 2020-09-24:
> I'm happy for you to use this issue to discuss problems with the realtime patches, and if you get to a stage where there is a combination of patches that work then I'll consider hosting them in an rpi-5.4.y-rt branch, just don't expect much support from the Pi devs.

Hi @pelwell, no problem, i'm just curious if somebody of the previous participants may have found a solution for this.

poerob, 2020-09-25:
Some News: I tried a precompiled 64bit 5.4 kernel i found [here](https://forum.linuxcnc.org/9-installing-linuxcnc/39779-rpi4-raspbian-64-bit-linuxcnc), which was patched with [5.4.54-rt32.patch.gz](https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/5.4/older/patch-5.4.54-rt32.patch.gz). Now everything just works fine. Something in my toolchain seems to be faulty, so the problem is on my side.

antonellocaroli, 2020-09-25:
@poerob  interesting, on rpi3 or rpi4?

poerob, 2020-09-25:
> @poerob interesting, on rpi3 or rpi4?

Pi 4B 4GB

antonellocaroli, 2020-09-25:
ok, on rpi4 my compilation had also worked...My problem is on rpi-3b

poerob, 2020-09-25:
Did you follow all steps from the Documentation for [Kernel Building ](https://www.raspberrypi.org/documentation/linux/kernel/building.md)? Otherwise i can't help you, since i have no Modell 3B+ at my place. Sorry!

antonellocaroli, 2020-09-26:
Just tried again with kernel 5.4.65 and patch-5.4.66-rt38.patch.gz

on aarch64.

Rpi4 no problem and it works (as I checked before)
Rpi3 same problem as above

kdoren, 2020-12-17:
I built a Raspberry Pi realtime Kernel 5.4.81-rt45.
sources are here:  https://github.com/kdoren/linux
also a pre-built 64-bit archive:  https://github.com/kdoren/linux/releases

FYI the old 4.19 kernel doesn't work on newer Pi4's, USB is broken.

antonellocaroli, 2020-12-17:
@kdoren 

does it also work on Rpi3?

kdoren, 2020-12-19:
I just tried it (64-bit rt kernel) on a Raspberry Pi 3B and it seems to work fine.

antonellocaroli, 2020-12-26:
@kdoren 
Sorry for the trivial question, but has the kernel been compiled with the **Fully Preemptible Kernel (RT)** option?
seeing the cpu usage, it seems not...

kdoren, 2020-12-27:
@antonellocaroli
The config options are here: https://github.com/kdoren/linux/blob/rpi-5.4.81-rt/rt64.config
including:
`CONFIG_PREEMPT_RT=y`

kdoren, 2020-12-28:
@antonellocaroli
You were correct, thanks for reporting.
Somehow I managed to build it with PREEMPT but not RT enabled.
I rebuilt it and posted a new file.  This one is really PREEMPT_RT:
```
$ uname -a
Linux jambox 5.4.81-rt45-v8+ #9 SMP PREEMPT_RT Mon Dec 28 00:13:29 PST 2020 aarch64 GNU/Linux
```
The bad news is that now it doesn't boot on a Pi3, same problem you ran into.
To deal with that I'm putting a "[pi4]" filter in /boot/config.txt so that it only boots the 5.4 rt kernel on a Pi4.
Sorry about the false alarm.

Ashram56, 2021-02-16:
@kdoren,

To clarify, has anyone been able to patch the 32b kernel for RT patches on RPI ?

kdoren, 2021-02-16:
@Ashram56 I released only a 64-bit build because I believe it's all that is needed - it works fine under 32-bit OS, arguably better.  It's running on hundreds of RPi4's with 32-bit OS.  I expect that 32-bit build of same code would work the same.  Which is to say, will work on Pi4 but fail to boot on Pi3.

Ashram56, 2021-02-17:
Thanks for the clarification
So for for the time being we have a solution for RPI4, but not RPI3

Regards

guysoft, 2021-03-16:
@kdoren Is there a point to push this to Rpi kernel branch? Or build and release RealtimePi with this? (Also your music distro looks cool)

kdoren, 2021-03-16:
@guysoft The rpi maintainers have a kernel merging process that I don't follow (and don't know), so I don't think that's an option.  The way I've been doing it is easy, and it works (on pi4 anyway, many hundreds of rpi's running my kernel builds, working well).

kdoren, 2021-03-18:
FYI I have built and released a 32-bit build of PREEMPT_RT kernel 5.10.16.  It works quite well on RPi4.  It boots on RPi3 but it will hang on rt_throttling fairly quickly if I.e. you start jack.

https://github.com/kdoren/linux

pelwell, 2021-03-18:
FYI, this is the email I sent to the original rpi-*-rt branch maintainer explaining how I thought it should work:

> I think the -rt variant should be a separate branch - unless the -rt branch goes upstream into mainline Linux I doubt we would switch completely.

> One of our internal rules is that, apart from back-porting patches to older kernels, we never change the upstream commits. This means that our development looks like this:

```
L-L-L-L-L-L-L---L-L-L-L-L-L-L     = Upstream Linux
             \       \       \
              P-P-P-P-M-P-P---M-P = Downstream rpi- branch
                  ^   ^       ^
                  A   B       C
```

> where L is an upstream Linux commit, P is a downstream Pi commit, and M is a merge commit. A represents the point where we stop rebasing our commits against the new kernel version, which corresponds to the point where it becomes our mainline branch. B and C are minor kernel version updates.

> Perhaps you could structure the -rt branch like this:

```
L-L-L-L-L-L-L----L-L-L-L-L-L-L     = Upstream Linux
            \         \       \
             R-R-R-R-M-M--M----M-M = rpi-rt branch
                    /    /     /
             P-P-P-P--P-P-----P    = Downstream rpi- branch
```

> I hope you can understand my diagram. The important point is that the L and P commits in this diagram are identical to those in the previous one, with the same commit hash. The nice thing about this scheme is that by not rebasing your existing commits you only need to think about (and fix) new commits.

Advantages of this scheme are that kernel updates are just minor increments on the previous one, which reduces the git churn a lot, and that commit IDs are shared with both the upstream kernel and the regular rpi- branches.

gtrainavicius, 2021-05-03:
Hi, has anyone got the realtime kernel based on 5.10 branch working on RPi 2 & 3?

guysoft, 2021-05-03:
@gtrainavicius There are talks here:
#4082

kdoren, 2021-05-04:
The rpi-4.19.y-rt branch, which works on RPi3, was built built by applying some additional patches on top of the "official" PREEMPT_RT patch set:   https://github.com/raspberrypi/linux/commits/rpi-4.19.y-rt

I tried these additional patches on 5.10 (the patches appear to apply correctly and cover all the intended cases).  It boots on RPi3 but is not stable.  So something more is needed but it's not clear what.

On RPi4 I have been using 5.10 with only the PREEMPT_RT patch set. (https://github.com/kdoren/linux/releases) This has worked well in most cases I care about (audio apps with jack & USB audio, lots of people successfully using it).

However there is at least one issue with this approach on Pi4: Jamulus server with multithreading enabled [no jack audio] can cause one of the cores running lxpanel to go to 100% CPU,  This does not happen on low-latency PREEMPT kernel build.  I don't know if this is a general problem with PREEMPT_RT, or is specific to my build for RPi4.

guysoft, 2021-05-04:
Does a Jamulus server benefits from a real-time kernel? Doesn't the internet protocol make the realtime not relevant?
I am also running one as was wondering.

kdoren, 2021-05-04:
@guysoft Jamulus server probably doesn't see any benefit  but I distribute the same image for multiple purposes.  Mostly people are running jamulus or other client, and the clients with jack audio are quite stable with the RT kernel build.  For that use, the RT kernel seems to have a small benefit over low-latency PREEMPT kernel.  This is without any real RT-specific tuning, so it might be possible to do better.

Jamulus server with multithreading enabled is not stable with my RT kernel build.  It will eventually trigger lxpanel to go to 100% CPU on one of the cores.  This can be avoided by running low-latency PREEMPT kernel instead of PREEMPT_RT, or by running jamulus server without multithreading.   I don't want to make a special image for jamulus server, so I may need to disable multithreading on jamulus-server, and give people instructions that if they want to enable multithreading they will need to use the PREEMPT kernel (I install both so it's easy to switch).  Either that or use PREEMPT for everything.

kdoren, 2021-05-16:
@guysoft @gtrainavicius I posted a release 5.10.35-rt39 that works on Pi3.  I applied the patches that were used on rpi-4.19.y-rt in addition to the PREEMPT_RT patch set, and that seems to work.

https://github.com/kdoren/linux/releases/tag/5.10.35-rt39

guysoft, 2021-05-17:
@kdoren it might be worth to build a branch and PR it in to this repo. Then I could build and ship.a new RealtimePi.

gtrainavicius, 2021-06-08:
@kdoren, how did you create your kernel config? (https://github.com/kdoren/linux/blob/rpi-5.10.35-rt/rt32.config)

Like what settings did you modify over the default configs? Is it meant for RPi 4, or also for other Pi versinos?

kdoren, 2021-06-09:
@gtrainavicius 

Latest release works on Pi3.

The .config files are part of the .deb package, and posted in repo.  changes from defaults are:

https://github.com/kdoren/linux/releases/tag/5.10.35-rt39
#Preemption model - Fully Preemptible Kernel (RT)
#Timer Freq - 1000 Hz
#cpufreq governor - performance
#hi res timers, dynticks tickless

I also have posted a release with (experimental) patches for Focusrite Scarlett interfaces
https://github.com/kdoren/linux/releases/tag/5.10.35-rt39-scarlett-2
This uses "tickless idle" instead of "dynticks tickless"

"Dynticks tickess" has worked well (I have been distributing it for last 6 months).
But I suspect (although I can't prove) that in rare cases that "dynticks tickless" can cause problems, so I am currently using "tickless idle".

guysoft, 2021-06-09:
Made a nightly build with the packages installed: http://unofficialpi.org/Distros/RealtimePi/nightly/2021-06-10_2021-05-07-realtimepi-buster-armhf-lite-0.5.0.zip

(packages from here: https://github.com/kdoren/linux/releases/tag/5.10.35-rt39)

guysoft, 2021-06-15:
If anyone wants to test it out: https://github.com/guysoft/RealtimePi/issues/29