raspberrypi/linux #3631: Raspi4/Buster USB-TV-stick causes endless errors

Max11223344, 2020-05-23:
i have a problem with the Terratec H6 (identified as Terratec H5 MKii) - Driver: dvb-usb-terratec-h5-drxk.fw.

I have 2 Raspis (Raspi3b+/Stretch) and Raspi4-4GB/Buster.
Problem is the different handling of the driver for the USB-TV stick (Terratec H6 or
H5MkII ).

With stretch/Leia 18.2/Raspi3b+ no errors
With buster/Leia 18.6/Raspi4 4GB always and never ending errors when TV playing

Below see dmesg when I connect the stick to the different Raspis:

Raspberry Pi 3B+
Raspbian Stretch (Kernel: Linux 4.19.66-v7+)
Kodi Leia 18.2 Git:20190428-a94ccc8ba5 - compiled 2019-04-30
Driver dvb-usb-terratec-h5-drxk.fw.

[ 327.813566] usb 1-1.3.3: new high-speed USB device number 21 using dwc_otg
[ 327.945084] usb 1-1.3.3: New USB device found, idVendor=0ccd, idProduct=10b2, bcdDevice= 1.00
[ 327.945095] usb 1-1.3.3: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 327.945100] usb 1-1.3.3: Product: TERRATCE H5 MKII
[ 327.945105] usb 1-1.3.3: Manufacturer: TERRATEC
[ 327.945110] usb 1-1.3.3: SerialNumber: 123456789ABCD
[ 327.945878] em28xx 1-1.3.3:1.0: New device TERRATEC TERRATCE H5 MKII @ 480 Mbps (0ccd:10b2, interface 0, class 0)
[ 327.945889] em28xx 1-1.3.3:1.0: Audio interface 0 found (Vendor Class)
[ 327.945895] em28xx 1-1.3.3:1.0: Video interface 0 found: isoc
[ 327.945900] em28xx 1-1.3.3:1.0: DVB interface 0 found: isoc
[ 328.013804] em28xx 1-1.3.3:1.0: chip ID is em2884
[ 328.090977] em28xx 1-1.3.3:1.0: EEPROM ID = 26 00 00 00, EEPROM hash = 0xbcd5a8cf
[ 328.090986] em28xx 1-1.3.3:1.0: EEPROM info:
[ 328.090991] em28xx 1-1.3.3:1.0: microcode start address = 0x0004, boot configuration = 0x00
[ 328.097948] em28xx 1-1.3.3:1.0: I2S audio, 5 sample rates
[ 328.097958] em28xx 1-1.3.3:1.0: 500mA max power
[ 328.097972] em28xx 1-1.3.3:1.0: Table at offset 0x27, strings=0x2298, 0x1c6a, 0x1286
[ 328.163517] em28xx 1-1.3.3:1.0: Identified as Terratec Cinergy H6 rev. 2 (card=101)
[ 328.163534] em28xx 1-1.3.3:1.0: Currently, V4L2 is not supported on this model
[ 328.163541] em28xx 1-1.3.3:1.0: dvb set to isoc mode.
[ 328.163719] em28xx 1-1.3.3:1.0: Binding audio extension
[ 328.163724] em28xx 1-1.3.3:1.0: em28xx-audio.c: Copyright (C) 2006 Markus Rechberger
[ 328.163729] em28xx 1-1.3.3:1.0: em28xx-audio.c: Copyright (C) 2007-2016 Mauro Carvalho Chehab
[ 328.163839] em28xx 1-1.3.3:1.0: Endpoint 0x83 high-speed on intf 0 alt 7 interval = 8, size 196
[ 328.163846] em28xx 1-1.3.3:1.0: Number of URBs: 1, with 64 packets and 192 size
[ 328.169691] em28xx 1-1.3.3:1.0: Audio extension successfully initialized
[ 328.169708] em28xx 1-1.3.3:1.0: Binding DVB extension
[ 328.263488] usb 1-1.3.4: new high-speed USB device number 22 using dwc_otg
[ 328.400889] usb 1-1.3.4: New USB device found, idVendor=0461, idProduct=037b, bcdDevice= 1.01
[ 328.400902] usb 1-1.3.4: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 328.400907] usb 1-1.3.4: Product: USB Scanner
[ 328.750619] drxk: status = 0x439260d9
[ 328.750642] drxk: detected a drx-3926k, spin A2, xtal 20.250 MHz
[ 332.766944] drxk: DRXK driver version 0.9.4300
[ 332.796015] drxk: frontend initialized.
[ 332.796054] tda18271 4-0060: creating new instance
[ 332.797813] tda18271: TDA18271HD/C2 detected @ 4-0060
[ 333.254709] dvbdev: DVB: registering new adapter (1-1.3.3:1.0)
[ 333.254739] em28xx 1-1.3.3:1.0: DVB: registering adapter 0 frontend 0 (DRXK DVB-C DVB-T)...
[ 333.257065] em28xx 1-1.3.3:1.0: DVB extension successfully initialized
[ 333.257087] em28xx 1-1.3.3:1.0: Registering input extension
[ 333.257479] Registered IR keymap rc-nec-terratec-cinergy-xs
[ 333.258173] rc rc0: 1-1.3.3:1.0 IR as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/rc/rc0
[ 333.258538] input: 1-1.3.3:1.0 IR as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/rc/rc0/input6
[ 333.259012] rc rc0: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
[ 333.259034] em28xx 1-1.3.3:1.0: Input extension successfully initialized
[ 333.357610] tda18271: performing RF tracking filter calibration
[ 335.932682] tda18271: RF tracking filter calibration complete
[ 335.932775] em28xx 1-1.3.3:1.0: DVB: adapter 0 frontend 0 frequency 0 out of range (47000000..864000000)
pi@raspberrypi3:~ $

Starting Kodi/TV there are no errors shown in dmesg.

Raspberry Pi4 4GB
[... 69 lines cut ...]
[ 1664.824232] drxk: 05 02 ..
[ 1664.824243] drxk: Error -22 on get_qam_lock_status
[ 1664.824251] drxk: Error -22 on get_lock_status
[ 1665.720279] drxk: i2c read error at addr 0x29
pi@raspberrypi:~ $ dmesg
[ 1634.827663] em28xx 1-1.4.3:1.0: write to i2c device at 0x52 failed with unknown error (status=1)
[ 1634.827669] drxk: i2c read error at addr 0x29
[ 1664.824224] drxk: ERROR: -32000
while sending cmd 0x0205 with params:
[ 1664.824232] drxk: 05 02 ..
[ 1664.824243] drxk: Error -22 on get_qam_lock_status
[ 1664.824251] drxk: Error -22 on get_lock_status
[ 1665.720279] drxk: i2c read error at addr 0x29
[ 1681.826525] em28xx 1-1.4.3:1.0: write to i2c device at 0x52 failed with unknown error (status=200)
[ 1681.826533] drxk: i2c read error at addr 0x29
[ 1683.826949] drxk: ERROR: -32768
while sending cmd 0x0205 with params:
[ 1683.826956] drxk: 05 02 ..
[ 1683.826967] drxk: Error -22 on get_qam_lock_status
[ 1683.826975] drxk: Error -22 on get_lock_status
[ 1688.827630] em28xx 1-1.4.3:1.0: write to i2c device at 0x52 failed with unknown error (status=200)
[ 1688.827641] drxk: i2c read error at addr 0x29
pi@raspberrypi:~ $

This Error-procedure goes on as long as the TV (Terratec H6) is playing and stops only when I stop TV

pelwell, 2020-05-23:
[ Moved to raspberrypi/linux, where it belongs ]

How are you powering your Pi 4 and the USB devices?

What does `rpi-eeprom-update` report on the Pi 4?

P33M, 2020-05-23:
Two things change between your 3B+ and 4B. The first is the actual hardware, the second is the kernel version.

Can you try using a fresh Buster image (and after doing `sudo apt update && sudo apt full-upgrade`) on your Pi 3 B+? This will help narrow it down.

Max11223344, 2020-05-24:
@pellwell

both raspberries are powered by the original Raspi power-supplies (2.5 and 3.1 A)

pi@raspberrypi:/ $ sudo rpi-eeprom-update
BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Thu 16 Apr 2020 05:11:26 PM UTC (1587057086)
 LATEST: Thu 16 Apr 2020 05:11:26 PM UTC (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad

@P33M

… sorry… took some time ….

-	downloaded new Raspbian (lite version)
-	made update
-	full-upgrade
-	installed Kodi
-	installed Tvheadend and USB stick
-	started TV

dmesg is on Pi3b+/Buster empty …. No errors 

so the software part seems to be fine?

Thanks for reading
Max11

pelwell, 2020-05-25:
I also asked how the other devices are powered, but it appears you don't have a hub attached.

Please try plugging the USB devices into a powered hub, if you have one.

Max11223344, 2020-05-26:
I connected the stick via a powered usb-hub. The errors are still there. So it isn't a question of power.

pelwell, 2020-05-26:
Thanks - we have to rule it out. Support is likely to be a bit slow for the next week or so, but this is on the list.

Max11223344, 2020-05-26:
Thank you ;-)   ... I will wait. At least TV is working.. even if it produces errors ;)

Max11223344, 2020-07-27:
Hello,

are there any new developments?
My dmesg still shows:

[ 3464.362265] drxk: Error -22 on get_qam_lock_status
[ 3464.362273] drxk: Error -22 on get_lock_status
[ 3477.092121] drxk: ERROR: -32000
                while sending cmd 0x0205 with params:
[ 3477.092128] drxk: 05 02                                            ..
[ 3477.092138] drxk: Error -22 on get_qam_lock_status
[ 3477.092146] drxk: Error -22 on get_lock_status
[ 3591.372881] drxk: ERROR: -32000
                while sending cmd 0x0205 with params:
[ 3591.372888] drxk: 05 02                                            ..
[ 3591.372898] drxk: Error -22 on get_qam_lock_status
[ 3591.372906] drxk: Error -22 on get_lock_status
[ 3597.373782] drxk: ERROR: -32000
                while sending cmd 0x0205 with params:
[ 3597.373789] drxk: 05 02                                            ..
[ 3597.373799] drxk: Error -22 on get_qam_lock_status
[ 3597.373808] drxk: Error -22 on get_lock_status
......
endless ....

ChrizZz90, 2021-03-28:
Hi @Max11223344, why did you close this issue? I have similar errors with another Terratec device. Can you please share the solution?

Max11223344, 2021-04-30:
Here we go again .... :(  
After updating/upgrading (stupid me) i have now
Linux raspberrypi 5.10.32-v7l+ #1413 SMP Tue Apr 27 14:43:13 BST 2021 armv7l GNU/Linux and i have too:
... endless errors .....
[dmesg.txt](https://github.com/raspberrypi/linux/files/6405780/dmesg.txt)
... and finally the TV stops totally ... which did not happen with the old errors  .... So the problem is worse now ...

How can i go back to e.g. raspberry 5.4 ?

Max11223344, 2021-04-30:
ok ... reading helps ;)

I just went back to raspberry 5.4.83 ...
No more errors ...

Max11223344, 2021-04-30:
This is crazy ...
I tried different versions down to 4.19.402  ... error still there ..
Then i switched the usb-boot- stick from the lower to the upper usb 3.0 slot, went back to 5.10.32 : no more errors ...
Put the stick back to the lower USB 3.0 port .... no more errors ......

Can someone pleae explain this behaviour??

am clueless ... and happy, that it is working (but for how long?)

Max11223344, 2021-05-02:
Hm ...
no more errors because i didn't start kodi (and tvheadend) 
As long as i stay within Raspbian there are no errors, but as soon as i start Kodi the errors appear. Googling around it seems it is a driver problem (?)
Any ideas?

Max

ChrizZz90, 2021-05-02:
I would try it at https://linuxtv.org/lists.php and share your issue which is related to the firmware. Maybe there was an update recently and someone is actively working on this chipset/firmware. Most of the manufacturers are actively reading and working on Linux TV.

Max11223344, 2021-05-04:
I now made -via SD Card Copier- a copy of the running Pi4 system. Connected the Terratec USB stick from Pi4 to Pi3.
Started the Pi3: no errors
[dmesg Raspi3 Terratec.txt](https://github.com/raspberrypi/linux/files/6423083/dmesg.Raspi3.Terratec.txt)
So the problem is in Rasperry Pi4 .. and not in the em28xx

nextlooper42, 2021-09-01:
Yes, something seems to be really broken in RPi4 codebase. I tried Hauppauge DualHD, OSMC DVB-T2/DVB-C TV dongle and random chinese dongles and all of them have corrupted output for no reason

Curiously this is NOT happening on RPi3B+ with exactly same setup (drivers also).

So problem is not in tuners itself, but with RPi4... related to #4243

Max11223344, 2021-09-01:
So you experience the same problem.
I updated in between to :
 uname -a
Linux raspberrypi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux

The never ending spamming error is still there :(

ChrizZz90, 2021-09-01:
well, I solved this issue just by testing lots of different devices. There are chips that work and I am using now this one: August DVB-T230 - I don't have errors, sometimes they appear and I have to plug the devices in again and I do a restart of my Pi4.

Oh, one thing I need to add: Yes it is a DVB-T device but DVB-C is also working ;)

nextlooper42, 2021-09-18:
Well that's **not solved issue with RPi4 FW**. That's just looking for device that may be using different drivers/stuff... Every em28xx driver based device still have these problems.

herrdeh, 2022-06-13:
On my system it looks as if the same problem occurs when I 

* first pause playback and 
* then terminate playback.

I'm using a [(MLD 5.5](minidvblinux.de) with RPi4 as well. 

`Jun 12 08:13:19 MLD-rpi4 user.err kernel: [37482.773550] drxk: Error -5 on get_lock_status
Jun 12 08:13:19 MLD-rpi4 user.warn kernel: [37482.883357] em28xx 1-1.3:1.0: writing to i2c device at 0x52 failed (error=-5)
Jun 12 08:13:19 MLD-rpi4 user.err kernel: [37482.883389] drxk: i2c read error at addr 0x29 `

A remedy for this problem would be greatly appreciated. Let me know if I can deliver more info or testing.
Cheers,
Wolf