raspberrypi/linux #3632: The hifiberry-dac code is not present

gotzl, 2020-05-23:
**Describe the bug**
The hifiberry-dac code (ie `sound/soc/bcm/hifiberry_dac.c`) is missing, even though the entry for `SND_BCM2708_SOC_HIFIBERRY_DAC` still exists in the respective `Kconfig`.
I don't know if this was done on purpose, but I guess, either one should adapt `Kconfig` or (what I would prefer ^^) add the code again.

**To reproduce**
Set `CONFIG_SND_BCM2708_SOC_HIFIBERRY_DAC=m` and compile the kernel.

**Expected behaviour**
Have a kernel module called `hifiberry-dac.ko`

**Actual behaviour**
No `hifiberry-dac.ko` module exists.

**Additional context**
I only checked the `rpi-4.19.y` branch and [created a patch](https://gist.github.com/gotzl/492fc565c12858f0293f9d4b9546b57a) as workaround.

pelwell, 2020-05-23:
That card is now managed by the rpi-simple-soundcard driver: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/sound/soc/bcm/rpi-simple-soundcard.c

gotzl, 2020-05-23:
Ah, I see! Thanks alot, closing this.

closed 2020-05-23 by gotzl