raspberrypi/linux #3634: Drop commit "zswap: Uncross module parameter setting functions"

hvenev, 2020-05-24:
This commit changes the meanings of the `zswap.compressor` and `zswap.zpool`. In particular, after 9497552330ea268fed5dc7fbb79f42af000585af, `zswap.compressor=` sets the `zpool` implementation used by zswap, and `zswap.zpool=` sets the compression algorithm. I find this very unintuitive compared to how it was before.

For example, consider the old implementation `zswap_compressor_param_set`.

That calls `__zswap_param_set(value_we_are_setting_for_compressor, kp, old_value_of_zswap_zpool_type), NULL)`

We enter `__zswap_param_set` with:
```
val = value_we_are_setting_for_compressor
kp = kp_for_compressor
type = old_value_of_zswap_zpool_type
compressor = NULL
```

In that function, we set

```
s = strstrip((char *)val)
  = strstrip(value_we_are_setting_for_compressor)
```

Then, we check

```
strcmp(s,  *(char**)kp->arg)
  = strcmp(s, *(char**)kp_for_compressor->arg)
  = strcmp(s, old_value_for_zswap_compressor)
```

Then, if they differ, we check

```
if(!type) type = s;
```

`type != NULL`, so this does not happen.

Then we do

```
if(!compressor) compressor = s;
```

As `compressor` is `NULL`, it is set to `s`.

Then, we get a `zpool` with the given `type` and `compressor = strstrip(value_we_are_setting_for_compressor)`.

Now, consider the new implementation of `zswap_compressor_param_set` (the one after commit 9497552330ea268fed5dc7fbb79f42af000585af "zswap: Uncross module parameter setting functions").

Inside `__zswap_param_set`, `type` is `NULL` and `compressor` contains the old value. `type` is then set to the new value we want for `compressor`. I think this is wrong.

hvenev, 2020-05-24:
In particular, I can confirm that on an unmodified 5.6 kernel, the following works:

```
echo z3fold > /sys/module/zswap/parameters/type
echo lz4 > /sys/module/zswap/parameters/compressor
```

I can also confirm that on an unmodified 5.6 kernel the following fails:

```
echo z3fold > /sys/module/zswap/parameters/compressor
echo lz4 > /sys/module/zswap/parameters/type
```

Note that I have not tested commit 9497552330ea268fed5dc7fbb79f42af000585af, but I expect that it would make things worse.

pelwell, 2020-05-25:
You're right - it's been reverted.

hvenev, 2020-05-26:
Thanks.

One more thing, what's the status of upstreaming https://github.com/raspberrypi/linux/commit/61596d496145cd85135ec9daea3f652be227d5f3? I couldn't find it in https://marc.info/?l=linux-mm.

pelwell, 2020-05-26:
It's on the list of things to do, but it's holiday season here at the moment. Feel free.

hvenev, 2020-05-26:
> It's on the list of things to do, but it's holiday season here at the moment. Feel free.

I don't feel like upstreaming code I haven't written and don't really use.

closed 2020-05-26 by hvenev