raspberrypi/linux #3639: Google Maps and OS Maps: incorrect rendering in 3D mode on Pi 4

andrum99, 2020-05-28:
**Describe the bug**
Google Maps in 3D mode has problems rendering correctly on the Raspberry Pi 4. Ordnance Survey's OS Maps product also has trouble. See below screenshots.

[image]

[image]

Screenshot of OS Maps from a Windows 10 PC showing correct rendering:
[image]

**To reproduce**
Google Maps:
1. Open Chromium (or Firefox ESR) within Raspberry Pi OS on a Pi 4.
2. Navigate to maps.google.com
3. Wait for the map to load. A small box will appear at the bottom left allowing you to switch to satellite view.
4. Click on the satellite view box to switch to satellite view
5. Click tiny globe icon near bottom right corner to switch to 3D mode.
6. Observe line of distortion on righthand side of satellite view.
7. Hold down CTRL and drag up with mouse to tilt view. Observe further distortion - see screenshot.

OS Maps:
1. Open Chromium (or Firefox ESR) within Raspberry Pi OS on a Pi 4.
2. Navigate to osmaps.ordnancesurvey.co.uk
3. Log in, or create a free account.
4. Click on the 'layers' icon in the bottom right corner to switch to 'Aerial 3D' view.
5. Zoom in on the map.
6. Hold down CTRL and drag up with the mouse to tilt the view.
7. Observe that the labels are shown as black boxes, instead of the correct rendering (see Windows screenshot for correct rendering).

**Expected behaviour**
Correct rendering

**Actual behaviour**
Dostorted rendering - see screenshots

**System**
Clean build of today's new Raspberry Pi OS 32-bit (2020-05-27). Problem exists on the previous version, and on both 32-bit and 64-bit versions of latest OS. Screenshots are from Raspberry Pi OS (32-bit) with desktop. Note that the 64-bit Raspberry Pi OS image uses the 5.4 kernel, so the problem exists on both kernel versions.

[raspinfo.txt](https://github.com/raspberrypi/linux/files/4695316/raspinfo.txt)

popcornmix, 2020-05-28:
@naushir I can reproduce this. Can you pass this bug report on?

naushir, 2020-05-28:
This has been identified:

> The offending commit was:
> 
> commit 63f24c3c016bec14bc549b71475ea0f7f7fdf893
> Author: Fritz Koenig <frkoenig@google.com>
> Date:   Tue Jul 30 14:40:33 2019 -0700
> 
>      gallium: Enable MESA_framebuffer_flip_y
> 
>      Reviewed-by: Kenneth Graunke <kenneth@whitecape.org>
>      Reviewed-by: Kristian H. Kristensen <hoegsberg@google.com>
> 
> The reason of enabling this extension in llvmpipe was mainly by Chromium
> interest to do testing using VM of the extension
> MESA_framebuffer_flip_y. The origin is Chromium Issue 955198 [1].
> 
> This was enabled at the chromium codepaths for llvmpipe, but they had to
> finally revert the use, because the had an extra flip. This is part of a
> work on "Enable 2D canvas and Webgl GpuMemoryBuffer code paths on
> ChromeOS" [2] to save battery power. There are no updates since end of 2019.
> 
> There are also some piglit tests [3] that never landed that check the
> extension.
> 
> With this findings there is another alternative to fix the issue in the
> specific version of chromium we have in raspbian. We can disable this
> extension when we have chromium with llvmpipe. Probably this could be
> easily included in the chromium launch scripts when HW is RPI[123].
> 
> MESA_EXTENSION_OVERRIDE=-GL_MESA_framebuffer_flip_y
> 

@XECDesign is aware of the change required and will get to it when he has some spare cycles.

naushir, 2020-05-28:
Sorry, I thought this might have already been identified, but that was not the case.  Will pass on the bug report.

andrum99, 2020-05-28:
Please ignore the bit about OS Maps - they must have fixed something at their end, as it is now workign correctly on the Pi 4.

Google Maps still shows the same problem.

txenoo, 2020-06-23:
This has been fixed and merged in Mesa master at https://gitlab.freedesktop.org/mesa/mesa/-/commit/ba15bb383f0756703e05bf202edc4771a1c54560

The corruption was caused by a compilation issue in shader 87 in google maps. It had an access to an indirect uniform that was used in a conditional after several ALU operations, but the indirect access was based of an input attribute. V3D was considering this situation as uniform-control of the conditional (all processes vertex will jump to the same branch when the condition is evaluated), but in this case each vertex processed could need a different branch of the conditional. So compiler should generate a non-uniform control structure in the GPU assembly situation.

At the end the fix is a one-liner patch.

JamesH65, 2020-06-23:
Thanks for the update! (And nice work finding the issue)

popcornmix, 2020-06-23:
Ping @XECDesign

andrum99, 2020-06-23:
Thanks for the update. Once the fix makes it into Raspberry Pi OS, this bug can be closed.

andrum99, 2020-12-31:
I discovered a way to run stock Raspberry Pi OS-supplied Chromium with Mesa 20.3.0 using binaries from https://github.com/smartavionics/Cura/releases/tag/20201121 using LD_LIBRARY_PRELOAD and can confirm this is now fixed. Given this is now fixed, and was not actually a bug in Linux, I will now close this issue.

(Hopefully it won't be too long before Raspberry Pi OS gets an updated version of Mesa).

closed 2020-12-31 by andrum99

XECDesign, 2021-01-06:
Sorry I missed the nudge and thank you for finding the appropriate patch. The mesa package has now been updated.

andrum99, 2021-01-06:
> Sorry I missed the nudge and thank you for finding the appropriate patch. The mesa package has now been updated.

Many thanks - I wasn't really expecting you to do a respin of the package just for this patch - what I meant was that I was assuming that the Mesa packages in Raspberry Pi OS would be getting a major version upgrade in the next few months to bring Vulkan etc to the OS, perhaps around the same time Raspberry Pi OS gets upgraded to kernel 5.10.

XECDesign, 2021-01-06:
Small patches are no biggie, but major Mesa package version bumps are risky and take time. I haven't had a look at the latest release, but in the past I've seen many changes to dependencies and general restructuring which would break other software if not handled carefully. I wouldn't expect a major bump before Bullseye.