raspberrypi/linux #3641: When I move windows there are artifacts around

xlazom00, 2020-05-28:
**Describe the bug**
When I move windows in  new 64bit raspbery debian based distro there are artifacts around windows

**To reproduce**
start 64bit debian raspberry pi distro on Pi4 with desktop environment, start terminal and move with window

**Expected behaviour**
no artifacts

**Actual behaviour**
artifacts around window

**System**
official beta 64bit debian 10  for raspberry pi 
Pi4 2GB ram
linux 5.4

pelwell, 2020-05-28:
We have a dedicated repo for the 64-bit OS: https://github.com/raspberrypi/Raspberry-Pi-OS-64bit

Please create the issue there.

xlazom00, 2020-05-28:
@pelwell same behavior on 32bit raspbian

xlazom00, 2020-05-28:
It is visible when I move from right to left and back

pelwell, 2020-05-28:
Are they temporary glitches or do they remain when you have stopped moving the window?

xlazom00, 2020-05-28:
Just temp. All fine when I stopped.
 I did same test with 32 bin raspian latest version and same behavior

xlazom00, 2020-06-01:
Are you able to reproduce my problem?

pelwell, 2020-06-02:
I see some tearing, but nothing that I would call artifacts. Try and capture a video showing the problem.

xlazom00, 2020-06-02:
Right it is tearing. any plans to fix this?

JamesH65, 2020-06-02:
It's on the list. Not sure yet what causes it, and because it doesn't affect functionality i.e. it's only cosmetic, it's low priority. Fixing tearing for video playback is also on the list, higher priority, may even be the same cause.

xlazom00, 2020-06-02:
PI4 is marketed as "desktop" computer and this is really lame bug

JamesH65, 2020-06-02:
Lame or not, its purely cosmetic and when we have fixed all the higher priority bugs that DO affect actual functionality, then we can look at it. Nothing to do with "desktop" marketing at all.

pelwell, 2020-06-02:
Closing this lame bug report.

closed 2020-06-02 by pelwell

codicusmaximus, 2020-09-02:
How on earth did this get closed when the problem wasn't even resolved?  Seems like a pretty hostile response @pelwell -- the number of downvotes should be pretty telling.

Just because this is "cosmetic" and there are higher priority bugs doesn't mean this is one should be marked as resolved, complete or closed @JamesH65

I propose this be reopened or if there is another bug in for the screen tearing issue, this issue be linked to that one for proper tracking.

JamesH65, 2020-09-03:
It was closed because 1) We know all about it and are looking in to it, but mainly 2) Because we are just getting an endless stream of "this is a lame bug" which are pointless additions to the issue.

We make 20 thousand Pi's a day, 11 downvotes it not that telling. BUT, we are working on this because we know it is an issue.

As for tracking, there is nothing to track. it'll get fixed when it get fixed (probably official release of KMS).

codicusmaximus, 2020-09-18:
So there is an issue, it is a known issue, work is being done on it, yet there's nothing to track.  In an issue tracker.  Makes sense to me 🙃

JamesH65, 2020-09-18:
Good, makes sense to us as well.

codicusmaximus, 2020-09-18:
The open source community at its finest!

codicusmaximus, 2020-09-18:
I'm just gonna return my Pi's and move to a platform with leadership that doesn't have their heads up their own butts.  Bye!

P33M, 2020-09-18:
> I'm just gonna return my Pi's and move to a platform with leadership that doesn't have their heads up their own butts. Bye!

Thank you for your valued contribution to open source software.