raspberrypi/linux #3642: Bug in i2c-rtc-overlay.dts (and its gpio version)

kpfleming, 2020-05-28:
https://github.com/raspberrypi/linux/blob/a9cf165a02c83810dc5d0e6b5358c1a21bf4e6a0/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts#L232

`abracon,tc-resistor` needs to have a `:0` suffix as the other entries do, otherwise the value provided in config.txt is incorrectly applied to the Device Tree. I've just tested this change locally and adding `:0` here resulted in the driver actually using the value I provided (it was ignored otherwise due to an -EOVERFLOW result from `of_property_read_u32`).

kpfleming, 2020-05-28:
I'd be happy to provide a PR for this, but I haven't got a clue how to set things up so it gets into all of the branches...

pelwell, 2020-05-28:
Thanks - it will probably be as easy for me to do it all, when I get back to a keyboard.

pelwell, 2020-05-29:
That's now fixed in all current branches. Thanks again.