raspberrypi/linux #3643: Not tagged: raspberrypi-kernel 1.20200512-2

Ve2mrx, 2020-05-28:
The package **raspberrypi-kernel 1.20200512-2** is available on the Raspberry Pi OS APT repo, but is not tagged for cloning. Custom build not possible (yet).

Solution: Please Tag it on GitHub.

Nice work with the new Pi4! 8GB will arrive soon :-)
Thanks,
Martin

sunfic, 2020-05-30:
I have similar issue. trying to build with  raspberrypi-kernel 1.20200512-2.

@Ve2mrx Can you guide me how you download the source of raspberrypi-kernel 1.20200512-2 right now? thanks

Ve2mrx, 2020-05-30:
> @Ve2mrx Can you guide me how you download the source of raspberrypi-kernel 1.20200512-2 right now? thanks

@sunfic All I can suggest is to try to compile from Git branch rpi-4.19.y and wish for the best until the release is tagged.

sunfic, 2020-05-31:
@Ve2mrx Thanks, seems right now it is 4.19-122 and I got some bugs in it. Hope we can get formal 1.20200512-2 release.

popcornmix, 2020-05-31:
@XECDesign is this something you normally do?

XECDesign, 2020-06-01:
Tags added

sunfic, 2020-06-05:
Thanks,

Another question
raspberrypi-kernel_1.20200512-2 and raspberrypi-kernel_1.20200601-1 are identical after I compared them. Why there are two tags there? anything updated?

XECDesign, 2020-06-05:
The tags correspond to the package version they come from in apt. The kernel and firmware both come from the same source package. If the firmware i updated, the kernel package gets a new version as well, even if the content doesn't change.

sunfic, 2020-06-06:
@XECDesign 
Thank you very much for you reply.

popcornmix, 2020-06-08:
@Ve2mrx okay to close?

Ve2mrx, 2020-06-08:
Sure!

closed 2020-06-08 by Ve2mrx