raspberrypi/linux #3644: Missing blkio and memory cgroup support on 3B+ and 3B, 5.4.42-v7+ kernel

DrumSergio, 2020-05-29:
**Describe the bug**
When I try to do anything related to containers in RPi, the software fails and messages that there is no memory cgroup. This issue comes from here: https://github.com/rancher/k3s/issues/1825

**To reproduce**
5.4.42-v7+ kernel, Raspbian 10 buster, only affecting 3B+ and 3B versions, at least not in the Pi 4.

**Expected behaviour**
Software to install successfully and allow management of containers

**Actual behaviour**
Printing of error messages related to cgroups

**System**
``` 
System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2

Linux K3sNode2 5.4.42-v7+ #1319 SMP Wed May 20 14:06:11 BST 2020 armv7l GNU/Linux
Revision	: a020d3
Serial		: 00000000a3b1c042
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

May 19 2020 22:31:46 
Copyright (c) 2012 Broadcom
version ba3a9fd1ea954b440ddea12f8f2b37b7ec1a2e23 (clean) (release) (start_cd)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       61112892 2034076  56564348   4% /
devtmpfs          466624       0    466624   0% /dev
tmpfs             499904       0    499904   0% /dev/shm
tmpfs             499904    6612    493292   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             499904       0    499904   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   54326    203769  22% /boot
tmpfs              99980       0     99980   0% /run/user/1000

Filename				Type		Size	Used	Priority

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20191105
openbox:
  Installed: (none)
lxpanel:
  Installed: (none)
pcmanfm:
  Installed: (none)
[... 372 lines cut ...]
[    7.662127] random: crng init done
[    7.662141] random: 7 urandom warning(s) missed due to ratelimiting
[    7.688048] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.720435] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[    9.293635] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    9.294698] 8021q: 802.1Q VLAN Support v1.8
[    9.298707] uart-pl011 3f201000.serial: no DMA platform data
[    9.318958] Bridge firewalling registered
[    9.639558] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   10.032600] 8021q: adding VLAN 0 to HW filter on device eth0
[   16.101378] Bluetooth: Core ver 2.22
[   16.101472] NET: Registered protocol family 31
[   16.101481] Bluetooth: HCI device and connection manager initialized
[   16.101507] Bluetooth: HCI socket layer initialized
[   16.101523] Bluetooth: L2CAP socket layer initialized
[   16.101548] Bluetooth: SCO socket layer initialized
[   16.118093] Bluetooth: HCI UART driver ver 2.3
[   16.118111] Bluetooth: HCI UART protocol H4 registered
[   16.118184] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.118427] Bluetooth: HCI UART protocol Broadcom registered
[   16.499671] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.499756] Bluetooth: BNEP filters: protocol multicast
[   16.499782] Bluetooth: BNEP socket layer initialized
[   44.848695] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

alemairebe, 2020-06-14:
same behavior here with `5.4.45-v7+ #1321 SMP Wed Jun 10 17:33:27 BST 2020 armv7l GNU/Linux`
`cat /etc/debian_version
10.4

cat /proc/cmdline
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory elevator=deadline fsck.repair=yes rootwait

cat /proc/cgroups
#subsys_name	hierarchy	num_cgroups	enabled
cpuset	2	1	1
cpu	3	24	1
cpuacct	3	24	1
devices	4	24	1
freezer	6	1	1
net_cls	5	1	1
pids	7	29	1`
while a previous version of raspbian works as expected : 
`4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

cat /etc/debian_version
10.3

cat /proc/cmdline
coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:1F:7D:55 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=18fd3f5e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

cat /proc/cgroups
#subsys_name	hierarchy	num_cgroups	enabled
cpuset	2	3	1
cpu	6	35	1
cpuacct	6	35	1
blkio	7	35	1
memory	3	226	1
devices	9	35	1
freezer	4	3	1
net_cls	8	3	1
pids	5	39	1`

idcrook, 2020-07-20:
This just manifest itself in "release" Raspberry Pi OS on my 3B+

Is there a known good kernel I can revert to?

```
grep "status installed" /var/log/dpkg.log | grep -e raspberrypi -e libraspberrypi
2020-07-20 06:30:07 status installed raspberrypi-kernel:armhf 1.20200717-1
2020-07-20 06:30:17 status installed raspberrypi-bootloader:armhf 1.20200717-1
2020-07-20 06:30:18 status installed libraspberrypi0:armhf 1.20200717-1
2020-07-20 06:30:18 status installed libraspberrypi-doc:armhf 1.20200717-1
2020-07-20 06:30:18 status installed libraspberrypi-bin:armhf 1.20200717-1
2020-07-20 06:30:18 status installed libraspberrypi-dev:armhf 1.20200717-1
```

kernel state

```
uname -a
Linux rpihp2 5.4.51-v7+ #1326 SMP Fri Jul 17 10:45:25 BST 2020 armv7l GNU/Linux

cat /proc/cgroups
#subsys_name	hierarchy	num_cgroups	enabled
cpuset	5	1	1
cpu	7	28	1
cpuacct	7	28	1
devices	4	28	1
freezer	6	1	1
net_cls	2	1	1
pids	3	32	1
```

memory cgroup not listed, even though it is in `/boot/cmdline.txt`

``` diff -u /boot/cmdline.txt /boot/cmdline_backup.txt
--- /boot/cmdline.txt	2020-07-20 07:30:54.000000000 -0600
+++ /boot/cmdline_backup.txt	2020-07-20 07:11:22.000000000 -0600
@@ -1 +1 @@
-console=serial0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes cgroup_memory=1 cgroup_enable=memory rootwait
+console=serial0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

mnorrsken, 2020-07-22:
Same problem here. Previously it worked without cmdline.txt parameters. Fixed by using 64 bit kernel and adding cgroup_enable=memory.

See my post at rpi forum: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=280656

idcrook, 2020-07-22:
> **Fix:** use 64 bit kernel and add cgroup_enable=memory
> (however this reduces IO performance on rpi3b+ with around 20%)

Your post at rpi form does explain or point to how to move to 64 bit kernel.
I'd rather not use this workaround on a 3B+ with 1GB RAM running Raspberry Pi OS as a k3s control plane node. 64 bit ubuntu for the Pi is my OS of choice on the 4GB/8GB RAM Pi 4s I have available for k3s until this raspberry pi kernel issue is fixed.

The "fix" would actually be to re-define the CONFIG_MEMCG and related CGROUP defines  (is there a reason why they stopped being defined?) in the kernel released...

idcrook, 2020-07-22:
I see that editing `/boot/config.txt` to include

```
arm_64bit=1
```
will choose the 64bit kernel, which includes the memory in cgroup-s, as @mnorrsken mentioned.

```shell
$ uname -a
Linux rpihp2 5.4.51-v8+ #1326 SMP PREEMPT Fri Jul 17 10:58:13 BST 2020 aarch64 GNU/Linux

$ cat /proc/cgroups
#subsys_name	hierarchy	num_cgroups	enabled
cpuset	9	14	1
cpu	3	41	1
cpuacct	3	41	1
blkio	5	41	1
memory	6	79	1
devices	8	41	1
freezer	2	14	1
net_cls	7	14	1
pids	4	45	1
```

Found here: [Pi4 64-bit raspbian kernel for testing - Focus on Pi4 - Raspberry Pi Forums](https://www.raspberrypi.org/forums/viewtopic.php?t=250730)

stevbev, 2020-07-24:
> This just manifest itself in "release" Raspberry Pi OS on my 3B+
> 
> Is there a known good kernel I can revert to?
> 
> ```
> grep "status installed" /var/log/dpkg.log | grep -e raspberrypi -e libraspberrypi
> 2020-07-20 06:30:07 status installed raspberrypi-kernel:armhf 1.20200717-1
> 2020-07-20 06:30:17 status installed raspberrypi-bootloader:armhf 1.20200717-1
> 2020-07-20 06:30:18 status installed libraspberrypi0:armhf 1.20200717-1
> 2020-07-20 06:30:18 status installed libraspberrypi-doc:armhf 1.20200717-1
> 2020-07-20 06:30:18 status installed libraspberrypi-bin:armhf 1.20200717-1
> 2020-07-20 06:30:18 status installed libraspberrypi-dev:armhf 1.20200717-1
> ```
> 
> kernel state
> 
> ```
> uname -a
> Linux rpihp2 5.4.51-v7+ #1326 SMP Fri Jul 17 10:45:25 BST 2020 armv7l GNU/Linux
> 
> cat /proc/cgroups
> #subsys_name	hierarchy	num_cgroups	enabled
> cpuset	5	1	1
> cpu	7	28	1
> cpuacct	7	28	1
> devices	4	28	1
> freezer	6	1	1
> net_cls	2	1	1
> pids	3	32	1
> ```
> 
> memory cgroup not listed, even though it is in `/boot/cmdline.txt`
> 
> ```diff
> --- /boot/cmdline.txt	2020-07-20 07:30:54.000000000 -0600
> +++ /boot/cmdline_backup.txt	2020-07-20 07:11:22.000000000 -0600
> @@ -1 +1 @@
> -console=serial0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes cgroup_memory=1 cgroup_enable=memory rootwait
> +console=serial0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
> ```

I just ran into this issue when the new kernel hit stable repository. I installed the updates across all my devices and took down my k8s cluster. The 20200717 and 20201723 kernels appear to be missing blkio and memory cgroup support. I resolved by downgrading to the previous 20200601 kernel packages. Commands below should help quickly revert.

wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20200601-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200601-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20200601-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-dev_1.20200601-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-doc_1.20200601-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20200601-1_armhf.deb
sudo dpkg -i *deb

pelwell, 2020-07-27:
The memory and block IO cgroups have been added to BCM2835-7, to match BCM2711 and previous kernels.

popcornmix, 2020-07-27:
Can you test rpi-update kernel and check if it has the features you need.

idcrook, 2020-07-27:
> The "fix" would actually be to re-define the CONFIG_MEMCG and related CGROUP defines (is there a reason why they stopped being defined?) in the kernel released...

and 

> Can you test rpi-update kernel and check if it has the features you need.

before update

```shell
$ cat /sys/firmware/devicetree/base/model
Raspberry Pi 3 Model B Plus Rev 1.3
$ uname -a
Linux rpihp2 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux
$ cat /proc/cgroups
#subsys_name	hierarchy	num_cgroups	enabled
cpuset	5	1	1
cpu	4	1	1
cpuacct	4	1	1
devices	2	25	1
freezer	7	1	1
net_cls	3	1	1
pids	6	31	1
$ sudo rpi-update
# ... reboot ...
```

*rpi-update* output
>  *** Updating kernel modules
>  *** depmod 5.4.51-v8+
>  *** depmod 5.4.51-v7+
>  *** depmod 5.4.51-v7l+
>  *** depmod 5.4.51+
>  *** If no errors appeared, your firmware was successfully updated to 56f62dee2a2385fe030724f7b39344fbc8c8d7ab

after update

```shell
$ uname -a
Linux rpihp2 5.4.51-v7+ #1328 SMP Mon Jul 27 16:51:11 BST 2020 armv7l GNU/Linux
$ cat /boot/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes cgroup_memory=1 cgroup_enable=memory rootwait
$ cat /proc/cgroups
#subsys_name	hierarchy	num_cgroups	enabled
cpuset	6	1	1
cpu	3	1	1
cpuacct	3	1	1
blkio	5	1	1
memory	8	56	1
devices	2	25	1
freezer	4	1	1
net_cls	7	1	1
pids	9	30	1
```

> **`blkio	5	1	1`**
> **`memory	8	56	1`**

So `cgroups_memory` is now present again.

idcrook, 2020-07-27:
> ```
> $ cat /proc/cgroups
> #subsys_name	hierarchy	num_cgroups	enabled
> blkio	5	1	1
> memory	8	56	1
> ```

> So `cgroups_memory` is now present again.

And I can confirm that `k3s` control plane node is now content runing again on a Raspberry Pi 3 Model B Plus Rev 1.3 with Raspberry Pi OS using the kernel version from recent `rpi-update`

bderenzo, 2020-07-27:
Same problem on RPi 4, I update to lastest kernel version using `rpi-update` but doesn't have the `cgroups_memory`.

```
hypi ~ $ sudo rpi-update 56f62dee2a2385fe030724f7b39344fbc8c8d7ab
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date
hypi ~ $ cat /boot/cmdline.txt
console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
hypi ~ $ uname -a
Linux hypi 5.4.51-v7l+ #1328 SMP Mon Jul 27 16:57:05 BST 2020 armv7l GNU/Linux
hypi ~ $ cat /sys/firmware/devicetree/base/model
Raspberry Pi 4 Model B Rev 1.1h
```

idcrook, 2020-07-27:
Your output shows that it indeed **does** have *cgroups* configured in the kernel

```
memory	0	33	0
```

enable in `cmdline.txt` by including : `cgroup_memory=1 cgroup_enable=memory `

pelwell, 2020-07-27:
Hint: Adding significant information by editing a comment effectively hides it from anyone following by email.

bderenzo, 2020-07-27:
@idcrook, thanks indeed it solve the problem. Is there a reason for this cgroup to not be loaded by default anymore ?
@pelwell, sorry for the mistake. I post and noticed then that I have forgotten the most important part.

mnorrsken, 2020-07-27:
Afaik memory cgroup enabled has a memory overhad of 8MB on a 1GB system (source: internet). So its understandable if default setting is disabled.

Also: great news, I will test the new kernel later.

bderenzo, 2020-07-27:
> Afaik memory cgroup enabled has a memory overhad of 8MB on a 1GB system (source: internet). So its understandable if default setting is disabled.
> 
> Also: great news, I will test the new kernel later.

8MB on 1GB is less than 1%. The problem is not so much the fact that it is not enabled, but the fact that it is no longer enabled by default.

swarred, 2020-09-24:
Hello, I was hesitant to do a firmware update as I did not want to have to deal with potential breaking changes for other configured software I had on one of my nodes. My issue was: 

`failed to find memory cgroup, you may need to add "cgroup_memory=1 cgroup_enable=memory" to your linux cmdline (/boot/cmdline.txt on a Raspberry Pi) ` 

when trying to join a node to my k3s cluster. 

My fix was to edit the `/boot/cmdline.txt`, as others have already suggested. However, do ensure that you are manually typing out `cgroup_memory=1 cgroup_enable=memory` on the same line. I do think it resolves the issue without firmware update needed.

bhkamath, 2020-10-03:
I have the same error; observe cgroups: memory related error even after a reboot

cat /boot/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=b7e4e595-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
cgroup_enable=cupset cgroup_memory=1 cgroup_enable=memory 

CGROUPS_MEMORY: missing
CGROUPS_HUGETLB: missing

	[WARNING SystemVerification]: missing optional cgroups: hugetlb
error execution phase preflight: [preflight] Some fatal errors occurred:
	[ERROR SystemVerification]: missing required cgroups: memory

Request your help.

bhkamath, 2020-10-04:
Got resolved, as it appeared to be a line separation issue.

ElDavoo, 2024-11-25:
Does anyone have issues with parameters being straight up ignored?
`[    0.000000] Unknown kernel command line parameters "cgroup_memory=1 cgroup_enable=memory", will be passed to user space.`
This is with a 6.12 kernel.

ElDavoo, 2024-11-25:
* it gets reverted by e196f8d and b413d2c does not put it back.

owlmind, 2024-11-29:
Could you pleas stop breaking cgroup_memory on my rpi?

popcornmix, 2024-11-29:
@owlmind are you on the experimental 6.12 kernel?

e-dong, 2024-12-12:
@ElDavoo I suggest creating a separate issue instead of necroing this issue from 2020, so it is clear that this is a regression with the recent 6.11 and 6.12 kernels.

pelwell, 2024-12-12:
Don't - it's already resolved: https://github.com/raspberrypi/linux/pull/6524

closed 2024-12-12 by pelwell

OfficialMuffin, 2026-02-22:
Sorry, unrelated to k8s but having a similar issue where memory is non-existent in `/proc/cgroups`. I have a Raspberry Pi 5 8GB on the `6.12.73-v8+` kernel. I have used `rpi-update` and rebooted with no avail. Heres the `/proc/croups/`

```
#subsys_name    hierarchy       num_cgroups     enabled
cpuset  0       77      1
cpu     0       77      1
cpuacct 0       77      1
blkio   0       77      1
devices 0       77      1
freezer 0       77      1
net_cls 0       77      1
perf_event      0       77      1
net_prio        0       77      1
pids    0       77      1

```

I see `cgroup_disable=memory` in dmesg log:

```
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave nvme.max_host_mem_s ize_mb=0  numa=fake=8 system_heap.max_order=0 iommu_dma_numa_policy=interleave s msc95xx.macaddr=REDACTED vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=a76987ea-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] cgroup: Disabling memory control group subsystem
```
However, its not written in `/boot/cmdline.txt`.

cmdline.txt:
`console=tty1 root=PARTUUID=a76987ea-02 rootfstype=ext4 fsck.repair=yes rootwait`

Could someone point me in the right direction please?