raspberrypi/linux #3646: vchiq_test: expected callback reason VCHIQ_BULK_TRANSMIT_DONE, got 5

lategoodbye, 2020-05-30:
**Is this the right place for my bug report?**
AFAIU this is more likely a kernel issue than a userspace one.

**Describe the bug**
If i try to run the functional test of vchiq_test on latest Raspberry Pi OS (64 bit), i aborts with an error.

**To reproduce**
If i try to run vchiq_test -f 1, i got the following error:
======== iteration 1 ========
vchiq_test: 1540: expected callback reason VCHIQ_BULK_TRANSMIT_DONE, got 5
vchiq_test: 856: func_error != 0

**Expected behaviour**
vchiq_test should succeed without any error. 

**Actual behaviour**
The functional test of vchiq_test aborts with an error.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Raspberry Pi 4 (4 and 8 GB)
* Which OS and version (`cat /etc/rpi-issue`)? Raspberry Pi OS (64 bit) (2020-05-27-raspios-buster-arm64)
* Which firmware version (`vcgencmd version`)? 2020-05-27 12:05 / 369ed4e44cb5a080a2dfa7f854ae4ff46b7c9ef9
* Which kernel version (`uname -a`)? Linux version 5.4.42-v8+ (dom@buildbot) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9)) #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020

pelwell, 2020-06-01:
This bug is already fixed in the current rpi-5.4.y branch by commits d8cbdaa729d5 ("staging: vchiq_arm: Clean up 40-bit DMA support") and 79495a5ecdfb ("ARM: dts: Update for new VCHIQ BCM2711 DMA support").

lategoodbye, 2020-06-01:
Okay, i will test as it's available via rpi-update.

lategoodbye, 2020-06-07:
I will retest after [this](https://github.com/raspberrypi/linux/commit/eac45bdf1e5f8ba6adf944c470746c48d2e0c661) commit is include. The kernel 5.4.44 crashes using vchiq_test.

closed 2025-01-02 by lategoodbye