raspberrypi/linux #3649: Pi Zero W on rpi-4.19.y-rt locks up if OTG USB used

Claudenw, 2020-06-01:
System works with raspberrypi-kernel but fails wit raspberrypi-kernel-rt

**Describe the bug**
When booting with raspberrypi-kernel-rt and a USB device attached the system shuts down.
When plugging in a USB device after boot the system locks up.

After reviewing the logs I believe that the issue is that the blue tooth operation conflicts with the OTG USB on the pi zero.

**To reproduce**

*** running lockup ***
using a pi zero W with blue tooth enabled running the raspberrypi-kernel and terminal via SSH

    sudo apt install raspberrypi-kernel-rt
    reboot
    
    log back in
    attach USB driver to OTG USB port.
    system locks up

*** running lockup ***
using a pi zero W with blue tooth enabled and the raspberrypi-kernel-rt installed, powered off.

    attach USB driver to OTG USB port.
    power on.
    system starts to boot and then shuts down.
    
**Expected behaviour**
Expect USB Storage to be recognized and OS to continue running.

**Actual behaviour**
System freezes or fails to boot.

**System**

[raspinfo.txt](https://github.com/raspberrypi/linux/files/4710441/raspinfo.txt)

**Logs**
Logs from boot with USB installed. 
[boot_with_usb.zip](https://github.com/raspberrypi/linux/files/4710461/boot_with_usb.zip)

Logs from pugging USB after boot (logs also contain entries from the above case)
[usb_after_boot.zip](https://github.com/raspberrypi/linux/files/4710487/usb_after_boot.zip)

pelwell, 2020-06-01:
You can report the issue here, as you've done, but you might not get a rapid response - we host the -rt branches, but we rely on others to support them.

P33M, 2020-06-01:
Is there any specific reason to use dwc_otg in this case? You appear to be using a single USB mass storage device, which dwc2 in host mode can handle just fine.

Claudenw, 2020-06-01:
I am just using the standard PI Zero W setup for the USB data setup.

pelwell, 2020-06-01:
Does that mean you are using the dwc2 overlay?

Claudenw, 2020-06-01:
I declare neither, so it looks like it is the dwc2 overlay by default for
the pi zero.  Sorry about mis-reporting the USB.