raspberrypi/linux #3650: Headphone jack doesn't work with kernel 5.4.

quantmint, 2020-06-01:
**Describe the bug**
Headphone jack doesn't work with kernel 5.4.
I used this config to build: https://github.com/moonman/MyPKGBUILDs/tree/master/linux-raspberrypi4.
Both dtoverlay=vc4-fkms-v3d and dtparam=audio=on are enabled in config.txt
**To reproduce**
aplay -l
aplay: device_list:274: no soundcards found...

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry pi 4 4gb
Arch linux arm 64 bit
* Which firmware version (`vcgencmd version`)?
version 21bfdeee3a6ea823e2113b983390acd1eec8edfb
* Which kernel version (`uname -a`)?
Linux pi4 5.4.42-1-ARCH #1 SMP PREEMPT Mon Jun 1 10:32:20 CEST 2020 aarch64 GNU/Linux

pelwell, 2020-06-01:
It works for me with the new 64-bit beta of the Raspberry Pi OS (https://www.raspberrypi.org/forums/viewtopic.php?t=275370), and also with a 64-bit kernel and 32-bit userspace. including with the current top-of-tree rpi-5.4.y kernel ("overlays: i2c-rtc-gpio: Fix trickle-resistor-ohms param").

pelwell, 2020-06-01:
```
pi@raspberrypi:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
```

closed 2021-09-02 by quantmint