raspberrypi/linux #3652: Confusion Pulseaudio device names.

audetto, 2020-06-01:
**Describe the bug**
Name of PulseAudio devices confusiong

**To reproduce**
Update to May 2020 version

**Actual behaviour**
When listing the available sinks in PulseAudio the HDMI and Analog devices have the same name

Only with a fully detailed ``pacmd list-sinks`` you can tell them apart

HDMI
	properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "bcm2835 HDMI 1"
		alsa.id = "bcm2835 HDMI 1"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "bcm2835 HDMI 1"
		alsa.long_card_name = "bcm2835 HDMI 1"
		alsa.driver_name = "snd_bcm2835"
		device.bus_path = "platform-soc:audio"
		sysfs.path = "/devices/platform/soc/soc:audio/sound/card0"
		device.form_factor = "internal"
		device.string = "hw:0"
		device.buffering.buffer_size = "131072"
		device.buffering.fragment_size = "131072"
		device.access_mode = "mmap+timer"
		device.profile.name = "analog-mono"
		device.profile.description = "Analog Mono"
		device.description = "Built-in Audio Analog Mono"
		alsa.mixer_name = "Broadcom Mixer"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"

Analog
	properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "bcm2835 Headphones"
		alsa.id = "bcm2835 Headphones"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "1"
		alsa.card_name = "bcm2835 Headphones"
		alsa.long_card_name = "bcm2835 Headphones"
		alsa.driver_name = "snd_bcm2835"
		device.bus_path = "platform-soc:audio"
		sysfs.path = "/devices/platform/soc/soc:audio/sound/card1"
		device.form_factor = "internal"
		device.string = "hw:1"
		device.buffering.buffer_size = "131072"
		device.buffering.fragment_size = "131072"
		device.access_mode = "mmap+timer"
		device.profile.name = "analog-mono"
		device.profile.description = "Analog Mono"
		device.description = "Built-in Audio Analog Mono"
		alsa.mixer_name = "Broadcom Mixer"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"

What matters in pavucontrol is the device.description which does not differentiate them
If the name could contian HDMI vs Analog it would be easier to use.

[image]

JamesH65, 2020-06-01:
We do not install Pulseaudio as standard (for reasons you may be encountering) , so its unsupported by us at the moment.

audetto, 2020-06-02:
Being such a wide used audio standard, it could be supported, even if not enabled by default.
And to be honest it works, just this little cosmetic improvement.

I am running a VNC client on a Pi and I redirect the audio back to the Pi over the network with pulseaudio.
For a standard desktop usage it works very well.

hvenev, 2020-06-06:
As we can see from https://cgit.freedesktop.org/pulseaudio/pulseaudio/tree/src/modules/alsa/alsa-util.c?h=v13.99.1#n905, PulseAudio first tries some sort of "default" name, and if that fails, generates a name based on the ALSA device name.

The default mechanism really likes to set the name to "Built-in Audio" if it detects that the device is built-in, as we can see from https://cgit.freedesktop.org/pulseaudio/pulseaudio/tree/src/pulsecore/sink.c?h=v13.99.1#n3510. The form factor is taken from udev's `SOUND_FORM_FACTOR` property.

For platform devices (such as the ones provided by `snd_bcm2835`), the form factor is set to `"internal"`, as we can see in https://github.com/systemd/systemd/blob/v245/rules.d/78-sound-card.rules#L56.

There are many ways to fix this. For example, improving udev's rules. I feel like `SOUND_FORM_FACTOR="internal"` should only be used for devices that are built-in speakers.

audetto, 2020-06-07:
The fact that it does not require a kernel change is good news.
But I am not too sure how to apply any of the above.

hvenev, 2020-06-07:
I simply added a rule that unsets the form factor if it was set to "internal":

```
SUBSYSTEM!="sound", GOTO="end"
ACTION!="change", GOTO="end"
KERNEL!="card*", GOTO="end"

ENV{SOUND_FORM_FACTOR}=="internal", ENV{SOUND_FORM_FACTOR}=""

LABEL="end"
```

Make sure to set the priority high enough (strictly greater than 78), so that the custom rules override the default ones.

hammerandtongs, 2020-06-07:
Sorry if its a derail but -

Is it working as intended that both the HDMI and the Heaphone are MONO just as listed in the OP?

Mine also come up that way on my Raspberry pi4...

audetto, 2020-12-06:
With the move to Pulseaudio this is becoming more relevant.

True that with the widget in the top bar one can see useful names, but the more standard and powerful pavucontrol still shows "Built-in Audio Analog Mono" for both.

BrunoMaire, 2021-04-10:
There is another possibility based on [https://gist.github.com/mpaolino/91c3587045fa8d1a9353b74bf102c83b](https://gist.github.com/mpaolino/91c3587045fa8d1a9353b74bf102c83b)
The problem I had was the wrong name in this example.
It had to be SOUND_DESCRIPTION instead of PULSE_NAME

To solve OP issue I would try
```
DEVPATH=="/devices/platform/soc/soc:audio/sound/card0", ENV{SOUND_DESCRIPTION}="HDMI1"
DEVPATH=="/devices/platform/soc/soc:audio/sound/card1", ENV{SOUND_DESCRIPTION}="Headphones"
```

Example:
For my Debian system I put a file named `96-static-soundcard-name.rules` in `/etc/udev/rules.d/`
with content:
```
SUBSYSTEM!="sound", GOTO="pa_naming_end"
ACTION!="change", GOTO="pa_naming_end"
KERNEL!="card*", GOTO="pa_naming_end"

DEVPATH=="/devices/platform/snd_aloop.0/sound/card?", ENV{SOUND_DESCRIPTION}="AlsaLoopback"
DEVPATH=="/devices/pci0000:00/0000:00:03.0/sound/card?", ENV{SOUND_DESCRIPTION}="IntelHDMI"
DEVPATH=="/devices/pci0000:00/0000:00:1b.0/sound/card?", ENV{SOUND_DESCRIPTION}="IntelPCH"
DEVPATH=="/devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card?", ENV{SOUND_DESCRIPTION}="NvidiaHDMI"

LABEL="pa_naming_end"
```