raspberrypi/linux #3653: kernel 5.4.42-v7l+ #1319

Billybangleballs, 2020-06-02:
Since the rpi-update installed the 5.4.42 kernel, I have strange entries in my kernel log

**System**
Pi4B

Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage4

Jun  1 2020 13:23:40
Copyright (c) 2012 Broadcom
version 6379679d1ec6a8c746d7e77e015f5b56b939976f (clean) (release) (start)

Linux sane4 5.4.42-v7l+ #1319 SMP Wed May 20 14:12:03 BST 2020 armv7l GNU/Linux

```
Jun  1 20:09:39 sane4 kernel: [12738.744120] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  1 20:09:39 sane4 kernel: [12738.744906] debugfs: Directory '1299' with parent 'clients' already present!
Jun  2 00:32:06 sane4 kernel: [28486.312966] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 00:32:06 sane4 kernel: [28486.314136] debugfs: Directory '1299' with parent 'clients' already present!
Jun  2 00:37:02 sane4 kernel: [28781.751161] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 00:37:02 sane4 kernel: [28781.751548] debugfs: Directory '1299' with parent 'clients' already present!
Jun  2 00:45:00 sane4 kernel: [29260.416102] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 00:45:00 sane4 kernel: [29260.416843] debugfs: Directory '1299' with parent 'clients' already present!
Jun  2 01:12:41 sane4 kernel: [30920.635114] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 01:12:41 sane4 kernel: [30920.635450] debugfs: Directory '1299' with parent 'clients' already present!
Jun  2 05:17:42 sane4 kernel: [45622.012919] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 05:17:42 sane4 kernel: [45622.013058] debugfs: Directory '1299' with parent 'clients' already present!
```

The process 1299 belongs to :-
`wonko     1299  1.0  4.4 683216 171328 ?       Sl   Jun01   8:13 /usr/lib/chromium-browser/chromium-browser-v7 --disable-quic --enable-tcp-fast-open --ppapi-flash-path=/usr/lib/chromium-browser/libpepflashplayer.so --ppapi-flash-args=enable_stagevideo_auto=0 --ppapi-flash-version=32.0.0.371 --enable-pinch`

Is this something I should be worried about?

jc-kynesim, 2020-06-04:
Offhand I have no ideas, but I could well believe that due to sandboxing in Chrome some part of an vcsm or mmal initialization process can't see that a file already exists & tries to recreate it anyway.

Billybangleballs, 2020-06-04:
It continued after I posted, eventually tiring of process 1299 and choosing 18792 instead. 

`Jun  2 06:00:57 sane4 kernel: [48216.757622] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 06:00:57 sane4 kernel: [48216.757783] debugfs: Directory '1299' with parent 'clients' already present!
Jun  2 13:48:03 sane4 kernel: [76243.288079] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 13:48:03 sane4 kernel: [76243.289216] debugfs: Directory '1299' with parent 'clients' already present!
Jun  2 18:47:56 sane4 kernel: [94237.284968] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 18:47:56 sane4 kernel: [94237.285199] debugfs: Directory '1299' with parent 'clients' already present!
Jun  2 20:28:29 sane4 kernel: [100270.074360] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  2 20:28:29 sane4 kernel: [100270.074557] debugfs: Directory '1299' with parent 'clients' already present!
Jun  3 02:28:02 sane4 kernel: [121843.844461] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  3 02:28:02 sane4 kernel: [121843.844603] debugfs: Directory '1299' with parent 'clients' already present!
Jun  3 03:09:44 sane4 kernel: [124345.416531] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  3 03:09:44 sane4 kernel: [124345.416716] debugfs: Directory '1299' with parent 'clients' already present!
Jun  3 15:14:55 sane4 kernel: [167857.618213] debugfs: Directory '1299' with parent 'alloc' already present!
Jun  3 15:14:55 sane4 kernel: [167857.618382] debugfs: Directory '1299' with parent 'clients' already present!
Jun  4 06:26:21 sane4 kernel: [222544.081034] debugfs: Directory '18792' with parent 'alloc' already present!
Jun  4 06:26:21 sane4 kernel: [222544.084183] debugfs: Directory '18792' with parent 'clients' already present!
Jun  4 06:33:31 sane4 kernel: [222974.294878] debugfs: Directory '18792' with parent 'alloc' already present!
Jun  4 06:33:31 sane4 kernel: [222974.295093] debugfs: Directory '18792' with parent 'clients' already present!
Jun  4 07:18:27 sane4 kernel: [225669.891203] debugfs: Directory '18792' with parent 'alloc' already present!
Jun  4 07:18:27 sane4 kernel: [225669.891500] debugfs: Directory '18792' with parent 'clients' already present!
Jun  4 12:00:31 sane4 kernel: [242594.080243] debugfs: Directory '18792' with parent 'alloc' already present!
Jun  4 12:00:31 sane4 kernel: [242594.080412] debugfs: Directory '18792' with parent 'clients' already present!
Jun  4 13:20:25 sane4 kernel: [247388.210624] debugfs: Directory '18792' with parent 'alloc' already present!
Jun  4 13:20:25 sane4 kernel: [247388.210982] debugfs: Directory '18792' with parent 'clients' already present!`

Again the process 18792 also belongs to chromium-browser, I have not noticed anything untoward happening, but it only started when the kernel version was bumped up to 5. I know very little about these things, but it was suggested I post the information here, so that minds immeasurably superior to mine could ponder the implications.

pelwell, 2020-06-04:
> so that minds immeasurably superior to mine could ponder the implications

If only there were some.

jbeale1, 2021-04-25:
I just got 104 lines in my 'dmesg' output of the form:
`[ 7179.245765] debugfs: Directory '1098' with parent 'clients' already present!`
I don't remember this happening before. Process 1098 is the chromium browser.
`$ ps aux | grep 1098
pi        1098  5.1  7.9 1194316 311668 ?      Sl   Apr24  40:00 /usr/lib/chromium-browser/chromium-browser-v7 --force-renderer-accessibility --disable-quic --enable-tcp-fast-open --check-for-update-interval=31536000 --enable-pinch --disable-features=AudioServiceOutOfProcess
$ uname -a
Linux rp49 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
`