raspberrypi/linux #3658: Kernel module on 5.4.42+ works on PI4 but not on PI0

JohnForecast, 2020-06-05:
For the last couple of years I've maintained a github repository to keep the DECnet code running in the latest Raspbian releases. Testing with 5.4.42+ shows a problem where the same kernel code (net/core/neighbour.c) works if built for a PI4 but fails when built for a PI0.

The DECnet code uses its own neighbour structure:
```
struct dn_neigh {
   struct neighbour n;
   __le16  addr;
};
```

struct neighbour (include/net/neighbour) has as its last component:
```
   u8    primary_key[0];
```

So when the neighbour.c code accesses "primary_key" it is really accessing "addr". This is how it works on a PI4 but when I build the kernel for PI0, there's an extra 4 bytes  inserted after "primary_key" in struct neighbour so "addr" is never accessed by the neighbour.c routines.

To reproduce the problem you do not have to include my DECnet code. Just  build the kernel with debug information included and select the defconfig for a PI0.

Use GDB on the resulting vmlinux:

Expected behaviour (what I see on a PI4 build):
```
(gdb) p/x sizeof(struct neighbour)
$3 = 0x128
(gdb)  p &((struct neighbour *)0)->primary_key
$4 = (u8 (*)[]) 0x128
(gdb) 
```
Both values should be the same.

Actual behaviour on a PI0 build:
```
(gdb) p/x sizeof(struct neighbour)
$1 = 0x118
(gdb)  p &((struct neighbour *)0)->primary_key
$2 = (u8 (*)[]) 0x114
(gdb) 
```
The structure is 4 bytes larger than the offset to the last component of struct neighbour.

The system I'm building this on:
```
Pi4B (2GB)

/etc/rpi-issue:
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2

vcgencmd version:
May 11 2020 18:59:41 
Copyright (c) 2012 Broadcom
version 21bfdeee3a6ea823e2113b983390acd1eec8edfb (clean) (release) (start)

uname -a:
Linux decnet 5.4.42-v7l+ #1 SMP Mon Jun 1 12:03:27 EDT 2020 armv7l GNU/Linux

I've tested with both the 5.4.y and 5.5.y branches:

5.4.y:  eac45bdf1e5f8ba6adf944c470746c48d2e0c661
5.5.y:  96a3278497cc6540d32dc4a7dba9100da6df16f0
```

6by9, 2020-06-05:
There's an interesting keyword at the end of struct neighbour - ```__randomize_layout;``` https://elixir.bootlin.com/linux/v5.4.44/source/include/net/neighbour.h#L161
Described in https://lwn.net/Articles/719732/

How does that work with a ```u8 var[0];``` at the end?
Have we managed to get it enabled for the Pi0/1 builds but not Pi4?

What compiler and version are you using to build the kernel in each case?

I assume this has worked on older kernel versions.

popcornmix, 2020-06-05:
`struct dn_neigh` looks invalid to me. See: https://gcc.gnu.org/onlinedocs/gcc/Zero-Length.html

> Declaring zero-length arrays in other contexts, including as interior members of structure objects or as non-member objects, is discouraged. Accessing elements of zero-length arrays declared in such contexts is undefined and may be diagnosed.

JohnForecast, 2020-06-05:
The DECnet code was originally written for the 2.6 kernel, I think, and I only picked it up a couple of years ago. The __randomize_layout keyword implies that it should randomize the order of structure members but there's a lot of networking code in the kernel which uses the neighbour.c routines (e.g. arp) so it can't be moving "primary_key" around.

It looks as though the way I'm accessing the primary key was modified some time ago so I'm going to change my usage to the way arp.c does it; leave the space for the key (member addr in struct dn_neigh) but when I need to access the key's value by reaching into struct neighbour and de-referencing primary_key (e.g. *(__le16 *)dn->n.primary_key). Fortunately there aren't too many places I need to change.

JohnForecast, 2020-06-05:
I implemented the changes noted above and everything works on the PI0 and PI4 for kernels 4.19.126 and 5.4.44.

closed 2020-06-05 by JohnForecast