raspberrypi/linux #3659: Raspberry Pi Zero - brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available

blitmap, 2020-06-07:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Incomplete firmware for the Raspberry Pi Zero's BCM43430/1 Broadcom chip

**To reproduce**
Download Raspberry Pi OS and start it up? :-)

**Expected behaviour**
No corresponding error in dmesg/logs

**Actual behaviour**
```
[   17.548468] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.548902] usbcore: registered new interface driver brcmfmac
[   17.944349] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.944542] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   17.945710] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
```

**System**
```
root@raspberrypi:/boot# raspinfo
System Information
------------------

Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2

Linux raspberrypi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
Revision        : 9000c1
Serial          : 0000000095f0f55c
Model           : Raspberry Pi Zero W Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Apr 15 2020 11:43:08
Copyright (c) 2012 Broadcom
version 82f9bb929ce2186eb1824178c1ae82902ad6275c (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7417360 1636504   5447492  24% /
devtmpfs          217064       0    217064   0% /dev
tmpfs             221396       0    221396   0% /dev/shm
tmpfs             221396    3120    218276   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             221396       0    221396   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   52235    205861  21% /boot
tmpfs              44276       0     44276   0% /run/user/1000

Filename                                Type            Size    Used    Priority
/var/swap                               file            102396  0       -2

Package version information
[... 482 lines cut ...]
[   29.411959] Bluetooth: SCO socket layer initialized
[   29.439873] Bluetooth: HCI UART driver ver 2.3
[   29.439890] Bluetooth: HCI UART protocol H4 registered
[   29.444583] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   29.444889] Bluetooth: HCI UART protocol Broadcom registered
[   30.274270] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   30.274284] Bluetooth: BNEP filters: protocol multicast
[   30.274310] Bluetooth: BNEP socket layer initialized
[  490.748940] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  490.749010] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  495.724354] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  498.695376] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
```

**Logs**
```
[   17.548468] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.548902] usbcore: registered new interface driver brcmfmac
[   17.944349] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.944542] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   17.945710] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
```

**Additional context**
(none)

pelwell, 2020-06-07:
The error message is incorrect in this case - the 43438 firmware has a built-in CLM blob.

blitmap, 2020-06-07:
Ahh okay, thanks for the quick reply :).  This can be closed then.

closed 2020-06-07 by pelwell