raspberrypi/linux #3660: Possible domain hijack that prevents installation of OMV and NextCloud

fatmanneedscake, 2020-06-08:
[image]
So I have been trying to install OpenMediaVault and Nextcloud on my Raspberry Pi 4 and a domain that is required for the scripts http://raspbian-us.ngc292.space is resolved to a random website https://113store.cr/es/ which seems to have nothing to do with Raspbian. I filed an issue with the developers of the script and they have pointed me over to this GitHub repository since it is a Raspbian domain. I have tried accessing the domain via TOR, 4G on my cellphone with no change and it still resolves to 113store. I apologize in advance if I posted this issue in the wrong place, but I felt it was important to bring it to a developer's attention. Thank you kindly.

JamesH65, 2020-06-08:
Does look very suspicious, be do not have any Raspbian domains, so this doesn't appear to be anything to do with us. 

You could try the Raspbian developers on www.raspbian.org but I suspect this is somethign else entirely. What scripts are being requested?

fatmanneedscake, 2020-06-08:
Alright, I will contact them at a later time.
__________________
https://raw.githubusercontent.com/OpenMediaVault-Plugin-Developers/installScript/master/install
That's a link to the OpenMediaVault script that I was trying to run on my RPI 4. It gets to the if statement and then gives me the error failed installing postfix because the domain resolves improperly. I not sure specifically how the script works, because I am a newbie. Sorry :(
if [[ ! "${omvInstall}" == "ii" ]]; then
  echo "Installing openmediavault required packages..."
  if ! apt-get --yes --no-install-recommends install postfix; then
    echo "failed installing postfix"
    exit 2
  fi

trejan, 2020-06-08:
raspbian-us.ngc292.space has already been removed from the Raspbian mirror pool. See https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=276340&p=1675102#p1675265