raspberrypi/linux #3667: vc4-kms-v3d-pi4 offscreen quad leaks to the left side of the screen buffer

oomek, 2020-06-11:
**Describe the bug**
When I draw a quad that extends beyond the right side of the screen buffer one column of pixels leaks to the left side shifted down by one pixel. This does not happen with `vc4-fkms-v3d` or if the `max_framebuffers=2`

**To reproduce**
`dtoverlay=vc4-kms-v3d-pi4
max_framebuffers=1`

**System**
 
* PI4
* Raspberry Pi reference 2020-02-13
* version 6379679d1ec6a8c746d7e77e015f5b56b939976f (clean) (release) (start)
* Linux retropie4 5.4.44-v7l+ #1320 SMP Wed Jun 3 16:13:10 BST 2020 armv7l GNU/Linux

**Screenshot**

[image]

popcornmix, 2020-06-11:
This is a known issue with the kms driver.
We had it with fkms driver initially but was resolved with very specific sequencing of enabling hardware blocks in the display pipeline.
In the kms world with generic linux code enabling the different blocks it is a harder problem to solve, but it's on the list of things to resolve.

ashimokawa, 2020-10-29:
I also had this when running 5.4.x almost every time, but it seems to be fixed in 5.9.x 
(it never occurs with the current "BRANCH=next" kernel)

6by9, 2020-10-29:
We are not intending to update 5.4 with the latest vc4-kms-v3d driver - there are far too many changes for it to be a trivial backport. 5.9 (and 5.10) will be the main branches for vc4-kms-v3d on Pi4.
I'm glad it's working on 5.9 though.

popcornmix, 2020-10-29:
It's not fixed on 5.9 - it's just intermittent. Had it on raspbian desktop yesterday and it's pretty common with kodi (which changes hdmi mode often).

ashimokawa, 2020-10-29:
@popcornmix 
Weird, I had that >90% after boot with 5.4 without changing mode, on 5.9 I never saw it. I might be lucky but there is definitely a huge improvement. I do not change modes often but the few occasions where I did it also did not happen on 5.9

ashimokawa, 2020-12-12:
In 5.10-rc7 this problem came back for my use case. I am sure it never happened for me during the 5.9 cycle and early 5.10-rcs.
Guess it is just random when some other code changes, timing changes, and the problem will get worse or better just by chance.

popcornmix, 2020-12-15:
Let us know what you do to provoke it. I can see issue but the kms guy is struggling to reproduce it.

ashimokawa, 2020-12-18:
@popcornmix 

Install retropie 4.7.1 on a sd-card

Change the config.txt

```
dtoverlay=vc4-kms-v3d-pi4,noaudio
max_framebuffers=1
disable_overscan=1
hdmi_force_hotplug=1
```

After booting, in the main menu there is a red horizontal line, you will notice that the leftmost part seems to "bend down", this is the rightmost column of the red line leaking to the left having an offset one pixel down.

This happened 90% of the time, so should be easy to reproduce. On 5.9.x to early 5.10.0-rcX I did not see that even once. At least since rc7 it is back.

popcornmix, 2021-06-25:
There's a [potential fix](https://github.com/raspberrypi/linux/pull/4410) for the 1-pixel wrap issue when using kms in latest rpi-update kernel.
If you are currently seeing this issue, then we'd be interested in reports of behaviour after updating.