raspberrypi/linux #3668: ERROR: linux-raspberrypi-1_5.4.40+gitAUTOINC+97badb4bae-r0 do_fetch: Fetcher failure: Unable to find revision 97badb4baeb8999263a9bc2fa85c3060ffff344b in branch rpi-5.4.y even from upstream

embdlover, 2020-06-11:
Hi,
 i am  build image on rpi 4 for 5.4.y kernel but i am getting following issue
pankaj@exaleap-Inspiron-3584:~/Oe-Raspberrypi4$ bitbake core-image-sato
Parsing recipes: 100% |######################################################################################################################################################################| Time: 0:01:54
Parsing of 2278 .bb files complete (0 cached, 2278 parsed). 3438 targets, 187 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "raspberrypi4"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 = "master:982b7f98b8423236cc986346379b1bde3694f131"
meta-raspberrypi     = "master:4982da8d48ce9e7025955d88d2cff6350b228c28"
meta-oe              
meta-python          
meta-networking      
meta-multimedia      
meta-perl            = "dunfell:e413c1ef621688e69bb7830bb3151ed23b30b73e"

Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:04
Sstate summary: Wanted 1094 Found 0 Missed 1094 Current 1154 (0% match, 51% complete)
NOTE: Executing Tasks
WARNING: libfm-extra-1.3.1-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/pcmanfm/libfm-1.3.1.tar.xz, attempting MIRRORS if available
WARNING: linux-raspberrypi-1_5.4.40+gitAUTOINC+97badb4bae-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;branch=rpi-5.4.y, attempting MIRRORS if available
ERROR: linux-raspberrypi-1_5.4.40+gitAUTOINC+97badb4bae-r0 do_fetch: Fetcher failure: Unable to find revision 97badb4baeb8999263a9bc2fa85c3060ffff344b in branch rpi-5.4.y even from upstream
ERROR: linux-raspberrypi-1_5.4.40+gitAUTOINC+97badb4bae-r0 do_fetch: Fetcher failure for URL: 'git://github.com/raspberrypi/linux.git;branch=rpi-5.4.y'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/pankaj/Oe-Raspberrypi4/tmp-glibc/work/raspberrypi4-oe-linux-gnueabi/linux-raspberrypi/1_5.4.40+gitAUTOINC+97badb4bae-r0/temp/log.do_fetch.10175
ERROR: Task (/home/pankaj/meta-raspberry/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2294 tasks of which 1551 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/pankaj/meta-raspberry/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_fetch
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
 
 can you help me, i am missing anything?

P33M, 2020-06-11:
Perhaps you should first ask whoever developed the script "bitbake". Hint: it wasn't us.

closed 2020-06-11 by P33M

timg236, 2020-06-12:
The bitbake script is working correctly. 

However, whoever created the bitbake config needs to be aware that the rpi-5.4.y kernel is not released and forced pushes to rebase will occur so commits refs will change. You might get away with a clean build, otherwise, you need you'll need to work with whoever created the bitbake config to cope with that.