raspberrypi/linux #3669: When will rpi-5.4.y stop being force-pushed/rebased?

mike-scott, 2020-06-11:
**Describe the bug**
It seems that the 5.4 kernel has been approved for general use as the firmware repository now has 5.4-based artifacts in it.  However, every week or 2 the git history of the rpi-5.4y branch is being re-written with a force push / rebase.  This means anyone who is depending on pinned SHAs as references in this repo will break every week or 2.

An easy example of this would be the meta-raspberrypi 5.4 kernel recipe:
https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux/linux-raspberrypi_5.4.bb

Is there a development time frame where the force-push/rebases will phase out and merge-only behavior begins?

Apologies if this is not the most appropriate place to ask this question.  There isn't a kernel-specific forum on the RaspberryPi forums.

Thank you!

pelwell, 2020-06-11:
Kernels are squashed, rebased and force pushed until they become the default branch, the one used to create. the current kernel package for Raspberry Pi OS & Raspbian. The aim is to create as clean a history as possible until the point where we want the tags for releases to persist.

P33M, 2020-06-11:
The 5.4 kernel has not been "approved for general use", at least not by us. Firmware builds retrieved from rpi-update (which indirectly use the raspberrypi/firmware repo) have been switched to use rpi-5.4.y, but these should never be used in production. 

The warning prominently printed to console whenever you invoke rpi-update warns you about this.

The canonical stable releases from this repository have git tags corresponding to their associated apt package releases.

c.f. the last time that someone started pulling from our development branch and got unexpected results: https://github.com/raspberrypi/linux/issues/2931

mike-scott, 2020-06-11:
@P33M @pelwell "approved for general use" is a misstatement on my part.  I apologize. 
 But it's not transparent "when" it is actually approved.  Would that be when 5.4 artifacts are pushed to the firmware's `stable` branch?  Is there any roadmap / timeline for that?

I mistakenly assumed that if there were 5.4 artifacts in the `master` branch that rebasing the 5.4 kernel sources would stop as that effectively makes reproducing those kernel binaries impossible (no pinned SHA is left).

mike-scott, 2020-06-11:
@P33M We don't use the Raspberry Pi OS, so we don't see the warning you are referring to.

mike-scott, 2020-06-11:
@P33M Thank you for linking the prior issue.  I read through it and @popcornmix comment at the end summarizes the information I was looking for very nicely.

```
You can see the current apt stable kernel from:
https://github.com/raspberrypi/firmware/blob/stable/extra/uname_string7

Or the source it was build from (latest tag) here:
https://github.com/raspberrypi/linux/tags

Also keep an eye on the Advanced Users forum. Threads like:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931

will typically give a few months notice of a planned major bump to the kernel.
```

If only that was available somewhere other than a closed issue :/

popcornmix, 2020-06-11:
A quick way of checking the stable kernel is to look at the default branch of this repo.
Currently https://github.com/raspberrypi/linux/ shows the rpi-4.19.y branch as the default option.
When we switch to 5.4 (in apt) the default branch will be rpi-5.4.y. And the default branch will not be rebased.

closed 2020-06-11 by mike-scott

Titaniumtown, 2020-06-12:
One question, how long does it usually take to transition to another stable branch? And is there an estimate of when 5.4.y will become default. 

Or put more simply, how far behind does the raspberry pi kernel lag behind?

pelwell, 2020-06-15:
It varies - we don't keep statistics and there are no hard targets. We choose LTS kernels for our stable branches for obvious reasons, switching to the next one when it feels ready, sometimes accelerated by the demand for a new feature.

Titaniumtown, 2020-06-15:
Oh ok, Thanks.