raspberrypi/linux #3671: WIFI Crash on RPI4

svenlimbach, 2020-06-12:
**Describe the bug**
If using the OnBoard WiFi in Client AP Mode the wifi crashes. The following messages are found in dmesg

[   25.866146] brcmfmac: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[   25.866154] brcmfmac: brcmf_cfg80211_scan: scan error (-52)

**To reproduce**
Install RaspAP with WiFi client AP mode

**Expected behaviour**
Wifi should not crash :-)

**Actual behaviour**
Wifi Crashes with the described error messages

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4 Model B Rev 1.2

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2

* Which firmware version (`vcgencmd version`)?
Jun  1 2020 13:23:40 
Copyright (c) 2012 Broadcom
version 6379679d1ec6a8c746d7e77e015f5b56b939976f (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

**Logs**
DMESG

pi@raspberrypi:/etc/hostapd $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.118-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1311 SMP Mon Apr 27 14:26:42 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s36928 r8192 d24512 u69632
[    0.000000] pcpu-alloc: s36928 r8192 d24512 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 cma=256M  smsc95xx.macaddr=DC:A6:32:5C:19:93 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=33458e9c-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3735552K/4050944K available (8192K kernel code, 690K rwdata, 2420K rodata, 2048K init, 860K bss, 53248K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 691 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 861 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[... 260 lines cut ...]
[    5.391029] random: crng init done
[    5.391043] random: 7 urandom warning(s) missed due to ratelimiting
[    5.506356] 8021q: 802.1Q VLAN Support v1.8
[    5.626654] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    5.719024] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    5.945015] bcmgenet: Skipping UMAC reset
[    6.048657] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
[    7.128278] bcmgenet fd580000.genet eth0: Link is Down
[   12.091580] Bluetooth: Core ver 2.22
[   12.091652] NET: Registered protocol family 31
[   12.091663] Bluetooth: HCI device and connection manager initialized
[   12.091683] Bluetooth: HCI socket layer initialized
[   12.091698] Bluetooth: L2CAP socket layer initialized
[   12.091742] Bluetooth: SCO socket layer initialized
[   12.107099] Bluetooth: HCI UART driver ver 2.3
[   12.107113] Bluetooth: HCI UART protocol H4 registered
[   12.107194] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.107419] Bluetooth: HCI UART protocol Broadcom registered
[   12.279738] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.279744] Bluetooth: BNEP filters: protocol multicast
[   12.279755] Bluetooth: BNEP socket layer initialized
[   16.093807] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[  365.740498] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[  365.953037] brcmfmac: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[  365.953054] brcmfmac: brcmf_cfg80211_scan: scan error (-52)

svenlimbach, 2020-06-12:
Maybe same error as for RPI3 Models

pelwell, 2020-06-12:
See https://github.com/raspberrypi/firmware/issues/1403 - there is an updated firmware to try, and instructions on how to enable logging from the WLAN firmware.

arend, 2020-06-13:
This is not a WiFi crash. The scan is simply failing. The error -52 seems to be [BCME_IE_NOTFOUND](https://android.googlesource.com/kernel/common/+/bcmdhd-3.10/drivers/net/wireless/bcmdhd/include/bcmutils.h#409)

arend, 2020-06-13:
So question is what IE is attempted to be set. Firmware only allows vendor-specific IEs so maybe the driver should have a filter for it.

PeterM7, 2020-06-17:
This issue exists in Buster. I have just tried to rebuild my Wheezy app in Raspi-OS and when that failed, dropped down to Buster, and it also failed. I am not sure how to find error information. All I know is that Wifi works until I add the Access Point Software  hostapd and dnsmasq, and after that I cannot see the wifi ESSID. 

This was tried on a Raspberry Pi B ver 2 using a Zyxel ZD1211 Wifi dongle and also a raspberry Pi Zero-W, on both of which the Wheezy version works.

pelwell, 2020-06-18:
There's another test firmware - the date is the same as the recent public release but the options list looks different (I have no other information): https://drive.google.com/file/d/10ivocg5PrOwVxAYFKOzJEdv_gdCd-IUF/view?usp=sharing

Be sure to rename to brcmfmac43455-sdio.bin when installing it.

PeterM7, 2020-06-18:
I found a solution to my problem in: https://unix.stackexchange.com/questions/119209/hostapd-will-not-start-via-service-but-will-start-directly

Because I found in syslog the following:
Jun 18 19:44:00 raspberrypi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
Jun 18 19:44:00 raspberrypi hostapd[15736]: Configuration file:
Jun 18 19:44:00 raspberrypi hostapd[15736]: Could not open configuration file '' for reading.
Jun 18 19:44:00 raspberrypi hostapd[15736]: Failed to set up interface with
Jun 18 19:44:00 raspberrypi hostapd[15736]: Failed to initialize interface
Jun 18 19:44:00 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited status=1
Jun 18 19:44:00 raspberrypi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Jun 18 19:44:00 raspberrypi systemd[1]: hostapd.service: Unit entered failed state.
Jun 18 19:44:00 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.

So it was necessary to edit the file /etc/default/hostapd with the line
DAEMON_CONF="/etc/hostapd/hostapd.conf"

This line was in the inital state commented and without the file name.

On a reboot the wifi came up happily.

This is something that I have not seen in any of the instructions about setting up hostapd before, and it was not necessary when I created my previous setup (unless it was automatically done fo me).

normanr, 2020-06-19:
I think BCME_IE_NOTFOUND is a red-herring. -52 is also -EBADE, which appears in drivers/net/wireless/broadcom/brcm80211 in many places. It should be possible to get more details by enabling tracing for brcmf_dbg. How I'm not sure.

arend, 2020-06-19:
> I think BCME_IE_NOTFOUND is a red-herring. -52 is also -EBADE, 
Yeah. Realized that last night. To see the actual firmware error you have to enable FIL debugging, ie. use modparam debug=0x1000.

normanr, 2020-06-19:
```
Jun 19 01:20:44 raspberrypi kernel: [  808.478487] brcmfmac: brcmf_fil_cmd_data Firmware error: BCME_BUSY (-16)
Jun 19 01:20:44 raspberrypi kernel: [  808.478507] brcmfmac: brcmf_run_escan: error (-52)
Jun 19 01:20:44 raspberrypi kernel: [  808.478517] brcmfmac: brcmf_cfg80211_scan: scan error (-52)
```

normanr, 2020-06-19:
In fact I can do better with debug=0x1006:

Jun 19 01:54:32 raspberrypi kernel: [ 1014.010667] brcmfmac: brcmf_cfg80211_scan Enter
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010700] brcmfmac: brcmf_vif_set_mgmt_ie bsscfgidx 0, pktflag : 0x10
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010735] brcmfmac: brcmf_fil_iovar_data_set ifidx=0, name=escan, len=132
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010743] brcmutil: data
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010758] 00000000: ... redacted ...
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010767] 00000010: ... redacted ...
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010773] 00000020: ... redacted ...
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010780] 00000030: ... redacted ...
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010794] brcmfmac: brcmf_sdio_bus_txctl Enter
Jun 19 01:54:32 raspberrypi kernel: [ 1014.010960] brcmfmac: brcmf_sdio_kso_control Enter: on=1
Jun 19 01:54:32 raspberrypi kernel: [ 1014.015706] brcmfmac: brcmf_sdio_bus_rxctl Enter
Jun 19 01:54:32 raspberrypi kernel: [ 1014.017723] brcmfmac: brcmf_sdio_isr Enter
Jun 19 01:54:32 raspberrypi kernel: [ 1014.018222] brcmfmac: brcmf_fil_cmd_data Firmware error: BCME_BUSY (-16)
Jun 19 01:54:32 raspberrypi kernel: [ 1014.018238] brcmfmac: brcmf_run_escan: error (-52)
Jun 19 01:54:32 raspberrypi kernel: [ 1014.018251] brcmfmac: brcmf_cfg80211_scan: scan error (-52)

arend, 2020-06-19:
let's get loud and use 0x1416.

normanr, 2020-06-19:
I also got this:

Jun 19 01:53:58 raspberrypi kernel: [  979.786931] ------------[ cut here ]------------
Jun 19 01:53:58 raspberrypi kernel: [  979.788087] WARNING: CPU: 0 PID: 5 at net/wireless/sme.c:756 __cfg80211_connect_result+0x388/0x400 [cfg80211]
Jun 19 01:53:58 raspberrypi kernel: [  979.788103] Modules linked in: brcmfmac brcmutil cfg80211 appletalk psnap ax25 cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic joydev 8021q garp stp llc evdev hid_microsoft sha256_generic raspberrypi_hwmon hwmon rfkill bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) media ir_rc6_decoder gpio_ir_recv gpio_ir_tx uio_pdrv_genirq fixed uio ip_tables x_tables ipv6 [last unloaded: brcmutil]
Jun 19 01:53:58 raspberrypi kernel: [  979.788239] CPU: 0 PID: 5 Comm: kworker/u2:0 Tainted: G        WC        4.19.118+ #1311
Jun 19 01:53:58 raspberrypi kernel: [  979.788244] Hardware name: BCM2835
Jun 19 01:53:58 raspberrypi kernel: [  979.789252] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Jun 19 01:53:58 raspberrypi kernel: [  979.789306] [<c0018008>] (unwind_backtrace) from [<c0014f50>] (show_stack+0x20/0x24)
Jun 19 01:53:58 raspberrypi kernel: [  979.789331] [<c0014f50>] (show_stack) from [<c06eac88>] (dump_stack+0x20/0x28)
Jun 19 01:53:58 raspberrypi kernel: [  979.789359] [<c06eac88>] (dump_stack) from [<c0024714>] (__warn+0xf4/0x11c)
Jun 19 01:53:58 raspberrypi kernel: [  979.789376] [<c0024714>] (__warn) from [<c002486c>] (warn_slowpath_null+0x4c/0x58)
Jun 19 01:53:58 raspberrypi kernel: [  979.790446] [<c002486c>] (warn_slowpath_null) from [<bf2825c4>] (__cfg80211_connect_result+0x388/0x400 [cfg80211])
Jun 19 01:53:58 raspberrypi kernel: [  979.792581] [<bf2825c4>] (__cfg80211_connect_result [cfg80211]) from [<bf2560ec>] (cfg80211_process_wdev_events+0x13c/0x1c8 [cfg80211])
Jun 19 01:53:58 raspberrypi kernel: [  979.794810] [<bf2560ec>] (cfg80211_process_wdev_events [cfg80211]) from [<bf2561b8>] (cfg80211_process_rdev_events+0x40/0x9c [cfg80211])
Jun 19 01:53:58 raspberrypi kernel: [  979.797198] [<bf2561b8>] (cfg80211_process_rdev_events [cfg80211]) from [<bf250258>] (cfg80211_event_work+0x24/0x2c [cfg80211])
Jun 19 01:53:58 raspberrypi kernel: [  979.798382] [<bf250258>] (cfg80211_event_work [cfg80211]) from [<c003d798>] (process_one_work+0x130/0x3a0)
Jun 19 01:53:58 raspberrypi kernel: [  979.798420] [<c003d798>] (process_one_work) from [<c003da3c>] (worker_thread+0x34/0x530)
Jun 19 01:53:58 raspberrypi kernel: [  979.798441] [<c003da3c>] (worker_thread) from [<c0043584>] (kthread+0x11c/0x158)
Jun 19 01:53:58 raspberrypi kernel: [  979.798462] [<c0043584>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
Jun 19 01:53:58 raspberrypi kernel: [  979.798469] Exception stack(0xdad01fb0 to 0xdad01ff8)
Jun 19 01:53:58 raspberrypi kernel: [  979.798483] 1fa0:                                     00000000 00000000 00000000 00000000
Jun 19 01:53:58 raspberrypi kernel: [  979.798495] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jun 19 01:53:58 raspberrypi kernel: [  979.798504] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jun 19 01:53:58 raspberrypi kernel: [  979.798515] ---[ end trace 5f15b547b308fecc ]---

I think might need mac80211 tracing (not enabled at compile time), to see what's going on inside of cfg80211_get_bss. Maybe we can also enable CONN (debug|=0x8000) to see what the driver is putting into the data structs.

normanr, 2020-07-11:
status update: I reverted to `Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f` and it didn't seem to help (maybe a tiny bit, but hard to tell because I didn't get many days of data). I disabled wicd.service service, and since that WiFi has been rock-solid (12 days so far).

pelwell, 2020-07-12:
Where did you get `wicd.service`?

normanr, 2020-07-12:
It got installed when I was adding X desktop to Raspbian Lite.  I don't remember if I chose it manually or if it was auto-installed. From https://bugs.debian.org/772996 it looks like it's bad news and quite probably the cause of my WiFi issues. 

I wonder if wicd and hostapd are tickling the same bug? It looks like a driver locking issue to me: something like two scans running at the same time, and then stomping on each other's results.