raspberrypi/linux #3676: Kernel 5.4 and missing GDB scripts

phryniszak, 2020-06-15:
After switching to kernel 5.4.(45) gdb kernel scripts stopped working.
Was working for my in 4.19.

I have in .config:

```
CONFIG_GDB_SCRIPTS=y
```

But in GDB no sign of lx- commands.

```
0xc001ebb4 in ?? ()
Reading symbols from ./vmlinux...
(gdb) lx-
Undefined command: "lx-".  Try "help".
(gdb) apropos lx
(gdb) c
```

Any clues?

pelwell, 2020-06-26:
Sorry - this one slipped past. You can't set `CONFIG_GDB_SCRIPTS=y` without also setting `CONFIG_GDB_INFO=y`.

pelwell, 2020-06-26:
Edited - that should have said `CONFIG_DEBUG_INFO=y`.

phryniszak, 2020-06-26:
I had CONFIG_DEBUG_INFO=y. I could see all symbols in GDB. 
One thing that didn't work were all python lx- commands. 
I try to rebuild 5.4 once more to confirm it.

phryniszak, 2020-06-27:
I tried again 5.4.49 and no luck. I used original rpi config and add in `CONFIG_DEBUG_INFO=y` and `CONFIG_GDB_SCRIPTS=y`
I can debug kernel with symbols but no lx- commands.
I have to stay with 4.19 for the moment.

r0ei, 2022-06-06:
Same here with _5.18.2_