raspberrypi/linux #3680: USB subsystem will stop working. Pi 4

JacoFourie, 2020-06-19:
**Describe the bug**
The USB subsystem will stop working at random after hours or days. You can no longer see any connected devices.

Is there a way to get the USB back without having to reboot. Some command I can run to reset all USB.

Only a reboot will fix the issue. Here you can see an lsusb shows no devices and a hub-ctrl says there are no hubs found.

[image]

**To reproduce**
Run my libusb program with a python program that communicates with a RS485 modbus 

**Expected behaviour**
The programs should keep on working. 

**Actual behaviour**
I will start the programs and they may run for days but at some point the USB subsystem will vanish

**System**
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage5

Linux PI4 4.19.118-v7l+ raspberrypi/firmware#1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
Revision        : c03111
Serial          : 1xxxxc
Model           : Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

May 11 2020 18:59:41
Copyright (c) 2012 Broadcom
version 21bfdeee3a6ea823e2113b983390acd1eec8edfb (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       30438344 18726248  10396652  65% /
devtmpfs         1966332        0   1966332   0% /dev
tmpfs            2000124        8   2000116   1% /dev/shm
tmpfs            2000124    18344   1981780   1% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs            2000124        0   2000124   0% /sys/fs/cgroup
/dev/mmcblk0p1    258096    52147    205949  21% /boot
tmpfs             400024        0    400024   0% /run/user/1000

Filename                                Type            Size    Used    Priority
/var/swap                               file            1048572 0       -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20200514
raspberrypi-sys-mods:
  Installed: 20200514
[... 668 lines cut ...]
[177164.922074] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -110
[177165.960854] usb 1-1.3: failed to send control message: -110
[177167.000846] usb 1-1.3: failed to receive control message: -110
[177167.000854] ch341-uart ttyUSB1: failed to read modem status: -110
[177167.130832] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -110
[177167.240834] pl2303 ttyUSB0: error sending break = -110
[179126.115965] pl2303 1-1.1:1.0: pl2303_vendor_write - failed to write [0008]: -110
[179126.225713] pl2303 1-1.1:1.0: pl2303_vendor_write - failed to write [0009]: -110
[179126.335696] pl2303 ttyUSB0: pl2303_get_line_request - failed: -110
[179126.445674] pl2303 ttyUSB0: pl2303_set_line_request - failed: -110
[179126.555698] pl2303 1-1.1:1.0: pl2303_vendor_write - failed to write [0000]: -110
[179126.665679] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -110
[179126.776066] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -110
[179127.116082] usb 1-1.3: failed to send control message: -110
[179128.166013] usb 1-1.3: failed to send control message: -110
[179128.905745] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -110
[179129.015744] pl2303 ttyUSB0: error sending break = -110
[179129.196028] usb 1-1.3: failed to receive control message: -110
[179129.196036] ch341-uart ttyUSB1: failed to read modem status: -110
[179161.676700] cdc_acm 1-1.2:1.0: failed to set dtr/rts
[179169.676559] cdc_acm 1-1.2:1.0: failed to set dtr/rts

**Additional context**
I did ask on the forum.
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262007&p=1597041#p1597041

P33M, 2020-06-19:
Please capture an xhci trace up until the first error message appears in the kernel log.

https://www.raspberrypi.org/forums/viewtopic.php?t=250990&start=75#p1560630

JacoFourie, 2020-06-19:
OK I am now running the trace but as I said it can be hours or days before it will crash.  We run this software on thousands of pis and see the same issue so it is not just the one I am developing on. If there is a way to restart the USB hub via a command that would be great. Then I can issue that if I detect that I can no loger connect to the USB devices.

JacoFourie, 2020-06-19:
I will stop / start the Trace every now and then until it fails so the trace file is not that big

P33M, 2020-06-19:
The trace is recorded to a ring buffer - oldest entries are evicted, so if you are able to intervene soon after an anomaly occurs, there is a good chance that it will span what happened either side of the event.

JacoFourie, 2020-06-21:
OK I managed to get it.

After the crash when I do a lsusb this it what I see

JacoFourie, 2020-06-21:
OK so I managed to get the crash again.

If I look in the trace file all is normal showing this.

[image]

Then after the crash I see this.

[image]

After a lsusb this is what I see

[image]

Here is a link to the trace file.

jdphotography.co.za/files/pi/error_trace.txt

timg236, 2020-06-22:
This should fully reset the XHCI controller but if the attached devices are in a bad state it won't
necessarily resolve the issue i.e. it might be better to reboot on failure

echo 1 >/sys/bus/pci/devices/0000:00:00.0/remove
sleep 2
echo 1 >/sys/bus/pci/rescan

P33M, 2020-06-22:
It would be useful to see what state the PCIe link is in when Linux decides that the xhci controller is dead. In the dead state, does `sudo lspci -vvv` return output for the VLI controller?

JacoFourie, 2020-06-22:
Hi. I have no idea. What commands should I type when it is dead, 'sudo lspci -vvv' ? Then I will share the output. I will have to wait for it to crash again as I rebooted the PI.

JacoFourie, 2020-06-22:
This is the output now while all is whell.

''''
pi@PI4:~ $ sudo lspci -vvv
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 10) (prog-if 00 [Normal decode])
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 51
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 00000000-00000fff
        Memory behind bridge: f8000000-f80fffff
        Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity+ SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: [48] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
        Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0
                        ExtTag- RBE+
                DevCtl: Report errors: Correctable+ Non-Fatal+ Fatal+ Unsupported+
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
                        ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt+
                RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
                RootCap: CRSVisible+
                RootSta: PME ReqID 0000, PMEStatus- PMEPending-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE# ARIFwd-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled ARIFwd-
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                AERCap: First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
        Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
        Capabilities: [240 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=8us PortTPowerOnTime=10us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=1us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=10us
        Kernel driver in use: pcieport

01:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host Controller (rev 01) (prog-if 30 [XHCI])
        Subsystem: VIA Technologies, Inc. VL805 USB 3.0 Host Controller
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 52
        Region 0: Memory at 600000000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [80] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot-,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [90] MSI: Enable+ Count=1/4 Maskable- 64bit+
                Address: 00000000fffffffc  Data: 6540
        Capabilities: [c4] Express (v2) Endpoint, MSI 00
[... 2 lines cut ...]
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <16us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp-
[... cut ...]
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                AERCap: First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
        Kernel driver in use: xhci_hcd
''''

P33M, 2020-06-22:
When the controller is in the dead state, capture the output of lspci first, before trying to do the rescan sequence.

JacoFourie, 2020-06-22:
OK another crash. This is the output of the command after the crash.

pi@PI4:~/Monitor $ sudo lspci -vvv
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 10) (prog-if 00 [Normal decode])
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 51
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 00000000-00000fff
        Memory behind bridge: f8000000-f80fffff
        Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity+ SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: [48] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
        Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0
                        ExtTag- RBE+
                DevCtl: Report errors: Correctable+ Non-Fatal+ Fatal+ Unsupported+
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
                        ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt+
                RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
                RootCap: CRSVisible+
                RootSta: PME ReqID 0000, PMEStatus- PMEPending-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE# ARIFwd-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled ARIFwd-
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                AERCap: First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
        Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
        Capabilities: [240 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=8us PortTPowerOnTime=10us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=1us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=10us
        Kernel driver in use: pcieport

01:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host Controller (rev 01) (prog-if 30 [XHCI])
        Subsystem: VIA Technologies, Inc. VL805 USB 3.0 Host Controller
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 52
        Region 0: Memory at 600000000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [80] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot-,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [90] MSI: Enable+ Count=1/4 Maskable- 64bit+
                Address: 00000000fffffffc  Data: 6540
        Capabilities: [c4] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
[... 14 lines cut ...]
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v1] Advanced Error Reporting
[... cut ...]
pi@PI4:~/Monitor $ echo 1 >/sys/bus/pci/devices/0000:00:00.0/remove
-bash: /sys/bus/pci/devices/0000:00:00.0/remove: Permission denied
pi@PI4:~/Monitor $ sleep 2
pi@PI4:~/Monitor $ echo 1 >/sys/bus/pci/rescan
-bash: /sys/bus/pci/rescan: Permission denied
pi@PI4:~/Monitor $ sudo echo 1 >/sys/bus/pci/devices/0000:00:00.0/remove
-bash: /sys/bus/pci/devices/0000:00:00.0/remove: Permission denied
pi@PI4:~/Monitor $ sleep 2
pi@PI4:~/Monitor $ sudo echo 1 >/sys/bus/pci/rescan
-bash: /sys/bus/pci/rescan: Permission denied
pi@PI4:~/Monitor $

timg236, 2020-06-22:
I think you need to do 'sudo su' and run the reset parameters from a root shell

pelwell, 2020-06-22:
Try either:
```
$ sudo sh -c "echo 1 > /sys/bus/pci/devices/0000:00:00.0/remove"
$ sudo sh -c "echo 1 > /sys/bus/pci/rescan"
```
or
```
$ echo 1 | sudo tee /sys/bus/pci/devices/0000:00:00.0/remove
$ echo 1 | sudo tee /sys/bus/pci/rescan
```

pelwell, 2020-06-22:
Unless you take one of the three approaches above, the `echo` command runs as `root` but the output of the command is still written to the /sys pseudo-files by the current shell, which is probably running as `pi`.

JacoFourie, 2020-06-22:
> sudo sh -c "echo 1 > /sys/bus/pci/rescan"

OK great. That works and I am able to get the USB back and all works again without a reboot. WOW OK that is great. Why does the USB stop working ??

JacoFourie, 2020-06-25:
This time it took 2 days before it crashed. The commands got it back again. Why would id do this? Is it hardware power or software?
Will these commands work on every PI 4,3 and zero? Is it always the same command (can I hardcode it) or do I need to find what to remove first?

JacoFourie, 2020-09-23:
Hi. I have the same issue on the Pi 3. HWat are the same commands to remove and rescan the bus like on the 4 as these commands dont seem to work on the 3. Also Will need for the 2 and Zero just in case.

JacoFourie, 2020-09-23:
I see there are /sys/bus/devices folders.

So I tried my luck with. But it does not work.

sudo sh -c "echo 1 > /sys/bus/usb/devices/remove"
sudo sh -c "echo 1 > /sys/bus/usb/rescan"

JacoFourie, 2020-09-23:
I found this script but it wont work as lsusb brings back nothing. I need to reset the full system not just one device and then rescan for new devices.

#!/bin/bash
#type lsusb to find "vendor" and "product" ID in terminal
 set -euo pipefail
 IFS=$'\n\t'

        #edit the below tow lines of vendor and product values using lsusb result
        dev=$(lsusb -t | grep usbdevicename | grep 'If 1' | cut -d' ' -f13|cut -d"," -f1)
        #VENDOR=05a3
        #PRODUCT=9230
        VENDOR=$(lsusb -s $dev | cut -d' ' -f6 | cut -d: -f1)
        PRODUCT=$(lsusb -s $dev | cut -d' ' -f6 | cut -d: -f2)

        for DIR in $(find /sys/bus/usb/devices/ -maxdepth 1 -type l); do
          if [[ -f $DIR/idVendor && -f $DIR/idProduct &&
                $(cat $DIR/idVendor) == $VENDOR && $(cat $DIR/idProduct) == $PRODUCT ]]; then
            echo 0 > $DIR/authorized
            sleep 0.5
            echo 1 > $DIR/authorized
          fi
        done

P33M, 2020-09-23:
Pi 1-3 use a completely different controller architecture. You can "rescan" the USB bus on a Pi 3 by forcing a disconnect/reconnect via

```
# echo 0 > /sys/devices/platform/soc/3f980000.usb/buspower
# echo 1 > /sys/devices/platform/soc/3f980000.usb/buspower
```
Note that the node address changes between BCM2835 and BCM2836/7.

JacoFourie, 2020-09-28:
@P33M thank you.  What do you mean by "Note that the node address changes between BCM2835 and BCM2836/7."

JacoFourie, 2020-09-28:
Also if I use this command I lose network connection.

P33M, 2020-09-28:
The 3f980000.usb directory turns into 20980000.usb on Pi 0/1.
You will temporarily lose the ethernet adapter on Pi 1-3 as it's connected via USB, but wireless and Pi 4 ethernet will be unaffected.

In practice, network protocols that don't send data during the bus reset typically survive - ssh sessions usually remain open.

amundgr, 2021-04-13:
I am using a RPB4 to boot loop test a USB-unit and has encountered the exact same problem. After about 1 day lsusb yields the following:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The two commands seems to fix it for me to:
```
$ sudo sh -c "echo 1 > /sys/bus/pci/devices/0000:00:00.0/remove"
$ sudo sh -c "echo 1 > /sys/bus/pci/rescan"
```

Are there any plans to fix this? I am not too familiar with how such things get processed, but it looks like a problem worth fixing? 

As mentioned, I am not too experienced with RBP or Linux, but if I can provide any logs or similar to help with the debugging I will happily do so.

JacoFourie, 2021-04-13:
I found that it was bad hardware that made the USB sub system fail. After replacing the hardware (copy of the prolific chip) I have not seen this again.

maratbn, 2021-12-05:
@JacoFourie 

> I found that it was bad hardware that made the USB sub system fail. After replacing the hardware (copy of the prolific chip) I have not seen this again.

By replacing the hardware, do you mean you replaced your whole RP4 with another, or did you replace (un-soldered & re-soldered) one of the chips on the motherboard, and if so was that chip the VLI VL805-Q6 USB controller?  Or do you mean you stopped using one of the USB peripherals you were using, replacing it with another, and after that the problem stopped?

JacoFourie, 2021-12-06:
@maratbn 

Replaced the hardware that connects to the Pi.