raspberrypi/linux #3683: Can't boot Pi1 B+ with threadirqs option

korakios, 2020-06-22:
**Describe the bug**
Raspberry just freezes on boot (can't get any debug info) when adding theadirqs on cmdline.txt

**To reproduce**
Add threadirqs option on boot parameters

**Expected behaviour**
Boot normal with threaded irqs

**Actual behaviour**
freezes on boot, can't ssh , tried to capture the screen via a video capture device from my laptop, nothing useful but I can upload a photo if needed

**System**
Raspberry Pi Model B Plus Rev 1.2

Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2

Apr 15 2020 11:43:08 
Copyright (c) 2012 Broadcom
version 82f9bb929ce2186eb1824178c1ae82902ad6275c (clean) (release) (start)

Linux raspberrypi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux

pelwell, 2020-06-22:
Do you have a particular reason to want to use threaded irqs for everything? It's likely that some drivers don't specify IRQF_NO_THREAD even though they ought to. Try adding "dtoverlay=dwc2" to config.txt, since the dwc_otg driver is probably a culprit here.

korakios, 2020-06-22:
dtoverlay=dwc2 doesn't help , I want to have irq threaded for a project making my Pi as a midi hub . 
I tried the rt kernel but it doesn't work either, so at least I wanted to boot with threadirqs with the stock kernel

pelwell, 2020-06-22:
Well it's on the long list of things to look at, but for now it's an unsupported option.