raspberrypi/linux #3686: USB boot breaks wireless keyboard/mouse

rtfmoz2, 2020-06-23:
**Describe the bug**
Booting with [UGREEN USB 3.0 SSD enclosure](https://www.amazon.com.au/gp/product/B06XWRRMYX) containing [Intel 600GB SSD](https://ark.intel.com/content/www/us/en/ark/products/56569/intel-ssd-320-series-600gb-2-5in-sata-3gb-s-25nm-mlc.html) while using [Logitech USB wireless](https://www.logitech.com/assets/41520/2/wireless-combo-mk220-qsg.pdf)  keyboard/mouse renders the keyboard/mouse unusable.

**To Reproduce**
Boot with devices attached. Cannot login as keyboard only responds occasionally. In desktop mode the mouse only moves in pulses.

**Expected behaviour**
Keyboard and Mouse work normally as they do when USB drive is not present.

**System**
Raspberry Pi4 4GB v1.1 64bit Beta with USB boot
https://gist.github.com/rtfmoz2/37f0c3579415c74e4b5d7d772c923215

**USB boot (please complete the following information):**
lsusb -vvv
https://gist.github.com/rtfmoz2/1d1ad48f4cdd5987bd688c4472c2b367

JamesH65, 2020-06-23:
If you put the keyboard receiver on a short extension cable, does the problem go away?

rtfmoz2, 2020-06-23:
Fixed. I am surprised there is so much signal interference around the Pi4. Is that normal or its the SSD that's contributing somehow?

rtfmoz2, 2020-06-23:
> > If you put the keyboard receiver on a short extension cable, does the problem go away?
> 
> Yes that fixed it. I am surprised there is so much signal interference around the Pi4. How is the SSD contributing to this?

As an experiment, I wrapped the SSD in alfoil and that made no difference. Maybe the interference is coming off the cord? I am using the Ethernet port as well.

rtfmoz2, 2020-06-23:
I tried an alfoil insulator between the two cords, no difference. I can only surmise it must be coming from the Pi4 itself. When I run the next meetup I will find someone with a signal tester and we can figure out where the noise is coming from.

rtfmoz2, 2020-06-23:
So I started testing wireless dongles, a newer logic unifying receiver had the identical problem. Then I tested a [Microsoft Sculpt Ergonomic](https://www.microsoft.com/accessories/en-us/products/keyboards/sculpt-ergonomic-desktop/l5v-00001) Desktop and it had the same problem.

rtfmoz2, 2020-06-24:
So I disabled wifi and bt on the Pi4. Preliminary testing looks good. I just need to do some negative testing as well to confirm. Will update this posts.

JamesH65, 2020-06-24:
USB3 and wireless are renowned for not getting along, and not just on the Pi, Apple products and others have reportedly the same issues. The problem is the very high speeds of USB3 causing RFI that interferes with the wireless.

rtfmoz2, 2020-06-26:
Ok well lets leave this one for now, its not something I think the linux repo can resolve anyway.

closed 2020-06-26 by rtfmoz2