raspberrypi/linux #3694: vc4-fkms-v3d: problems with interlaced modes

DarkAyron, 2020-06-28:
The vc4 driver doesn't handle interlaced modes correctly. When an interlaced video mode is set in config.txt, as soon as the vc4 module is loaded, output switches to progressive.

To reproduce the bug add the following to config.txt:
hdmi_group=1
hdmi_mode=20
hdmi_drive=2
hdmi_pixel_encoding=3
dtoverlay=vc4-fkms-v3d

reboot and watch the output of the HDMI port. The Pi starts with 1080i and as soon as the vc4 kernel module loads, the output changes to 1080p. The same thing happens with mode 5, 6, 7, 21 and 22 but changing to 480p, 576p or 1080p respectively.

Especially when driving a CRT not supporting 1080p or only SD, this causes it to display garbage or nothing at all (and in the worst case it ends with a puff of smoke and a broken horizontal deflection driver).

This was tested on an Raspberry Pi 3B+ and 4B. Both behave the same.
Image used is release 2020-05-27
Kernel version 4.19.118-v7+ #1311

6by9, 2020-06-29:
Interlacing is a non-trivial one from the configuration perspective as to field order, offsets, and the like. AFAICT the kernel frameworks don't give you that information and rely on the driver working it out for itself by magic. There isn't a generic answer for those settings (eg PAL is upper field first, but NTSC is lower field first), and varies based on output format as well.

Without having an easy solution on that, it was quicker to disable interlaced modes through FKMS until time has been found to resolve the differences. It's a matter of priorities.

Interlaced modes are supported via composite output as there the mode is fixed by the firmware, and the kernel just follows what is set in config.txt.

6by9, 2020-06-30:
I had a quick look and it turned out simpler than anticipated.
PR up for the kernel change, but there's also a firmware update required.

bellum07, 2020-07-25:
Unfortunately the patch doesn’t work here with a Raspberry Pi 2B and a Sony TV.
I’ve upgraded my Raspberry Pi OS lite to the latest version with Kernel:
   Linux raspberrypi 5.4.51-v7+ #1326 SMP Fri Jul 17 10:45:25 BST 2020 armv7l GNU/Linux
and added
   hdmi_group=1
   hdmi_mode=20

Without FKMS this works and the TV displays 1080i.
   root@raspberrypi:~# tvservice -sn
   state 0xa [HDMI CEA (20) RGB lim 16:9], 1920x1080 @ 50.00Hz, interlaced
   device_name=SNY-SONY_TV

But I want to use gstreamer with kmssink.
So I added 
    [all]
    dtoverlay=vc4-fkms-v3d
in config.txt but now the TV does only show a black screen and the error message “unsupported signal”.

Data for analysis:
root@raspberrypi:~# cat /proc/cmdline
coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1
video=HDMI-A-1:1920x1080M@50i,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48
vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=d2bdc27e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

Raw EDID in base64:
AP///////wBN2QJ0AQEBAQEWAQOAWTJ4Cg3JoFdHmCcSSEwhCACBgAEBAQEBAQEBAQEBAQEBAjqA
GHE4LUBYLEUAdvIxAAAeAR0AclHQHiBuKFUAdvIxAAAeAAAA/ABTT05ZIFRWCiAgICAgAAAA/QAw
Pg5GDwAKICAgICAgAcACAzvwUx8QFAUTBCAiPD4SFgMHERUCBgEmCQcHFQdQgwEAAHQDDAAgALgt
L9AKAUAAfyAwcICQduIA+wI6gNByOC1AECxFgHbyMQAAHgEdALxS0B4guChVQHbyMQAAHgEdgNBy
HBYgECwlgHbyMQAAngAAAAAAAAAAAAAAAAAAVQ==

Why does it work in normal but not in FKMS mode?
If you need additional data or tests, just say.

DarkAyron, 2020-07-25:
I've tried this with a Raspberry Pi 3B+ and 4B with CEA mode 21. The output was a blank video with incorrect synchronisation.
This is how it looks on my monitor:
[photo](http://0wx.cat/076e6d5ca5286dd4b34cf1d4533e9880.jpg) The H/V-Delay button was not pressed.

DaanPeters, 2020-07-28:
I have the same here with my Raspberry 4. The HDMI signal gets recognized as 1920x540 in stead of 1080i.

bellum07, 2020-07-28:
I've compiled modetest on my raspberry an ran it with the fautly HDMI mode in a ssh session: modetest -M vc4 (output see below)

What makes me suspicious are the following lines:

CRTCs:
id	fb	pos	size
52	61	(0,0)	(1920x1080)
  #0 1920x1080i 25.00 1920 2448 2492 2640 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver

To me it seems that the vc4 dirver selects only 25Hz for the 1920x1080i resolution.
I think this sould be 50Hz instead but I'm not sure.
Perhaps an expert can clarify this.
It would be great if this would be working in the future.

-----

pi@raspberrypi:~ $ modetest -M vc4
Encoders:
id	crtc	type	possible crtcs	possible clones	
53	52	TMDS	0x00000001	0x00000000

Connectors:
id	encoder	status		name		size (mm)	modes	encoders
54	53	connected	HDMI-A-1       	890x500		28	53
  modes:
	index name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  #0 1920x1080 60.00 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: preferred, driver
  #1 1920x1080 59.94 1920 2008 2052 2200 1080 1084 1089 1125 148352 flags: phsync, pvsync; type: driver
  #2 1920x1080i 30.00 1920 2008 2052 2200 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  #3 1920x1080i 29.97 1920 2008 2052 2200 1080 1084 1094 1125 74176 flags: phsync, pvsync, interlace; type: driver
  #4 1920x1080 50.00 1920 2448 2492 2640 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: driver
  #5 1920x1080i 25.00 1920 2448 2492 2640 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  #6 1920x1080 30.00 1920 2008 2052 2200 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  #7 1920x1080 29.97 1920 2008 2052 2200 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  #8 1920x1080 24.00 1920 2558 2602 2750 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  #9 1920x1080 23.98 1920 2558 2602 2750 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  #10 1280x1024 60.02 1280 1328 1440 1688 1024 1025 1028 1066 108000 flags: phsync, pvsync; type: driver
  #11 1280x720 60.00 1280 1390 1430 1650 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #12 1280x720 59.94 1280 1390 1430 1650 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  #13 1280x720 50.00 1280 1720 1760 1980 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #14 1280x720 30.00 1280 3040 3080 3300 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #15 1280x720 29.97 1280 3040 3080 3300 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  #16 1280x720 24.00 1280 3040 3080 3300 720 725 730 750 59400 flags: phsync, pvsync; type: driver
  #17 1280x720 23.98 1280 3040 3080 3300 720 725 730 750 59341 flags: phsync, pvsync; type: driver
  #18 1024x768 60.00 1024 1048 1184 1344 768 771 777 806 65000 flags: nhsync, nvsync; type: driver
  #19 800x600 60.32 800 840 968 1056 600 601 605 628 40000 flags: phsync, pvsync; type: driver
  #20 720x576 50.00 720 732 796 864 576 581 586 625 27000 flags: nhsync, nvsync; type: driver
  #21 720x576i 25.00 720 732 795 864 576 580 586 625 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #22 720x480 60.00 720 736 798 858 480 489 495 525 27027 flags: nhsync, nvsync; type: driver
  #23 720x480 59.94 720 736 798 858 480 489 495 525 27000 flags: nhsync, nvsync; type: driver
  #24 720x480i 30.00 720 739 801 858 480 488 494 525 13514 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #25 720x480i 29.97 720 739 801 858 480 488 494 525 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #26 640x480 60.00 640 656 752 800 480 490 492 525 25200 flags: nhsync, nvsync; type: driver
  #27 640x480 59.94 640 656 752 800 480 490 492 525 25175 flags: nhsync, nvsync; type: driver
  props:
	1 EDID:
		flags: immutable blob
		blobs:

		value:
			00ffffffffffff004dd9027401010101
			01160103805932780a0dc9a057479827
			12484c21080081800101010101010101
			010101010101023a801871382d40582c
			450076f23100001e011d007251d01e20
			6e28550076f23100001e000000fc0053
			4f4e592054560a2020202020000000fd
			00303e0e460f000a20202020202001c0
			02033bf0531f101405130420223c3e12
			16030711150206012609070715075083
[... 193 lines cut ...]
			 NV21:  LINEAR
			 P030:  (UNKNOWN MODIFIER)
	47 alpha:
		flags: range
		values: 0 65535
		value: 65535
	48 rotation:
		flags: bitmask
		values: rotate-0=0x1 rotate-180=0x4 reflect-x=0x10 reflect-y=0x20
		value: 1
	49 COLOR_ENCODING:
		flags: enum
		enums: ITU-R BT.601 YCbCr=0 ITU-R BT.709 YCbCr=1 ITU-R BT.2020 YCbCr=2
		value: 0
	50 COLOR_RANGE:
		flags: enum
		enums: YCbCr limited range=0 YCbCr full range=1
		value: 0
	51 zpos:
		flags: range
		values: 0 127
		value: 0

Frame buffers:
id	size	pitch

6by9, 2020-07-28:
DRM always works in frames per second. 1080i50 is 50 fields per second, or 25 frames per second, and each field is 1920x540.

Please can you provide the output of ```tvservice -s```? I'll try to have another look at it later in the week. It was working fine for me with the monitor happily reporting 1080i.

bellum07, 2020-07-28:
> Please can you provide the output of tvservice -s?

In which mode should i run tvservice -s?
The non working FKMS mode or the working non FKMS mode?

6by9, 2020-07-29:
> In which mode should i run tvservice -s?
The non working FKMS mode or the working non FKMS mode?

At least FKMS, but ideally both.

DaanPeters, 2020-07-29:
The output for tvservice -s is the following:
FKMS:
state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 50.00Hz, interlaced

non FKMS:
state 0xa [HDMI CEA (20) RGB lim 16:9], 1920x1080 @ 50.00Hz, interlaced

Where the non FKMS version is working and the FKMS version fails.

bellum07, 2020-07-29:
Same here:
Output of tvservice -s

FKMS:
state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 50.00Hz, interlaced

non FKMS:
state 0xa [HDMI CEA (20) RGB lim 16:9], 1920x1080 @ 50.00Hz, interlaced

Non FKMS version is working and the FKMS version produces an error message "unsupported signal" on my TV.

6by9, 2020-07-29:
OK, I'd missed some of the vertical timing changes required for interlaced when NOT also double clocked, so 480i was working, but not 1080i.
I've got a potential patch, but now want to go through all the combinations of standard modes before pushing it.

bellum07, 2020-07-30:
Sounds great.
Looking forward to test the fix...
Thanks!

popcornmix, 2020-08-04:
Should be in latest rpi-update firmware

bellum07, 2020-08-06:
I'm sorry, I'm quite new to the Raspberry Pi - is 'rpi-update firmware' the command to get and install the fixed firmware?
If not, how to test?
Thanks!

JamesH65, 2020-08-07:
Just

`rpi-update`

6by9, 2020-08-07:
https://www.raspberrypi.org/documentation/raspbian/applications/rpi-update.md

bellum07, 2020-08-08:
> Just
> `sudo rpi-update`

Thanks rpi-update worked and I can confirm that now 1080i mode with hdmi_group=1 and hdmi_mode=20 in config.txt now also works with my TV correctly.
Thanks for fixing this.
When will it go to the official 5.4 kernel?

6by9, 2021-04-16:
Fixed back in August - closing.

closed 2021-04-16 by 6by9