raspberrypi/linux #3699: V4L2 failure in arm64 with HQ camera

satmandu, 2020-07-01:
Connected Forum post here: https://www.raspberrypi.org/forums/viewtopic.php?p=1688634#p1688634

**Describe the bug**
Using raspios 64-bit, installed cheese and getting this error on the RPI4 8GB (with no image):

**System**

https://paste.debian.net/1154643/

**Logs**
```
[   53.146306] x21: 0000000004b00000 x20: 000000000000000f 
[   53.146309] x19: 0000000000000cc4 x18: ffffffc010e38888 
[   53.146312] x17: 0000000000000000 x16: 0000000000000000 
[   53.146315] x15: ffffffc010f71670 x14: 00000000fffffffe 
[   53.146318] x13: ffffffc010f712cf x12: ffffff81f51f5bb0 
[   53.146321] x11: 0000000000000000 x10: 0000000000000168 
[   53.146325] x9 : 0000000000000003 x8 : 0000000000000168 
[   53.146328] x7 : 0000000000000000 x6 : ffffff81f71c3158 
[   53.146331] x5 : ffffff81f71c3158 x4 : 0000000000000000 
[   53.146334] x3 : 0000000000000000 x2 : 0000000000000000 
[   53.146337] x1 : 0000000000000000 x0 : ffffffc010e38888 
[   53.146340] Call trace:
[   53.146345]  __alloc_pages_nodemask+0x2d4/0x338
[   53.146348]  __dma_direct_alloc_pages+0x180/0x230
[   53.146351]  arch_dma_alloc+0xe0/0x190
[   53.146354]  dma_direct_alloc+0x80/0x98
[   53.146358]  dma_alloc_attrs+0x88/0xf8
[   53.146367]  vb2_dc_alloc+0x70/0x130 [videobuf2_dma_contig]
[   53.146377]  __vb2_queue_alloc+0x198/0x438 [videobuf2_common]
[   53.146383]  vb2_core_reqbufs+0x2ac/0x400 [videobuf2_common]
[   53.146389]  vb2_ioctl_reqbufs+0x98/0xc8 [videobuf2_v4l2]
[   53.146411]  v4l_reqbufs+0x50/0x60 [videodev]
[   53.146427]  __video_do_ioctl+0x234/0x3f0 [videodev]
[   53.146443]  video_usercopy+0x274/0x6b0 [videodev]
[   53.146459]  video_ioctl2+0x3c/0x50 [videodev]
[   53.146475]  v4l2_ioctl+0x64/0x90 [videodev]
[   53.146481]  do_vfs_ioctl+0xc8/0xa08
[   53.146484]  ksys_ioctl+0x50/0x80
[   53.146488]  __arm64_sys_ioctl+0x28/0x38
[   53.146494]  el0_svc_common.constprop.1+0xe0/0x1a8
[   53.146497]  el0_svc_handler+0x34/0xa0
[   53.146500]  el0_svc+0x8/0xc
[   53.146503] ---[ end trace 381d2cb7073dfff1 ]---
[   53.146513] bcm2835-isp bcm2835-isp: dma_alloc_coherent of size 78643200 failed
[   53.148730] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x000000001e600000
[   53.148862] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x0000000023100000
[   53.148950] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x0000000027c00000
[   53.491781] cma: cma_alloc: alloc failed, req-size: 19200 pages, ret: -12
[   53.491810] bcm2835-isp bcm2835-isp: dma_alloc_coherent of size 78643200 failed
[   53.492345] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x000000001e600000
[   53.492468] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x0000000023100000
[   53.492551] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x0000000027c00000
```
**Additional context**
Add any other relevant context for the problem.

satmandu, 2020-07-01:
Also on next to most recent firmware: https://paste.debian.net/1154645/

JamesH65, 2020-07-01:
You missed the first, and probably important line of the log. I suspect that the CMA heap has run out. 

[   67.638242] cma: cma_alloc: alloc failed, req-size: 19200 pages, ret: -12`

satmandu, 2020-07-01:
Ah yes, bad copy.
I have ```cma=256M``` in config.txt. That doesn't seem to help though.

pelwell, 2020-07-01:
Try using `dtoverlay=vc4-kms-v3d,cma-size=0x18000000`.

satmandu, 2020-07-01:
So that gets rid of the first error message.

I still get this (and no picture in cheese):
```
[   96.565070] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x0000000016300000
[   96.565240] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x000000001ae00000
[   96.565356] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x000000001f900000
[   96.565467] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x0000000024400000
[   96.907181] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x0000000016300000
[   96.907372] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x000000001ae00000
[   96.907457] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x000000001f900000
[   96.907543] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x0000000024400000
```

https://paste.debian.net/1154651/

satmandu, 2020-07-01:
Also trying with vlc...
[image]

pelwell, 2020-07-01:
We get the idea. It will be some DMA address mapping issue.

satmandu, 2020-07-01:
Sorry, didn't mean to pester! I can be patient and wait for a kernel rev to address this.

6by9, 2020-07-01:
The fact that it logs
```
[   53.146513] bcm2835-isp bcm2835-isp: dma_alloc_coherent of size 78643200 failed
```
```bcm2835-isp``` should be a huge clue that it's using the wrong /dev/video node.  That's going to be a Cheese bug and not a kernel one. Why is it asking for a 78MB image?!
According to man, you can tell Cheese to use a specific device node by launching as ```cheese -d DEVICE```, but it's not obvious what format it wants for DEVICE (it's not apparently /dev/video0).

I'd also suggested qv4l2.

The cache alias message does indicate something wrong in the dma-ranges, but the driver fixes those up anyway.

satmandu, 2020-07-01:
Thanks all. I'll try qv4l2.

satmandu, 2020-07-01:
qv4l2 definitely works.

I do get a pink picture, but I have the IR filter off, so that's somewhat expected.

Thanks all!

satmandu, 2020-07-01:
Update: pink cast resolved in qv4l2 by making the armhf vcdbg run in the 64-bit Raspi-OS and doing ```sudo LD_LIBRARY_PATH=/opt/vc/lib /usr/bin/vcdbg set awb_mode 0``` since ```awb_auto_is_greyworld=1``` in config.txt isn't doing anything. 

Thanks again for your patience in working with me to figure this out.

closed 2020-07-01 by satmandu