raspberrypi/linux #3700: Use of the uart1-midi overlay results in -22 (EINVAL) error when attempting to register UART1.

billj-korg, 2020-07-01:
**Describe the bug**
This is a follow-up to issue #3312. 

```
[    0.186001] Serial: 8250/16550 driver, 1 ports, IRQ sharing enabled
[    0.186598] bcm2835-aux-uart 3f215040.serial: unable to register 8250 port - -22
[    0.186649] bcm2835-aux-uart: probe of 3f215040.serial failed with error -22
```

In that issue, @pelwell asked to try it without using the uart1-midi overlay. I have tried that and verified that UART1 successfully registers. However, when trying to use the uart1-midi overlay, the above error is encountered.

**To reproduce**

**Expected behaviour**
Should be able to use uart1-midi overlay without error.

**Actual behaviour**
EINVAL error code encountered when attempting to register UART1 when the uart1-midi overlay is specified.

**System**
Linux Korg 4.19.120-v8 #1 SMP PREEMPT Mon May 4 13:25:21 UTC 2020 aarch64 GNU/Linux

Running on a CM3.

vcgencmd version:
May  1 2020 18:00:21 
Copyright (c) 2012 Broadcom
version 688a8f8f3d84e788f07f53c93693e1705f68c407 (clean) (release) (start)

**Logs**

**Additional context**

pelwell, 2020-07-02:
There's a nasty initialisation-order problem caused by the fact that certain clocks are initialised very early. In this case, the fixed-factor-clock that adjusts the normal UART baud rate is initialised before the parent clock, and the clock framework handles this by pretending that the source clock is running at 0Hz. Unfortunately the UART only asks for the clock rate once, and zero isn't a useful number, so it is better to interpret that as meaning that the source clock isn't ready and returning -EPROBE_DEFER.

The fix is in the rpi-5.4.y source tree (857e41a). There will likely be an kernel+firmware release in the next few days, at which point you'll be able to `sudo rpi-update` to get the fix.

billj-korg, 2020-07-02:
I am using a custom yocto build, so the rpi-update method is not available. However, I was able to apply your patch to the 4.19 version that our recipe calls for and that allows the midi overlay to be used with UART1 successfully registering. I have a few more hoops to jump through here to be able to confirm that the serial port is indeed functioning as expected. I'll confirm those results as soon as I can test. Thank you for the quick response!

billj-korg, 2020-07-03:
Also confirmed that the serial port is working at the MIDI baud rate. This confirms that the fix in the patch above works when applied to the 4.19.120-v8 kernel.

pelwell, 2020-07-03:
I'm going to close this issue then.

closed 2020-07-03 by pelwell