raspberrypi/linux #3701: RPi4: Dual HDMI display fails with vc4-kms-v3d-pi4 overlay

mszyprow, 2020-07-02:
**Describe the bug**
Dual-TV display over HDMI doesn't work with vc4-kms-v3d-pi4 overlay on RPi4 (1GB version).

**To reproduce**
Connect 2 FHD HDMI TVs to RPi4, add "dtoverlay=vc4-kms-v3d-pi4" to config.txt, run rpi-update to switch to v5.4 kernel, "apt-get install libdrm-tests" to get modetest tool. 

Then run:
`modetest -Mvc4 -s 32:1920x1080 -v`
[wait a while and press ENTER]
`modetest -Mvc4 -s 38:1920x1080 -v`
[wait a while and press ENTER]

DRM crashes and system runs slowly. Kernel dumps "Disabling IRQ #32" message. Futher runs of modetest tool fails.

**System**
2020-05-27-raspios-buster-lite-armhf.img + rpi-update to switch to v5.4 kernel.

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4
* Which OS and version (`cat /etc/rpi-issue`)?
Raspbian GNU/Linux 10
* Which firmware version (`vcgencmd version`)?
Jun 26 2020 14:48:17 
Copyright (c) 2012 Broadcom
version b078012603a6a094178f959805996ad225a0d83b (clean) (release) (start)
* Which kernel version (`uname -a`)?
Linux raspberrypi 5.4.49-v7l+ #1323 SMP Fri Jun 26 14:40:18 BST 2020 armv7l GNU/Linux

**Additional context**
`grep -v '#' /boot/config.txt | tr -s "\n"`
dtparam=audio=on
[pi4]
dtoverlay=vc4-kms-v3d-pi4
[all]
enable_uart=1
core_freq_min=500

**Logs**
`modetest -Mvc4 -c`
Connectors:
id      encoder status          name            size (mm)       modes   encoders
32      31      connected       HDMI-A-1        1280x720                38      31
  modes:
        name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  1920x1080 60 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: preferred, driver
  1920x1200 60 1920 2056 2256 2592 1200 1203 1209 1245 193476 flags: nhsync, pvsync; type: userdef
  1920x1080 60 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: driver
  1920x1080 60 1920 2008 2052 2200 1080 1084 1089 1125 148352 flags: phsync, pvsync; type: driver
  1920x1080i 60 1920 2008 2052 2200 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  1920x1080i 60 1920 2008 2052 2200 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  1920x1080i 60 1920 2008 2052 2200 1080 1084 1094 1125 74176 flags: phsync, pvsync, interlace; type: driver
  1920x1080 50 1920 2448 2492 2640 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: driver
  1920x1080i 50 1920 2448 2492 2640 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  1920x1080i 50 1920 2448 2492 2640 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  1920x1080 30 1920 2008 2052 2200 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  1920x1080 30 1920 2008 2052 2200 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  1920x1080 25 1920 2448 2492 2640 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  1920x1080 24 1920 2558 2602 2750 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  1920x1080 24 1920 2558 2602 2750 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  1920x1080 24 1920 2558 2602 2750 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  1280x1024 60 1280 1328 1440 1688 1024 1025 1028 1066 108000 flags: phsync, pvsync; type: driver
  1360x768 60 1360 1424 1536 1792 768 771 777 795 85500 flags: phsync, pvsync; type: driver
  1280x720 60 1280 1390 1430 1650 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  1280x720 60 1280 1390 1430 1650 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  1280x720 50 1280 1720 1760 1980 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  1024x768 60 1024 1048 1184 1344 768 771 777 806 65000 flags: nhsync, nvsync; type: driver
  800x600 60 800 840 968 1056 600 601 605 628 40000 flags: phsync, pvsync; type: driver
  720x576 50 720 732 796 864 576 581 586 625 27000 flags: nhsync, nvsync; type: driver
  720x576 50 720 732 796 864 576 581 586 625 27000 flags: nhsync, nvsync; type: driver
  720x576i 50 720 732 795 864 576 580 586 625 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver
[... 205 lines cut ...]
[ 1089.294574] [<c02022f8>] (__do_softirq) from [<c0227d54>] (irq_exit+0x100/0x110)
[ 1089.294578]  r10:0000016a r9:c1200000 r8:da848800 r7:00000001 r6:00000000 r5:00000000
[ 1089.294582]  r4:c10a82e4
[ 1089.294588] [<c0227c54>] (irq_exit) from [<c0283030>] (__handle_domain_irq+0x70/0xc4)
[ 1089.294593] [<c0282fc0>] (__handle_domain_irq) from [<c02022b8>] (gic_handle_irq+0x4c/0x88)
[ 1089.294598]  r8:f0815000 r7:f0814000 r6:c1201ed0 r5:f081400c r4:c1205a14 r3:c1201ed0
[ 1089.294603] [<c020226c>] (gic_handle_irq) from [<c0201a3c>] (__irq_svc+0x5c/0x7c)
[ 1089.294607] Exception stack(0xc1201ed0 to 0xc1201f18)
[ 1089.294611] 1ec0:                                     c0209b94 00000000 60000093 c021c160
[ 1089.294616] 1ee0: c1204fb4 c1200000 c1204ffc 00000001 c12a30ad c106ca48 0000016a c1201f2c
[ 1089.294620] 1f00: c12053ac c1201f20 00000000 c0209b98 60000013 ffffffff
[ 1089.294625]  r8:c12a30ad r7:c1201f04 r6:ffffffff r5:60000013 r4:c0209b98 r3:3dd3f420
[ 1089.294633] [<c0209b64>] (arch_cpu_idle) from [<c0a58e8c>] (default_idle_call+0x34/0x48)
[ 1089.294640] [<c0a58e58>] (default_idle_call) from [<c0255af0>] (do_idle+0xec/0x170)
[ 1089.294645] [<c0255a04>] (do_idle) from [<c0255e50>] (cpu_startup_entry+0x28/0x2c)
[ 1089.294649]  r8:ffffffff r7:c12b5440 r6:c106ca48 r5:c12b5498 r4:000000ce r3:c1201f48
[ 1089.294655] [<c0255e28>] (cpu_startup_entry) from [<c0a52294>] (rest_init+0xbc/0xc0)
[ 1089.294662] [<c0a521d8>] (rest_init) from [<c1000bec>] (arch_call_rest_init+0x18/0x1c)
[ 1089.294666]  r4:c1204f88 r3:c109d270
[ 1089.294672] [<c1000bd4>] (arch_call_rest_init) from [<c10010d4>] (start_kernel+0x4e4/0x518)
[ 1089.294678] [<c1000bf0>] (start_kernel) from [<00000000>] (0x0)
[ 1089.294681] handlers:
[ 1089.294720] [<5b038e4c>] vc4_hvs_irq_handler [vc4]
[ 1089.294724] Disabling IRQ #32
[ 1099.379128] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:72:crtc-3] flip_done timed out

pelwell, 2020-07-02:
`dtoverlay=vc4-kms-v3d-pi4` should not be necessary with an up-to-date firmware - `dtoverlay=vc4-kms-v3d` is sufficient. If you find that it _is_ necessary then I would update your firmware so that it can't be causing other problems.

mszyprow, 2020-07-02:
Okay, I didn't know that `dtoverlay=vc4-kms-v3d` is now supported on RPi4 too. Anyway, the issue is same when I used it.

DylanGauchFutureDesigns, 2020-10-06:
I am having the same issue. When I enable dtoverlay=vc4-kms-v3d, I am able to access the i2c buses, but am not able to use both HDMI ports. However, if I switch it back to dtoverlay=vc4-fkms-v3d I am able to use both HDMI ports and not able to access the i2c busses. I need to be able to do both.

JamesH65, 2020-10-06:
Standard KMS is not yet out of beta, so should not be expected to work in all cases. As for I2C not working with FKMS, I suspect you have a bad DT tree overlay entry (or none at all).

DylanGauchFutureDesigns, 2020-10-06:
When using FKMS, I am having the same issues as described at issue #3152.

6by9, 2020-10-09:
> When using FKMS, I am having the same issues as described at issue #3152.

Correct. With FKMS the firmware is in charge of the HDMI I2C interfaces.

If you want to use vc4-kms-v3d on a Pi4, then please compile your own kernel from the rpi-5.9.y branch as that is now fairly stable with the patches that have been accepted for the mainline kernel.
There is still a call to make as to whether we attempt to backport to rpi-5.4.y, or jump forward to rpi-5.9.y sooner than we might otherwise.