raspberrypi/linux #3702: enc28j60 random disconnects

ComexioDev, 2020-07-03:
**Describe the bug**
I use a enc28j60 ethernet module with spi on my CM3+. Sometimes it randomly disconnects (about once every day) with following error (dmesg):
`eth0 (enc28j60): transmit queue 0 timed out`

**To reproduce**
- use a enc28j60
- wait for ethernet to disconnect

**Expected behaviour**
- Work without disconnects

**Actual behaviour**
- Ethernet disconnects randomly

**System**
`System Information
------------------

Raspberry Pi Compute Module 3 Plus Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage4

Linux raspberrypi 5.4.47-v7+ #1322 SMP Wed Jun 17 17:52:39 BST 2020 armv7l GNU/Linux
Revision	: a02100
Serial		: 000000001ef8fa95
Model		: Raspberry Pi Compute Module 3 Plus Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jun 19 2020 15:44:36 
Copyright (c) 2012 Broadcom
version 31eda665dcc45e61650a2ef018ef2cf94091c46f (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Dateisystem    1K-Blöcke Benutzt Verfügbar Verw% Eingehängt auf
/dev/root        7189324 5617380   1230756   83% /
devtmpfs          441236       0    441236    0% /dev
tmpfs             474516       0    474516    0% /dev/shm
tmpfs             474516   12352    462164    3% /run
tmpfs               5120       4      5116    1% /run/lock
tmpfs             474516       0    474516    0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   54429    203666   22% /boot
tmpfs              94900       4     94896    1% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installiert:           1.20200611
raspberrypi-sys-mods:
  Installiert:           20200514
openbox:
  Installiert:           3.6.1-8+rpt4
lxpanel:
  Installiert:           0.10.0-2+rpt11
[... 574 lines cut ...]
[ 2536.103142]  r8:00000000 r7:80c67a40 r6:80daa200 r5:80daa258 r4:000000ce r3:80d01f50
[ 2536.103154] [<80153750>] (cpu_startup_entry) from [<808cab48>] (rest_init+0xbc/0xc0)
[ 2536.103169] [<808caa8c>] (rest_init) from [<80c00bd8>] (arch_call_rest_init+0x18/0x1c)
[ 2536.103175]  r4:80d04f48 r3:80c96260
[ 2536.103187] [<80c00bc0>] (arch_call_rest_init) from [<80c01090>] (start_kernel+0x4b4/0x4e8)
[ 2536.103198] [<80c00bdc>] (start_kernel) from [<00000000>] (0x0)
[ 2536.103204] ---[ end trace 694e100ebaed3721 ]---
[ 2536.137096] enc28j60 spi0.0 eth0: link down
[ 2536.174263] enc28j60 spi0.0 eth0: multicast mode
[ 2536.174817] enc28j60 spi0.0 eth0: multicast mode
[ 2536.199726] enc28j60 spi0.0 eth0: link up - Half duplex
[ 2536.223212] enc28j60 spi0.0 eth0: multicast mode
[ 2536.370228] enc28j60 spi0.0 eth0: multicast mode
[ 2536.370387] enc28j60 spi0.0 eth0: multicast mode
[ 2542.255617] enc28j60 spi0.0 eth0: multicast mode
`

**Logs**
see above

**Additional context**
I tested it with the current stable kernel and i testet it with the new kernel. Both have the same Problem.
While i was searching i found a forum post [https://www.raspberrypi.org/forums/viewtopic.php?t=232969](https://www.raspberrypi.org/forums/viewtopic.php?t=232969). It doesn't contain any answer and no link to any bugreport.

I hope this is the right way of reporting this and i am sorry for any inconvenience. Feel free to correct me or ask for any additional information.

pelwell, 2020-07-03:
I have been unable to reproduce the "transmit queue 0 timed out" failure, which makes it hard to diagnose.

ComexioDev, 2020-07-03:
Can i give you any additional information to help you reproduce?

pelwell, 2020-07-03:
That's so hard to say - you have given a fairly full, clear report. Perhaps you could say something about the other hardware attached?

ComexioDev, 2020-07-03:
/boot/config:

# Display
hdmi_force_hotplug=1
dtoverlay=dpi24
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=800
framebuffer_height=445
framebuffer_aspect=-1
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x6f007
dpi_timings=800 0 40 48 88 480 0 13 3 32 0 0 0 60 0 32000000 6

# Ethernet
dtparam=spi=on
dtoverlay=spi-gpio35-39-com
dtoverlay=enc28j60,int_pin=35

# Touchscreen
dtparam=i2c_vc=on
dtoverlay=i2c0,pins_44_45=1
dtoverlay=touchScreen

As you can see, i use spi and i²c. 

The enc28j60 int_pin is wired to pin 35, so i had to modify the gpio35-39 overlay to take pin 40 instead of pin 35.

pelwell, 2020-07-03:
The `touchScreen` overlay isn't a standard one - what's in it, or at least how is the touchscreen device attached?

ComexioDev, 2020-07-06:
Its a Focaltech touchscreen with the following overlay, connected with i²c on GPIO 44-45:

```
/dts-v1/;
/plugin/;
/ {
        compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";
        fragment@0 {
                target = <&i2c0>;
                __overlay__ {
			#address-cells = <1>;
                        #size-cells = <0>;
						
                         polytouch: edt-ft5x06@38 {
                                compatible = "edt,edt-ft5406", "edt,edt-ft5x06";
                                reg = <0x38>;
                                pinctrl-names = "default";
                                interrupt-parent = <&gpio>;
                                interrupts = <42 2>; /* rising edge */
                                touchscreen-size-y = <800>;
                                touchscreen-size-x = <480>;
                                touchscreen-swapped-x-y;
                                reset-gpios = <&gpio 43 1>; /* active low */
                        };
                };
        };
};
```

FD-, 2020-07-21:
I'm seeing the same problem (there also seems to be a fairly recent forum thread about it: https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=232969).

Similar to @ComexioDev I also have additional device attached to the Pi: I have 3 devices on the I2C bus and am using both SPI interfaces (one to an STM32 and one to the ENC28J60).

I'm on a Pi 3A+ using a 4.19.97 kernel (newest Raspbian image when obtained from the official website back in March).

bhjelstrom, 2022-04-26:
Still an issue on 5.15.32:
[ 2891.055731] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:477 dev_watchdog+0x2fc/0x300
[ 2891.068407] NETDEV WATCHDOG: eth0 (enc28j60): transmit queue 0 timed out
[ 2891.076666] Modules linked in: sha256_generic cfg80211 rfkill enc28j60 8021q spidev garp stp llc bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon dwc2 videodev roles vc_sm_cma(C) mc spi_bcm2835 uio_pdrv_genirq uio fixed sch_fq_codel drm fuse drm_panel_orientation_quirks backlight ipv6
[ 2891.123514] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.15.32-v7 #1
[ 2891.135528] Hardware name: BCM2835
[ 2891.140376] Backtrace: 
[ 2891.144244] [<80a53438>] (dump_backtrace) from [<80a5367c>] (show_stack+0x20/0x24)
[ 2891.154744]  r7:000001dd r6:00000000 r5:80d1933c r4:60000113
[ 2891.161903] [<80a5365c>] (show_stack) from [<80a57c80>] (dump_stack_lvl+0x70/0x94)
[ 2891.172520] [<80a57c10>] (dump_stack_lvl) from [<80a57cbc>] (dump_stack+0x18/0x1c)
[ 2891.183254]  r7:000001dd r6:80927194 r5:00000009 r4:80d980b4
[ 2891.190520] [<80a57ca4>] (dump_stack) from [<8011fcfc>] (__warn+0xfc/0x114)
[ 2891.199104] [<8011fc00>] (__warn) from [<80a53c70>] (warn_slowpath_fmt+0xa8/0xe8)
[ 2891.209726]  r7:80927194 r6:000001dd r5:80d980b4 r4:80d98078
[ 2891.216956] [<80a53bcc>] (warn_slowpath_fmt) from [<80927194>] (dev_watchdog+0x2fc/0x300)
[ 2891.228230]  r8:00000000 r7:82599000 r6:81003d00 r5:825992e8 r4:00000000
[ 2891.236498] [<80926e98>] (dev_watchdog) from [<801aa3b8>] (call_timer_fn+0x40/0x18c)
[ 2891.247330]  r9:81000000 r8:0003f420 r7:00000100 r6:80926e98 r5:825992e8 r4:825992e8
[ 2891.258226] [<801aa378>] (call_timer_fn) from [<801ab8c4>] (run_timer_softirq+0x630/0x6b4)
[ 2891.269852]  r8:0003f420 r7:00000000 r6:81138ce4 r5:825992e8 r4:81001dc4
[ 2891.278248] [<801ab294>] (run_timer_softirq) from [<80101498>] (__do_softirq+0x190/0x440)
[ 2891.289805]  r10:81000000 r9:00000282 r8:81137da0 r7:00000100 r6:00000001 r5:00000002
[ 2891.300998]  r4:81003084
[ 2891.305129] [<80101308>] (__do_softirq) from [<80126ad8>] (irq_exit+0xd8/0x120)
[ 2891.314130]  r10:80f54a60 r9:81000000 r8:81137da0 r7:00000001 r6:00000000 r5:00000000
[ 2891.325238]  r4:ffffe000
[ 2891.329350] [<80126a00>] (irq_exit) from [<80188a80>] (handle_domain_irq+0x70/0x88)
[ 2891.340226]  r5:00000000 r4:80f9210c
[ 2891.345369] [<80188a10>] (handle_domain_irq) from [<80101268>] (bcm2836_arm_irqchip_handle_irq+0x5c/0x60)
[ 2891.358115]  r7:81001f04 r6:ffffffff r5:60000013 r4:80108974
[ 2891.365352] [<8010120c>] (bcm2836_arm_irqchip_handle_irq) from [<80100aec>] (__irq_svc+0x4c/0x64)
[ 2891.377351] Exception stack(0x81001ed0 to 0x81001f18)
[ 2891.383943] 1ec0:                                     00b1a34c 00000000 00000001 80119700
[ 2891.395124] 1ee0: 81000000 00000000 81005058 810050a8 81137da0 80d0f7bc 80f54a60 81001f2c
[ 2891.406289] 1f00: 81001f30 81001f20 80108970 80108974 60000013 ffffffff
[ 2891.414421] [<8010892c>] (arch_cpu_idle) from [<80a63ea4>] (default_idle_call+0x40/0xd8)
[ 2891.425525] [<80a63e64>] (default_idle_call) from [<8015baf0>] (do_idle+0x10c/0x16c)
[ 2891.436369] [<8015b9e4>] (do_idle) from [<8015be84>] (cpu_startup_entry+0x28/0x2c)
[ 2891.447126]  r9:81150000 r8:00000000 r7:ffffffff r6:81005000 r5:80f54a60 r4:000000e4
[ 2891.458102] [<8015be5c>] (cpu_startup_entry) from [<80a5b694>] (rest_init+0xc8/0xd0)
[ 2891.469196] [<80a5b5cc>] (rest_init) from [<80f00c24>] (arch_call_rest_init+0x18/0x1c)
[ 2891.480500]  r5:80f54a60 r4:81150068
[ 2891.485727] [<80f00c0c>] (arch_call_rest_init) from [<80f0137c>] (start_kernel+0x6c4/0x710)
[ 2891.497423] [<80f00cb8>] (start_kernel) from [<00000000>] (0x0)
[ 2891.505070] ---[ end trace 6103e215c512cb35 ]---
[ 2891.540906] enc28j60 spi0.0 eth0: link down
[ 2891.550357] enc28j60 spi0.0 eth0: link down
[ 2891.611752] enc28j60 spi0.0 eth0: link up - Half duplex
[ 2891.636598] enc28j60 spi0.0 eth0: multicast mode
[ 2897.335100] enc28j60 spi0.0 eth0: multicast mode

[image]

BushyTaa, 2022-05-12:
I was also seeing the same issue with a Zero 2 W. 
The timeouts were occuring when RPi.GPIO atempted to set up on the same channel as enc28j60's interrupt (GPIO 25 in my case).

gerwsx, 2024-01-16:
I tried both enc28j60 and w5500 on a Raspberry Zero 2W with latest Raspberry Pi OS (bookworm). Both seem to show this issue (latest updates applied, `6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux`). I can easily reproduce it by sending pings in a timely manner. 

`mtr -i 0.5 -s 1400 -n -o "LSRD NBAW  X" hostonmynetwork`

When executing with lower/higher intervals(-i) the issue is harder to reproduce or doesn't happen at all.

```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 706626  bytes 461926722 (440.5 MiB)
        RX errors 0  dropped 147456 overruns 0  frame 0
        TX packets 261872  bytes 358885678 (342.2 MiB)
        TX errors 2  dropped 0 overruns 0  carrier 0  collisions 0

[56819.328223] NETDEV WATCHDOG: eth0 (w5100): transmit queue 0 timed out
[56819.328334] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:525 dev_watchdog+0x278/0x280
[56819.328372] Modules linked in: sctp ip6_udp_tunnel udp_tunnel rfkill snd_bcm2835(C) binfmt_misc snd_pcm bcm2835_v4l2(C) bcm2835_isp(C) videobuf2_vmalloc raspberrypi_hwmon bcm2835_codec(C) bcm2835_mmal_vchiq(C) snd_timer v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev snd mc vc_sm_cma(C) raspberrypi_gpiomem uio_pdrv_genirq uio i2c_dev drm fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 w5100_spi w5100 spidev spi_bcm2835
[56819.328626] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G         C         6.1.0-rpi7-rpi-v8 #1  Debian 1:6.1.63-1+rpt1
[56819.328644] Hardware name: Raspberry Pi Zero 2 W Rev 1.0 (DT)
[56819.328653] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[56819.328669] pc : dev_watchdog+0x278/0x280
[56819.328684] lr : dev_watchdog+0x278/0x280
[56819.328698] sp : ffffffc00800bdc0
[56819.328705] x29: ffffffc00800bdc0 x28: ffffffe222c612c4 x27: ffffffc00800bee0
[56819.328732] x26: ffffffe223345008 x25: 0000000000000000 x24: ffffffe22366d898
[56819.328758] x23: ffffffe223666000 x22: 0000000000000000 x21: ffffff80027ea3dc
[56819.328784] x20: ffffff80027ea000 x19: ffffff80027ea488 x18: ffffffffffffffff
[56819.328809] x17: ffffff9df3864000 x16: ffffffc008008000 x15: ffffffc00800b9b0
[56819.328835] x14: ffffffe223894a05 x13: 74756f2064656d69 x12: 7420302065756575
[56819.328860] x11: 712074696d736e61 x10: ffffffe2236e47e8 x9 : ffffffe22230eb68
[56819.328885] x8 : 00000000ffffefff x7 : ffffffe2236e47e8 x6 : 80000000fffff000
[56819.328910] x5 : ffffff8016baba10 x4 : 0000000000000040 x3 : 0000000000000001
[56819.328934] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8001e60000
[56819.328959] Call trace:
[56819.328966]  dev_watchdog+0x278/0x280
[56819.328982]  call_timer_fn+0x3c/0x1cc
[56819.329001]  __run_timers+0x258/0x314
[56819.329018]  run_timer_softirq+0x28/0x4c
[56819.329034]  __do_softirq+0x170/0x468
[56819.329049]  ____do_softirq+0x18/0x24
[56819.329065]  call_on_irq_stack+0x24/0x54
[56819.329082]  do_softirq_own_stack+0x24/0x3c
[56819.329098]  __irq_exit_rcu+0xb8/0xe0
[56819.329119]  irq_exit_rcu+0x18/0x40
[56819.329138]  el1_interrupt+0x38/0x70
[56819.329157]  el1h_64_irq_handler+0x18/0x2c
[56819.329175]  el1h_64_irq+0x64/0x68
[56819.329187]  arch_cpu_idle+0x18/0x2c
[56819.329206]  default_idle_call+0x6c/0x1a8
[56819.329222]  do_idle+0x260/0x2a0
[56819.329237]  cpu_startup_entry+0x3c/0x44
[56819.329251]  secondary_start_kernel+0x130/0x154
[56819.329272]  __secondary_switched+0xb0/0xb4
[56819.329294] ---[ end trace 0000000000000000 ]---
```

mikaku, 2025-03-24:
Hello, I'm getting the same message on a Raspberry Pi Zero 2W with Rocky Linux 9 (kernel 6.1.74-6) specially when the system boots:

```
[...]
[   10.241585] NET: Registered protocol family 38
[   10.301534] cryptd: max_cpu_qlen set to 1000
[   12.218143] enc28j60 spi0.0 eth0: link down
[   12.221953] enc28j60 spi0.0 eth0: normal mode
[   12.225698] enc28j60 spi0.0 eth0: multicast mode
[   13.052615] enc28j60 spi0.0 eth0: link up - Half duplex
[   13.626080] enc28j60 spi0.0 eth0: multicast mode
[   20.006878] enc28j60 spi0.0 eth0: multicast mode
[   21.619876] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   29.118569] ------------[ cut here ]------------
[   29.118606] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:473 dev_watchdog+0x330/0x334
[   29.118618] NETDEV WATCHDOG: eth0 (enc28j60): transmit queue 0 timed out
[   29.118627] Modules linked in: cmac algif_hash aes_arm_bs crypto_simd cryptd algif_skcipher af_alg bnep bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem brcmfmac brcmutil hci_uart bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) sha256_generic libsha256 snd_bcm2835(C) videobuf2_vmalloc btbcm videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 cfg80211 bluetooth snd_seq videobuf2_common enc28j60 spidev ecdh_generic snd_seq_device snd_pcm videodev ecc rfkill mc snd_timer spi_bcm2835 snd raspberrypi_hwmon vc_sm_cma(C) fixed uio_pdrv_genirq uio ip_tables x_tables ipv6 crc_ccitt
[   29.118852] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        WC        5.4.225-v7.1.el7 #1
[   29.118857] Hardware name: BCM2835
[   29.118862] Backtrace: 
[   29.118883] [<809449e4>] (dump_backtrace) from [<80944d60>] (show_stack+0x20/0x24)
[   29.118896]  r7:ffffffff r6:00000000 r5:60000113 r4:80e9a6a0
[   29.118911] [<80944d40>] (show_stack) from [<8094bebc>] (dump_stack+0xd8/0x11c)
[   29.118927] [<8094bde4>] (dump_stack) from [<80120354>] (__warn+0xe0/0x108)
[   29.118940]  r10:80ea0420 r9:00000009 r8:808289e4 r7:000001d9 r6:00000009 r5:808289e4
[   29.118947]  r4:80c2081c r3:00000000
[   29.118962] [<80120274>] (__warn) from [<809453c8>] (warn_slowpath_fmt+0xa4/0xd8)
[   29.118972]  r7:000001d9 r6:80c2081c r5:80e04f88 r4:80c207e0
[   29.118987] [<80945328>] (warn_slowpath_fmt) from [<808289e4>] (dev_watchdog+0x330/0x334)
[   29.119001]  r9:00000001 r8:80e03d00 r7:9300c000 r6:96f46500 r5:9300c2a8 r4:00000000
[   29.119018] [<808286b4>] (dev_watchdog) from [<8019dc3c>] (call_timer_fn+0x40/0x1b0)
[   29.119031]  r9:9bf6f440 r8:00000000 r7:ffff9630 r6:808286b4 r5:00000100 r4:9300c2a8
[   29.119046] [<8019dbfc>] (call_timer_fn) from [<8019e4a0>] (run_timer_softirq+0x46c/0x640)
[   29.119058]  r8:00000000 r7:ffff9630 r6:00000000 r5:80e01db4 r4:9300c2a8
[   29.119073] [<8019e034>] (run_timer_softirq) from [<80102504>] (__do_softirq+0x194/0x444)
[   29.119086]  r10:ffffe000 r9:80e00000 r8:00000082 r7:00000100 r6:00000001 r5:00000002
[   29.119092]  r4:80e03084
[   29.119107] [<80102370>] (__do_softirq) from [<801264bc>] (irq_exit+0xdc/0x100)
[   29.119120]  r10:80baa80c r9:80e00000 r8:9743d000 r7:00000001 r6:00000000 r5:00000000
[   29.119126]  r4:ffffe000
[   29.119141] [<801263e0>] (irq_exit) from [<80183130>] (__handle_domain_irq+0x70/0xc4)
[   29.119149]  r5:00000000 r4:80da82e4
[   29.119163] [<801830c0>] (__handle_domain_irq) from [<801022a0>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa8)
[   29.119176]  r9:80e00000 r8:00000001 r7:80e01f04 r6:ffffffff r5:60000013 r4:00000000
[   29.119190] [<80102240>] (bcm2836_arm_irqchip_handle_irq) from [<80101abc>] (__irq_svc+0x5c/0x7c)
[   29.119197] Exception stack(0x80e01ed0 to 0x80e01f18)
[   29.119207] 1ec0:                                     80109908 00000000 60000093 60000093
[   29.119220] 1ee0: ffffe000 80e04fa8 80e04ff0 00000001 00000001 80e9f9df 80baa80c 80e01f2c
[   29.119230] 1f00: 80e00000 80e01f20 00000000 8010990c 60000013 ffffffff
[   29.119238]  r5:60000013 r4:8010990c
[   29.119255] [<801098d8>] (arch_cpu_idle) from [<80952e18>] (default_idle_call+0x40/0x48)
[   29.119270] [<80952dd8>] (default_idle_call) from [<801555f0>] (do_idle+0x124/0x168)
[   29.119281] [<801554cc>] (do_idle) from [<8015590c>] (cpu_startup_entry+0x28/0x2c)
[   29.119295]  r10:9bfff980 r9:80d6ca60 r8:00000000 r7:ffffffff r6:80eb1000 r5:80eb1054
[   29.119302]  r4:000000d1 r3:80d9d27c
[   29.119314] [<801558e4>] (cpu_startup_entry) from [<8094c0a4>] (rest_init+0xb4/0xbc)
[   29.119329] [<8094bff0>] (rest_init) from [<80d00d0c>] (arch_call_rest_init+0x18/0x1c)
[   29.119337]  r5:80eb1054 r4:80e04f88
[   29.119352] [<80d00cf4>] (arch_call_rest_init) from [<80d01238>] (start_kernel+0x4b0/0x4e4)
[   29.119365] [<80d00d88>] (start_kernel) from [<00000000>] (0x0)
[   29.119373] ---[ end trace 27582ef93847ae8b ]---
[   29.124469] enc28j60 spi0.0 eth0: link down
[   29.182671] enc28j60 spi0.0 eth0: link up - Half duplex
[...]
```

Any news on this issue?