raspberrypi/linux #3708: Dev/vcio error

kosepsea, 2020-07-05:
I keep getting this error "raise OSError("No memory to allocate with /dev/vcio")". What can i do to avoid getting that?

JamesH65, 2020-07-05:
What are you doing that promotes this error?

kosepsea, 2020-07-05:
I run this pyton script

import os
import mmap
import struct
import re
import fcntl
import array
import atexit
import ctypes

# Raspberry Pi registers
# https://www.raspberrypi.org/wp-content/uploads/2012/02/BCM2835-ARM-Peripherals.pdf
RPI1_PERI_BASE = 0x20000000
RPI2_3_PERI_BASE = 0x3F000000
# detect board version
try:
    with open("/proc/cpuinfo", "r") as f:
        d = f.read()
        r = re.search("^Revision\s+:\s+(.+)$", d, flags=re.MULTILINE)
        h = re.search("^Hardware\s+:\s+(.+)$", d, flags=re.MULTILINE)
        RPI_1_REVISIONS = ['0002', '0003', '0004', '0005', '0006', '0007',
                           '0008', '0009', '000d', '000e', '000f', '0010',
                           '0011', '0012', '0013', '0014', '0015', '900021',
                           '900032']
        if h is None:
            raise ImportError("This is not raspberry pi board.")
        elif r.group(1) in RPI_1_REVISIONS:
            PERI_BASE = RPI1_PERI_BASE
        elif "BCM2" in h.group(1):
            PERI_BASE = RPI2_3_PERI_BASE
        else:
            raise ImportError("Unknown board.")
except IOError:
    raise ImportError("/proc/cpuinfo not found. Not Linux device?")
PAGE_SIZE = 4096
GPIO_REGISTER_BASE = 0x200000
GPIO_INPUT_OFFSET = 0x34
GPIO_SET_OFFSET = 0x1C
GPIO_CLEAR_OFFSET = 0x28
GPIO_FSEL_OFFSET = 0x0
GPIO_PULLUPDN_OFFSET = 0x94
GPIO_PULLUPDNCLK_OFFSET = 0x98
PHYSICAL_GPIO_BUS = 0x7E000000 + GPIO_REGISTER_BASE

# registers and values for DMA
DMA_BASE = 0x007000
DMA_CS = 0x00
DMA_CONBLK_AD = 0x04
DMA_TI_NO_WIDE_BURSTS = 1 << 26
DMA_TI_SRC_INC = 1 << 8
DMA_TI_DEST_INC = 1 << 4
DMA_SRC_IGNORE = 1 << 11
DMA_DEST_IGNORE = 1 << 7
DMA_TI_TDMODE = 1 << 1
DMA_TI_WAIT_RESP = 1 << 3
DMA_TI_SRC_DREQ = 1 << 10
DMA_TI_DEST_DREQ = 1 << 6
DMA_CS_RESET = 1 << 31
DMA_CS_ABORT = 1 << 30
DMA_CS_DISDEBUG = 1 << 28
DMA_CS_END = 1 << 1
DMA_CS_ACTIVE = 1 << 0
DMA_TI_PER_MAP_PWM = 5
DMA_TI_PER_MAP_PCM = 2
DMA_TI_PER_MAP = (lambda x: x << 16)
DMA_TI_TXFR_LEN_YLENGTH = (lambda y: (y & 0x3fff) << 16)
DMA_TI_TXFR_LEN_XLENGTH = (lambda x: x & 0xffff)
DMA_TI_STRIDE_D_STRIDE = (lambda x: (x & 0xffff) << 16)
DMA_TI_STRIDE_S_STRIDE = (lambda x: x & 0xffff)
[... 155 lines cut ...]
        self._dma.write_int(self._DMA_CHANNEL_ADDRESS + DMA_CS, cs)
        cs &= ~DMA_CS_ACTIVE
        self._dma.write_int(self._DMA_CHANNEL_ADDRESS + DMA_CS, cs)
        cs |= DMA_CS_RESET
        self._dma.write_int(self._DMA_CHANNEL_ADDRESS + DMA_CS, cs)

    def is_active(self):
        """ Check if DMA is working. Method can check if single sequence
            still active or cycle sequence is working.
        :return: boolean value
        """
        cs = self._dma.read_int(self._DMA_CHANNEL_ADDRESS + DMA_CS)
        if cs & DMA_CS_ACTIVE == DMA_CS_ACTIVE:
            return True
        return False

    def current_control_block(self):
        """ Get current dma control block address.
        :return: Currently running DMA control block offset in bytes or None
                 value if DMA is not running.
        """
        cb = self._dma.read_int(self._DMA_CHANNEL_ADDRESS + DMA_CONBLK_AD)
        if cb == 0:
            return None
        return cb - self._phys_memory.get_bus_address()

kosepsea, 2020-07-05:
This is for pycnc ( gcode interpreter for raspberry) its uses dma , I send
full folder , i am running this in raspberry pi 3b , buster os

pelwell, 2020-07-05:
This isn't really a Linux issue - you're just using Linux as a platform to run Python that talks directly to the harware - and it's a question rather than a real problem report, so I'm going to close this issue. You should really start a topic in the Raspberry Pi Forums - [Advanced Users](https://www.raspberrypi.org/forums/viewforum.php?f=29) might be best.

In the meantime, you could:

1. Print out the value of `size` at the point it fails to see if it is a sensible value and to help you calculate a value for point 2.

2. Try adding a `gpu_mem=` setting in config.txt, e.g `gpu_mem=128` for 128MB. If that works you could reduce it to 100 or lower.

By the way, I feel I should say that directly manipulating the hardware from user space is _not_ recommended.

closed 2020-07-05 by pelwell

kosepsea, 2020-07-06:
You mean it's because of gpu_mem memory allocated